o-gs/dji-firmware-tools #176: Error when attempting to calibrate mavic pro gimbal.

thedadio123, 2020-09-27:
Hi im desperate to calibrate my repaired gimbal with no luck.
Mavic is on com5
I have installed python
Install pyserial
I get this error when attempijt calibration. "no response on calibration command jointcoarse request" (the gimbal does not move at all)

I've managed to install the dji virtual com port driver. And I am able to access the drone using file manager and dji assistant 2.

Note: the gimbal goes limp when plugged into USB. Once unplugged it goes back to its stabilised state

Thank you in advance.

robxs1, 2020-10-08:
If you "calibrate " the gimbal in the DJIGO 4 app - what happens ?

thedadio123, 2020-10-10:
Its goes through the calibration fine, but I still have issue with roll it will not hold the horizon

WigHlod, 2022-04-27:
Got almost the same issue here.

I'm able to began the calibration process by launching the DJI Assistant then close it just before run the cmd line.
The problem is the process never finish well. I reproduce this result over MacOS / Windows on different computer : 

I```
Calibration process started; monitoring progress.
Progress: received 33 reports.
Progress: received 48 reports.
Progress: received 63 reports.
Error: No progress tick on calibration request.
```
If i want to relauch the process i must re-open the DJI Assistant, it's like if the door to acces to the Mavic was only open for few seconds before closing.. 

My gimbal work very well, i'm also able to perform a regular calibration from DJI Go app. I just got an hard gimbal knock at startup that i would like to resolve.

robxs1, 2022-04-28:
calibration wont improve you knock issue - the knock could be due to work bands on the vibration board.