o-gs/dji-firmware-tools #177: Mavic Mini 4k unlock?

kapsolas, 2020-10-12:
I'm new to DJI firmware modifications, but understand the process from experience in other areas. 

The Mavic Mini currently supports 2.7K, but comes equipped with an Ambarella H22. Per the datasheet, it should support 4K. 

Has anyone looked into or explored if it can be enabled? I tried to find a FW download of the latest version, but couldn't find out how to get it. 

I suspect it has been looked into and was not possible?

mefistotelis, 2020-10-12:
You may want to ask for that on slack. Invite is here:
https://github.com/Bin4ry/deejayeye-modder#want-to-contribute-or-learn-some-stuff

For firmware download, best way is to use Dank Drone Downloader.

When enabling such functionalities, you have to take into account many possible points of failure - ie. overheating, or power budget. But yeah, that might be doable. For whether anyone tried -> see Slack.

pratikalwaysrocks, 2020-11-28:
@mefistotelis how can I contact you privately? There is something I wanna show you that will be useful for future development.

mefistotelis, 2020-11-28:
> how can I contact you privately?

Isn't my email included in every commit I push to the repository?
It's the same as my github name, at gmail.com.

DataGreed, 2021-07-19:
Hi, did anyone had any luck with this?