o-gs/dji-firmware-tools #178: Mavic Air 2 'Unrecognised response to calibration command JointCoarse request' Error?

SergeantAsh, 2020-10-13:
Hey folks, just tried to use this tool with the Mavic Air 2 following gimbal/camera replacement but upon running each of the two commands, I get this error:

- Unrecognised response to calibration command JointCoarse request
- Unrecognised response to calibration command LinearHall request

You can see the whole process that I used in this video:
https://www.youtube.com/watch?v=6MMpbMhy20Y

While this procedure HAS fixed the wonky yaw (it's now pointing perfectly forwards) there are multiple 'Incorrect gimbal IMU data' and 'camera calibration required, contact DJI' errors and more within the app upon startup.

Any ideas on how to solve this? Happy to jump onto a call with some of the techies to try to troubleshoot if useful!

mefistotelis, 2020-10-13:
I didn't looked into that in detail. There probably is a one-time command to pair the drone with new gimbal.
The "Unrecognised response" errors are most likely not related to the "Incorrect gimbal IMU data".

If you'd like to ask a wider audience for it, repair people are usually discussing things in `#hardware` channel on Slack. Invite is here:
https://github.com/Bin4ry/deejayeye-modder#want-to-contribute-or-learn-some-stuff

SergeantAsh, 2020-10-13:
Thanks @mefistotelis, think that you're right in there being a command there somewhere, I hope that someone finds it soon.

sQueezy1332, 2020-10-16:
Same issue with mini, drone is detected com3 and reconnect like flash storage

pilgrimage80, 2020-11-04:
There is gimbal movement with this command.
py comm_og_service_tool.py COM4 WM160 GimbalCalib JointCoarse

py comm_og_service_tool.py COM4 WM160 GimbalCalib LinearHall
There is no movement on the gimbal with this command. What could be the reason for this?

as a result, gimbal calibration is not successful. What could be the reason for this? 2. I tried it with a different gimbal frame. Room doesn't work. Which cable enables the movement of the gimbal?

thankyou.

pilgrimage80, 2020-11-04:
> Same issue with mini, drone is detected com3 and reconnect like flash storage

To see it as com3 or com4, you need to open assistant dji mavic 2 as step 1, then close the assistant and you can operate with the command.