o-gs/dji-firmware-tools #180: Context guide

kieppie, 2020-11-06:
Hi folks,

Nice project, thanks.
Good way to free me from vendor's hooks. 

I'd like to request basic a starting/context guide, please.
I appreciate the bold statement on the readme: "Step by step instruction: Such instruction will not be provided.", but I'd just like to get some context for the scripts before I brick my device(s).

I'm relatively comfortable in a CLI shell & flashing embedded devices, but just would like to know what I'm running the scripts against - mobile app, remote controller or UAV itself - and if there are any prerequisites I need to satisfy, eg. gain root (if applicable), flashing custom firmware, running on a shell on the device, over serial (directly via UART?), via controller WiFi, or from the connected host, etc. Looks like I may need to pull a binary first?
Also not super clear if there are measures on the vendor software that could cause issues (I know, use at own risk, but...HBD), so that I know to roll back tweaks when app updates or need to send in for servicing.

I'm working from a Linux host, so threads elsewhere referencing "DJI assistant" may or may not be relevant here (see no ref's to dev/tty*)

Any help on a good place to start (besides RTFM/RTFC) would be appreciated, thanks.

Not asking for anyone to hold my hand, certainly no GUI, but rather

mefistotelis, 2020-11-07:
That might be interesting exercise.

So can you propose a specific structure or points which you think should be a part of documentation for each tool?

Maybe you could also read the description of each tool and point out what are you uncertain about?

Or do you lack some general overview rather than specific info about each tool?

mefistotelis, 2020-11-17:
No answer?

So I guess that was one of these cases which I specifically included in the README.
Pity.

closed 2020-11-17 by mefistotelis