o-gs/dji-firmware-tools #181: Mavic Mini Linear Hall calibration failed

Agrestbar, 2020-11-17:
Hi there!
Just tried to perform the calibration of the Linear Hall sensor but it failed. Do you have any ideas what should i check when it comes to the hardware problems? Which part is responsible for the linear hall sensor calibration?

Denis14071982, 2020-11-24:
Был ответ от "Мефистотелис" по этой проблеме! Но я пока не разобрался! Нужен пример!)
"Чтобы это исправить, нам понадобится ответный пакет. Запустите команду с параметром «-vv» и вставьте полный журнал. 
Он должен содержать ответ в виде шестнадцатеричной строки."

mefistotelis, 2020-11-24:
The "Unrecognized response" error has no effect on the actual calibration.

Also, please use English.

Denis14071982, 2020-11-25:
Thanks! That does not affect, I was convinced! But how to remove the message in DJI Fly with code 40011?

MarcoBarba77, 2021-02-13:
hello i have the problem with gimball calibration of my MAVIC MINI .After crush i have put a new cable on the gimball, but i can see only the camera but the gimball dosn't move .Also after calibration , nothink happen , only the JointCoarse move a little be the gimball , but the other command nothing do .
If you have some suggestion , please let me know .
sorry for my english