o-gs/dji-firmware-tools #182: MAVIC MINIGimbal Calibration Error Code 40011

MAG65-MAG65, 2020-11-24:
MAVIC MINI
Can you help me please?

After doing the calibration, I got the error 40011 "Gimbal Calibration Error. Restart aircraft or contact DJI Support if the issue persists (Code :40011)"

how can i fix this error 

HELP ME PLEASE

nettunoroberto, 2020-12-06:
hi, you have solve this error?

moinmoin68, 2020-12-06:
Have the same problem, built the Gimbal
In another Mainboard, error is away, 

So the error is on thé mainboard

denverhan, 2020-12-16:
try downgrage  and than upgrage, if u made it success let me know pls

edyno, 2020-12-16:
Use downgrading to v01.00.0320, gimbal calibration with apk DJI FLY and after flash firmware what you want..

denverhan, 2020-12-16:
> Use downgrading to v01.00.0320, gimbal calibration with apk DJI FLY and after flash firmware what you want..

i expect to fix the code 40011，but i dont have the device to make the test

edyno, 2020-12-16:
> 
> 
> > Use downgrading to v01.00.0320, gimbal calibration with apk DJI FLY and after flash firmware what you want..
> 
> i expect to fix the code 40011，but i dont have the device to make the test

Yes, it fix, tested now.

denverhan, 2020-12-16:
> > > Use downgrading to v01.00.0320, gimbal calibration with apk DJI FLY and after flash firmware what you want..
> > 
> > 
> > i expect to fix the code 40011，but i dont have the device to make the test
> 
> Yes, it fix, tested now.

great,thanks
i may do my own test on sunday,new gimble is flying to me : )
expecting

moinmoin68, 2020-12-16:
Yes that works

robxs1, 2021-01-11:
Ideas on how to calibrate Mini 2 gimbal on V 1.02.000 , antiroll back =1

lrgc99, 2021-07-08:
On mini 2 and air 2 it is no longer possible to downgrade to the necessary versions.
Has anyone found an alternative way?

mEfis-afk, 2021-08-23:
> No mini 2 e no air 2 não é mais possível fazer o downgrade para as versões necessárias.
> Alguém encontrou uma maneira alternativa?

I solve this problem!

moinmoin68, 2021-10-19:
> WhatsApp
> Dji account, not app

You need the Factory firmware.

mEfis-afk, 2021-10-19:
> > […](#)
> 
> VPN DJI Account?

Man, I said I solve, not teach how. This is a public forum, do you think they'll be posting something internal for everyone to see?

Most of you just want free solutions. Idiots!

954426470, 2021-10-19:
> > > […](#)
> > 
> > VPN DJI Account?
> 
> Man, I said I solve, not teach how. This is a public forum, do you think they'll be posting something internal for everyone to see?
> 
> Most of you just want free solutions. Idiots!

a gk****@163.com for dji account

guewald, 2021-10-19:
> > > > […](#)
> > > 
> > > 
> > > VPN DJI Account?
> > 
> > 
> > Man, I said I solve, not teach how. This is a public forum, do you think they'll be posting something internal for everyone to see?
> > Most of you just want free solutions. Idiots!
> 
> a gk****@163.com for dji account

i have interest guewald@gmail.com

ace811, 2021-12-08:
> > > > […](#)
> > > 
> > > 
> > > VPN DJI Account?
> > 
> > 
> > Man, I said I solve, not teach how. This is a public forum, do you think they'll be posting something internal for everyone to see?
> > Most of you just want free solutions. Idiots!
> 
> a gk****@163.com for dji account

Hi ... how to deal with you on the issue?

maurosv, 2022-02-26:
Good afternoon, who solves this error in the mini 2, how much does it cost to give the solution?

mEfis-afk, 2022-02-26:
I decided. Leave your WhatsApp and I will contact you

maurosv, 2022-02-26:
> thank you very much my number is 57-3216166612

954426470, 2022-02-28:
> > thank you very much my number is 57-3216166612
> 
> 

You can send a message to my mailbox

ja4who, 2022-06-25:
> > No mini 2 e no air 2 não é mais possível fazer downgrade para as versões necessárias. 
> > Alguém encontrou uma forma alternativa?
> 
> Eu posso resolver este problema, se você precisar entrar em contato comigo.

MAVIC MINI 2
Can you help me please?

After doing the calibration, I got the error 40011 "Gimbal Calibration Error. Restart aircraft or contact DJI Support if the issue persists (Code :40011)"

how can i fix this error

HELP ME PLEASE

mEfis-afk, 2022-06-25:
> 

Yes, leave your WhatsApp

alxgacc, 2022-10-11:
Who needs remove error 40011, you can write me on WhatsApp +62 812-8430-9143

Oleg73s, 2024-08-13:
Всем доброго! купил новый дрон а там 2 ошибки 40011 и 40021 подскажите как исправить как удалить?

fcoaajr, 2024-08-13:
Coloque o módulo CCD original e calibra! Resolve na hora!

Obter o Outlook para Android<https://aka.ms/AAb9ysg>