o-gs/dji-firmware-tools #184: [03 06] Firmware upgrade finish failed (step = 2, err = 0xaa)

sduyangyi, 2020-11-30:
Here is my story:
I'm trying to changes some parameters of my P3A.
First I got  "V01.11.0030_P3s_dji_system.bin"  from https://github.com/cs2000/DankDroneDownloader
changed the name and flashed it to the drone, and the log shows it's working. 

Then try modify parameters:
Used dji_xv4_fwcon.py -vv -x -p V01.11.0030_P3s_dji_system.bin
Then dji_flyc_param_ed.py -vv -x -m V01.11.0030_P3s_dji_system_m0306.bin
Changed these parameters in the "flyc_param_infos" file:
Parameters:
advanced_function.height_limit_enabled = 2
flying_limit.max_height = 9000

control.horiz_vel_atti_range_0 = 35 (Gps)
control.atti_limit_0 = 50 (Atti)
control.atti_range_0 = 50 (atti upper limit)

control.vert_up_vel = 8
control.vert_down_vel = 8

airport_limit_cfg.cfg_search_radius = 1
airport_limit_cfg.cfg_disable_airport_fly_limit = 1
airport_limit_cfg.cfg_limit_data = 20250910
airport_limit_cfg.cfg_enable[FLY_LIMIT_TYPE_SPECIAL] = 1

advanced_function.driver_license_limit_enable = 2
advanced_function.viechle_license_limit_enable = 2

voltage2.user_set_smart_bat = 0 
voltage2.level_2_function = 0
voltage2.level2_smart_battert_land = 0
voltage2.level_2_voltage = 0
voltage.level_2_protect_type = 0 

Renamed file name to PMCAPPFw3.bin

The flashing stared and ended with long beep and right light.
[P3S_FW_LOG_AB.txt](https://github.com/o-gs/dji-firmware-tools/files/5616992/P3S_FW_LOG_AB.txt)

In the log it said:
 [00024341]========== remo-con disconnect. boot(15) ============
 [00024402]Firmware [C:\PMCAPPFw3.bin] detected, card sn [0x65676124].
 [00024471][03 06] Firmware upgrade start...
 [00067032][03 06] Firmware upgrade finish failed (step = 2, err = 0xaa).
 [00097093]Done.

What should I do at this moment?
Thanks you very much.

mefistotelis, 2020-11-30:
Well, your FC either just got bricked, or it didn't accepted the firmware.

If it got bricked, you may still try to recover by shorting "BOOT" pin an then re-trying the update. The bootloader is still in working condition after all.

If it didn't accepted the firmware - the FC can't just refuse to flash, it does not do any checks to the data it receives. So maybe it didn't receive the data properly? Like ie there was a communication issue between the camera and FC? Hard to tell.

sduyangyi, 2020-11-30:
Hi mefistotelis,
Thanks for the quick reply.
The drone is still working as before. (celling still there)
I'll change to another SD card and give it another try.
Thank you very much.

sduyangyi, 2020-11-30:
other sd card won't help， I guess I just live with whatever I have :)  
Thank you for you help and have a nice day.

closed 2020-11-30 by sduyangyi