o-gs/dji-firmware-tools #186: Dji mini 2 WM161?

ginnolo, 2020-12-31:
DO YOU KNOW THE CODE OF THE MINI 2?

UBFly, 2021-01-20:
> DO YOU KNOW THE CODE OF THE MINI 2?

Did you find out the code? I want to calibrate my gimbal after replacement and need this code.

LeandroBrasill, 2021-01-21:
alguém já sabe ??????

UBFly, 2021-01-23:
I'm looking to use the code to calibrate my gimbal when it arrives. If anyone has successfully done this please let me know. 

I'll post my findings once I try it out probably next week.

LeandroBrasill, 2021-01-23:
change the complete gimbal and it was crooked, has anyone managed to update ??

> Estou tentando usar o código para calibrar meu gimbal quando ele chegar. Se alguém tiver feito isso com sucesso, por favor me avise.
> 
> Vou postar minhas descobertas assim que tentar, provavelmente na próxima semana.

change the complete gimbal and it was crooked, has anyone managed to update ??

UBFly, 2021-01-25:
> change the complete gimbal and it was crooked, has anyone managed to update ??

Try following the instructions here. For model number use WM161.

LeandroBrasill, 2021-01-28:
> > troquei o gimbal completo e ficou torto, alguém conseguiu atualizar ??
> 
> Tente seguir as instruções aqui https://www.youtube.com/watch?v=6MMpbMhy20Y . Para o número do modelo, use WM161.

I already followed all the steps in this video with the code fo dji Mavic mini and it doesn't work and using the code WM161 also doesn't fucio and says that this code doesn't exist in the script, please can someone update the dji mini 2 guimbal for the love of God

UBFly, 2021-01-28:
> I already followed all the steps in this video with the code fo dji Mavic mini and it doesn't work and using the code WM161 also doesn't fucio and says that this code doesn't exist in the script, [...]

Where did you buy the replacement gimbal? Can you ask the seller how to calibrate it?

LeandroBrasill, 2021-01-28:
> > > > troquei o gimbal completo e ficou torto, alguém conseguiu atualizar ??
> > > 
> > > 
> > > Tente seguir as instruções aqui https://www.youtube.com/watch?v=6MMpbMhy20Y . Para o número do modelo, use WM161.
> > 
> > 
> > Já segui todos os passos deste vídeo com o código fo dji Mavic mini e não funciona e usando o código WM161 também não funciona e diz que este código não existe no script, por favor alguém pode atualizar o dji mini 2 guimbal pelo amor de Deus
> 
> Onde você comprou o gimbal substituto? Você pode perguntar ao vendedor como calibrá-lo?

he replied that he cannot do anything, you managed to update yours, he and a Dji mini 2 ???

UBFly, 2021-01-28:
That seller is being useless. My gimbal isn't received yet but I expect it to arrive hopefully tomorrow. I'll keep you posted after I try.

UBFly, 2021-01-29:
Just installed the new gimbal and managed to fix the crooked gimbal issue!

The tricky part is that the DJI Mini 2 is only detected on the COM port for about 5 seconds. Afterwards it disappears. To have it show up again you have to turn off the drone and on again. 

I ran the command as soon as the Mini 2 was detected by windows (you hear the ding). Then the gimbal did the calibration and now it's no longer crooked! Good luck. 

[image]

mefistotelis, 2021-01-29:
Use Dji Assistant to keep the port active. After it stays active, you can close the Assistant.

LeandroBrasill, 2021-01-31:
> Acabei de instalar o novo gimbal e consegui consertar o problema do gimbal torto!
> 
> A parte complicada é que o DJI Mini 2 só é detectado na porta COM por cerca de 5 segundos. Depois, ele desaparece. Para que apareça novamente, você deve desligar o drone e ligá-lo novamente.
> 
> Executei o comando assim que o Mini 2 foi detectado pelo windows (você ouve o ding). Então o gimbal fez a calibração e agora não está mais torto! Boa sorte.
> 
> Editar: o código WM160 funcionou.
> 
> [image]

> Acabei de instalar o novo gimbal e consegui consertar o problema do gimbal torto!
> 
> A parte complicada é que o DJI Mini 2 só é detectado na porta COM por cerca de 5 segundos. Depois, ele desaparece. Para que apareça novamente, você deve desligar o drone e ligá-lo novamente.
> 
> Executei o comando assim que o Mini 2 foi detectado pelo windows (você ouve o ding). Então o gimbal fez a calibração e agora não está mais torto! Boa sorte.
> 
> Editar: o código WM160 funcionou.
> 
> [image]

friend how do i contact you for you explain to me in more detail how did you manage to correct the crooked guimbal ????

UBFly, 2021-01-31:
I followed the instructions given in this video https://www.youtube.com/watch?v=6MMpbMhy20Y. 

For device code use :WM160. 

The only different instructions to that video is running the commands within 5 seconds of connecting the drone. However according to mefistotelis above this may not be required if you install DJI Assistant.

LeandroBrasill, 2021-02-01:
> Segui as instruções fornecidas neste vídeo https://www.youtube.com/watch?v=6MMpbMhy20Y .
> 
> Para usar o código do dispositivo: WM160.
> 
> As únicas instruções diferentes para aquele vídeo são executar os comandos em até 5 segundos após conectar o drone. No entanto, de acordo com o mefistotelis acima, isso pode não ser necessário se você instalar o DJI Assistant.
> 
> Que mais instruções você precisa?

thank you friend i managed to update my gimbal thank you very much, the problem was precisely that the drone didn’t stay in the port for a long time (COM) only just 5 seconds and didn’t show up anymore, more like the friend say that VC entering the Dji assistant there  yes it keeps showing up there and just do the update with the WM160 command that works perfectly, thank you very much

LeandroBrasill, 2021-02-02:
[image]

how to remove this 2 errors that was after the gimbal update ????
has anyone managed to solve this problem ???

UBFly, 2021-02-02:
I get the same two errors and there doesn't seem to be a way to get rid of it. The gimbal works perfectly fine so I don't really care :)

LeandroBrasill, 2021-02-02:
> Recebo os mesmos dois erros e não parece haver uma maneira de me livrar deles. O gimbal funciona perfeitamente bem, então eu realmente não me importo :)

If you downgrade to a previous version and then update again these errors disappear ??

UBFly, 2021-02-03:
Don't know buddy and at this point I'm happy that my mini 2 is in working condition. If you figure it out please share here.

miglsilva, 2021-05-02:
Hello Guys, 
Thank you very much for all content shared. I think it would be very helpful for all Dji mini 2 users.
I also broke my gimbal and I want to replace it by myself. Anyone can please share where I can buy a original gimbal for dji mini 2?

Thanks!

UBFly, 2021-05-03:
> Hello Guys,
> Thank you very much for all content shared. I think it would be very helpful for all Dji mini 2 users.
> I also broke my gimbal and I want to replace it by myself. Anyone can please share where I can buy a original gimbal for dji mini 2?
> 
> Thanks!

They are available for sale on EBay.

miglsilva, 2021-05-03:
My idea is just to buy the Gimbal without the camera. Otherwise, I think that would not worth the value.

UBFly, 2021-05-03:
Unfortunately I couldn't find the gimbal only so had to buy the whole unit

miglsilva, 2021-05-03:
After some hours of research, I found it on an Italian website https://flytodiscover.it/prodotto/dji-mini-2-gimbal/

mefistotelis, 2021-05-20:
For future readers which may be confused by this thread:

There is no special 'code for calibration'. The tool takes aircraft code name just to select proper calibration procedure; the supported platforms are visible inside the script.

Anyway, closing. Nothing interesting here.

Disregard the link, it is just for indexing of wiki:
https://github-wiki-see.page/m/o-gs/dji-firmware-tools/wiki

closed 2021-05-20 by mefistotelis

darkfps, 2023-09-05:
guys i got this error after replacing the gimbal,
everything else working...
Do you know of any alternative to remove it? thanks
[image]

odahcam, 2025-05-04:
I did the process using the instructions from https://mavicpilots.com/threads/fix-incompatible-gimball-error-40021-on-mavics-02-2020.106521/ with a few tweaks to make it work on Mac OS. Yep the Mac COM port name style works just fine, I guess you should know how Python scripts work in order to use this as well.

Mainly I did install `pyserial` using `pip`, found the COM port and ran the commands:

```
python comm_og_service_tool.py --port /path/to/COM WM160 GimbalCalib JointCoarse

python comm_og_service_tool.py --port /path/to/COM WM160 GimbalCalib LinearHall
```

Well, Mini 2 code is WM161 but the tool supports only Mini 1 code, WM160. It works and calibrates the gimbal, but an additional error appears in the DJI Fly app, like @LeandroBrasill showed in his screenshot. I did not yet found a way to get rid of those errors.

odahcam, 2025-05-08:
I could fix 40021 using a paid software that overrides the camera serial number plus a paid firmware from dank drone downloads that is known as calibration software, it won't allow you to fly the drone, but allows you to calibrate and remove error 40011, then I flashed the original firmware back into the drone. It is all green now.