o-gs/dji-firmware-tools #187: No connection to Mavic Pro 2 Gimbal

DerKlotz74, 2021-01-09:
Hi, i have problems to run " py comm_og_service_tool.py com6 WM240 GimbalCalib JointCoarse". I receive the following error message: 
"Traceback (most recent call last):
 File "comm_og_service_tool.py", line 35 , in  <module> 
 import serial
ModuleNotFoundError: No module named ´serial´
Somebody an idea.

mefistotelis, 2021-01-09:
Try reading the error message. Google it if you must.

mc197910, 2021-01-21:
hi need help

DerKlotz74, 2021-01-21:
> 
> 
> Hi, i have problems to run " py comm_og_service_tool.py com6 WM240 GimbalCalib JointCoarse". I receive the following error message:
> "Traceback (most recent call last):
> File "comm_og_service_tool.py", line 35 , in
> import serial
> ModuleNotFoundError: No module named ´serial´
> Somebody an idea.

I sold the camera... Thanks

closed 2021-01-21 by mefistotelis