o-gs/dji-firmware-tools #188: any solution to unsign, decrypt & read fw of mini 2 (wm161) ?

kwankiu, 2021-01-09:
I am New to this tool, trying to kind of extract, decrypt and explore what's inside the firmware of dji mini 2.
tried to unsign the fw and this is what i got

wm161_0306_v03.04.09.43_20200828.pro.fw.sig: Opening for extraction and un-signing
wm161_0306_v03.04.09.43_20200828.pro.fw.sig: Warning: The 2018 format is not fully supported.
wm161_0306_v03.04.09.43_20200828.pro.fw.sig: Warning: Header field 'reserved2' is non-zero; the tool is not designed to handle this.
wm161_0306_v03.04.09.43_20200828.pro.fw.sig: Warning: Header field 'reserved3' is non-zero; the tool is not designed to handle this.
wm161_0306_v03.04.09.43_20200828.pro.fw.sig: Unpacking image...
OrderedDict([('magic', b'IM*H'),
            ('header_version', 2),
            ('size', 1704736),
            ('reserved', b'\x00\x00\x00\x00'),
            ('header_size', 224),
            ('signature_size', 256),
            ('payload_size', 1704256),
            ('target_size', 1704736),
            ('os', 0),
            ('arch', 0),
            ('compression', 0),
            ('anti_version', 0),
            ('auth_alg', 262146),
            ('auth_key', 'PRAK'),
            ('enc_key', 'UFIE'),
            ('scram_key', b'\x04\xb0\xe4kK\\^\x9c\x82F\x02\xe8v\xf5/\x0c'),
            ('name', '0306'),
            ('type', 'FC'),
            ('version', 50596139),
            ('date', 538970152),
            ('reserved2', b'\x8e\xd0\xae\xd7\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'),
            ('userdata', b''),
            ('entry', b'\x00\x00\x00\x00\x00\x00\x00\x00'),
            ('reserved3', b'\xa2WP`'),
            ('chunk_num', 1),
            ('payload_digest', b'\xafl\xe40~\xac\x0f%Pn\xd0\x01i\x02\xd6\x0b\x1c\xdfb\xf38;\x10\xef]\x06B\x05\x8b\xc6\xaan')])
Error: Image file head signature verification failed.

LeandroBrasill, 2021-01-21:
> Eu sou novo nessa ferramenta, tentando meio que extrair, descriptografar e explorar o que está dentro do firmware do dji mini 2.
> tentei cancelar a assinatura do fw e foi isso que consegui
> 
> wm161_0306_v03.04.09.43_20200828.pro.fw.sig: Abrindo para extração e
> desassinatura wm161_0306_v03.04.09.43_20200828.pro.fw.sig: Aviso: O formato 2018 não é totalmente compatível.
> wm161_0306_v03.04.09.43_20200828.pro.fw.sig: Aviso: O campo de cabeçalho 'reservado2' é diferente de zero; a ferramenta não foi projetada para lidar com isso.
> wm161_0306_v03.04.09.43_20200828.pro.fw.sig: Aviso: O campo de cabeçalho 'reservado3' é diferente de zero; a ferramenta não foi projetada para lidar com isso.
> wm161_0306_v03.04.09.43_20200828.pro.fw.sig: Descompactando imagem ...
> OrderedDict ([('magic', b'IM * H '),
> (' header_version ', 2),
> (' size ', 1704736),
> ( 'reservado', b '\ x00 \ x00 \ x00 \ x00'),
> ('tamanho_cabeçalho', 224),
> ('tamanho_assinatura',
> 
> ('target_size', 1704736),
> ('os', 0),
> ('arch', 0),
> ('compressão', 0),
> ('anti_version', 0),
> ('auth_alg', 262146),
> (' auth_key ',' PRAK '),
> (' enc_key ',' UFIE '),
> (' scram_key ', b' \ x04 \ xb0 \ xe4kK \ ^ \ x9c \ x82F \ x02 \ xe8v \ xf5 / \ x0c '),
> ( 'nome', '0306'),
> ('tipo', 'FC'),
> ('versão', 50596139),
> ('data', 538970152),
> ('reservado2', b '\ x8e \ xd0 \ xae \ xd7 \ x00 \ x00 \ x00 \ x00 \ x00 \ x00 \ x00 \ x00 \ x00 \ x00 \ x00 \ x00 \ x00 \ x00 \ x00 \ x00 '),
> (' userdata ', b' '),
> (' entrada ', b '\ x00 \ x00 \ x00 \ x00 \ x00 \ x00 \ x00 \ x00'),
> ('reserved3 ', b' \ xa2WP` '),
> (' chunk_num ', 1),
> ('payload_digest', b '\ xafl \ xe40 ~ \ xac \ x0f% Pn \ xd0 \ x01i \ x02 \ xd6 \ x0b \ x1c \ xdfb \ xf38; \ x10 \ xef] \ x06B \ x05 \ x8b \ xc6 \ xaan ')])
> Erro: A verificação da assinatura do cabeçalho do arquivo de imagem falhou.

Amigo no caso você tentou usa com o código WM161 e não deu certo ??
Estou precisando fazer essa atualização também mais não vi nada a respeito do dji mini 2 tem o código do Mavic mini né isso o WM160 se por acaso eu usa ele no mini 2 irá funcionar VC já fez esse teste????

mefistotelis, 2021-05-20:
This is now supported, so closing.
(though decryption of 2nd level FC firmware is not public)

Disregard the link, it is just for indexing of wiki:
https://github-wiki-see.page/m/o-gs/dji-firmware-tools/wiki

closed 2021-05-20 by mefistotelis

493505110, 2025-08-08:
> This is now supported, so closing.现在已支持，因此关闭。 (though decryption of 2nd level FC firmware is not public)（尽管二级 FC 固件的解密并非公开）
> 
> Disregard the link, it is just for indexing of wiki:忽略该链接，它只是维基索引用 https://github-wiki-see.page/m/o-gs/dji-firmware-tools/wiki

Will 2nd level FC firmware be public?

mefistotelis, 2025-08-08:
> Will 2nd level FC firmware be public?

Who knows. If it goes public, then it will likely be added to this repo as well.

A hint - after AES or other mathematically complex encryption, you do not get repeating patterns like this:

[image]

493505110, 2025-08-08:
> > Will 2nd level FC firmware be public?
> 
> Who knows. If it goes public, then it will likely be added to this repo as well.
> 
> A hint - after AES or other mathematically complex encryption, you do not get repeating patterns like this:
> 
> <img alt="Image" width="927" height="975" src="https://private-user-images.githubusercontent.com/11463396/475940887-7c1d1d4a-364c-4d20-a522-bb4b912cd882.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQ2NDU2MDQsIm5iZiI6MTc1NDY0NTMwNCwicGF0aCI6Ii8xMTQ2MzM5Ni80NzU5NDA4ODctN2MxZDFkNGEtMzY0Yy00ZDIwLWE1MjItYmI0YjkxMmNkODgyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODA4VDA5MjgyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIxNjdjOTNhOGI5OGIyNjM1ZmQyYTM5YzFlYjFjY2YzNWY2NDFkMTYyMWNhYTdkOWZlNjI4MTdlYTk5YzM5NDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.vv12YoHA1t98K8xLbcVeZMovqV84iHSxODqLfOmcusA">

Thank you,but i dont know the key

493505110, 2025-08-09:
> Who knows. If it goes public, then it will likely be added to this repo as well.
>
> A hint - after AES or other mathematically complex encryption, you do not get repeating patterns like this:
>
> <img [...]

btw,i got these by wm161_0100's squashfs
is it useful?

[image]

mefistotelis, 2025-08-09:
> it seems success decode some string?

No, you would have to write a separate decoder for this format.
All it found is the value 8 which happened to be in position of module type.

```
# grep 'video encoder' *.py
dji_xv4_fwcon.py:    DjiModuleTarget( 8,-1, "VENC",    "video encoder in air"),
dji_xv4_fwcon.py:    DjiModuleTarget( 8, 0, "DM368",   "video encoder 'DaVinci Dm368 Linux'"), # P3X
dji_xv4_fwcon.py:    DjiModuleTarget( 8, 1, "IG810LB2","video encoder 'IG810 LB2_ENC'"),
```

The 'xV4' is often used as magic by DJI, because it's just incrementing digits in hex. I guess they use that as a placeholder but then are to lazy to replace with something meaningful.

For the factory test scripts - yes, checking them is useful for gaining more knowledge on how the AC works. In particular, the diagnostic packets generated with `dji_mb_ctrl`.