o-gs/dji-firmware-tools #190: The 2018 format is not fully supported - Mini 2

9930wilson, 2021-01-19:
I have the auth_key, enc_key & scram_key for mini 2.. need to decrypt the scram_key and make changes to this script.. So that it can decrypt the 2018+ firmware modules..
Can anyone help??

python3 dji_imah_fwsig.py -vv -u -i 0905v2.sig
0905v2.sig: Opening for extraction and un-signing
0905v2.sig: Warning: The 2018 format is not fully supported.
0905v2.sig: Warning: Header field 'reserved2' is non-zero; the tool is not designed to handle this.
0905v2.sig: Warning: Header field 'reserved3' is non-zero; the tool is not designed to handle this.

mefistotelis, 2021-01-19:
For a first try, the tool itself can '--help' you. There is a force option, which continues regardless of unrecognized fields.

Mavic Mini 2 firmware package uses UFIE key for encryption and PRAK key for authentication. If you have these keys, consider sharing. Such things are typically discussed on our Slack group, dji-rev.slack.com.

9930wilson, 2021-01-19:
The log was too long.. So shared just the starting few lines.
Please refer the attachment..  I still have to understand how to use to use the enc_key to decrypt the scram_key..

My background is of mechanical and aircraft engineering. I don't know much about coding and software.. So I'm stuck with the script.

I tried to join the dji-rev.slack.com but the old invites don't work anymore.. And I don't know anyone who can let me in.

[zIMAH mini2.docx](https://github.com/o-gs/dji-firmware-tools/files/5835551/zIMAH.mini2.docx)

danieltroger, 2021-01-19:
have you tried https://join.slack.com/t/dji-rev/shared_invite/zt-b1dubybk-K9M~z7zS8h_8f97F9LbYug

mefistotelis, 2021-01-19:
@Bin4ry is keeping a working invite here:
https://github.com/Bin4ry/deejayeye-modder#want-to-contribute-or-learn-some-stuff

Let me know if it doesn't work and I will ping him.

Note that to decrypt the firmware, you need to have the actual keys. Some keys are included in the public script - view its content to see them. Some more keys are known but not public, and for some drones - we just don't have them. Dji gave 4-letter identifiers to the keys, though the identifiers are often re-used (ie. there are multiple different keys used as PUEK key, only 2 of these are public).

The public script doesn't support new version of firmware container, because adding the support would change nothing - without the keys, it would be still impossible to decrypt the firmware. Though if you're interested in making that support - patches are welcome.

9930wilson, 2021-01-19:
I saw the .py script and the keys.. My bad.
I thought since the "PRAK" auth_key is already known for the previous version.. those are the same keys and we can use for the M2 firmware.

The second part of the attachment seems like alien language to me.. So check if it has some new info which may help you guys to understand how to find the keys. I have no idea what the writer means by uint8_t and uint32_t..

[image]

mefistotelis, 2021-01-19:
> So check if it has some new info which may help you guys to understand how to find the keys.

No. All that is defined in the public script as well.

> I have no idea what the writer means by uint8_t and uint32_t..

https://www.gnu.org/software/libc/manual/html_node/Integers.html

LeandroBrasill, 2021-01-24:
for god's sake someone help me please do this update on Dji mini 2 please

mefistotelis, 2021-05-13:
The 218 format is now supported. Closing.

closed 2021-05-13 by mefistotelis