o-gs/dji-firmware-tools #191: Dji mini 2 código alguém já sabem como

LeandroBrasill, 2021-01-21:
Alguém já encontrou o código do dji mini 2 "WM???" Alguém saber qual e o código????????????

mefistotelis, 2021-01-21:
Sianie wszędzie wiadomosciami w twoim lokalnym języku nie jest dobrym pomysłem, jeżeli masz nadzieję że ktoś to przeczyta i jeszcze odpowie.

closed 2021-01-21 by mefistotelis

LeandroBrasill, 2021-01-21:
> Sianie wszędzie wiadomosciami w twoim lokalnym języku nie jest dobrym pomysłem, jeżeli masz nadzieję że ktoś para przeczyta i jeszcze odpowie.

friend help me please do you know tell me the update code for dji mini 2?

mefistotelis, 2021-01-21:
"update code"?

Anyway, this is an open-source project which does accept contributions.
You need something, you implement it. Then you share with everyone else.

LeandroBrasill, 2021-01-21:
> "atualizar código"?
> 
> De qualquer forma, este é um projeto de código aberto que aceita contribuições.
> Você precisa de algo, você o implementa. Então você compartilha com todos os outros.
because the Mavic mini and WM160 code is correct, what would be the dji mini 2 code?

mefistotelis, 2021-01-21:
The best way to figure out the WM model marking of a drone is to look into its firmware files.
If you have the drone, and updated the FW already - the update file should be cached somewhere on your mobile device.
If not, download FW update from the internet.

LeandroBrasill, 2021-01-21:
> A melhor maneira de descobrir a marcação do modelo WM de um drone é examinar seus arquivos de firmware.
> Se você tem o drone e já atualizou o FW - o arquivo de atualização deve ser armazenado em cache em algum lugar do seu dispositivo móvel.
> Caso contrário, baixe a atualização do FW da Internet.

if I put the code WM160 that would be the Mavic mini code would work on the Dji mini 2?

LeandroBrasill, 2021-01-21:
I saw here that probably the code WM161 and for the Dji mini 2 and I verify that?