o-gs/dji-firmware-tools #194: DJI Phantom 3 Professional Ambarella

rsrmdp, 2021-02-19:
Good I tried to install the Firmware m0100 by USB UART, without success:

using the methods of:
https://github.com/o-gs/dji-firmware-tools/wiki/Flashing-firmware-via-SD-card#by-firmware-module-in-camera
https://github.com/o-gs/dji-firmware-tools/wiki/Firmware-m0100

Extracting: P3X_FW_V01.11.0020_m0100.bin and renaming it to PFC300XFw3.bin

I tried it with several P3X_FW, with no positive results:

Also probe: Program m0101 the same way as you did with m0100: extract modules, but instead of P3S_FW _ * - m0100.bin use them P3S_FW _ * - m0101.bin. Rename Me PFC300SFw3.bin. As you put mefistotelis at https://github.com/o-gs/dji-firmware-tools/issues/172#issuecomment-734972633 with no result.

Luckily the led of the gimbal keeps flashing, which gives me hope that the plate still gives me signals, since I have another that does not turn on the led and does not show me data (it is a simple deduction)

LEAVE SOME LOGS:

P3X_FW_LOG_AB: LOG

[00015162] ========== remo-con disconnect. boot (15) ============
 [00015294] Packet [C: \ P3X_FW_V01.11.0020.bin] detected, card sn [0x9b54d1c4].
 [00015329] Packet upgrade start ...

 [00015364] Packet checking ...
 [00015399] Packet vlink 01.11.0020 <-> 01.11.0020.
 [00015469] Done.

 [00015500] Version checking [1] ...
 [00015578] [03 06] [00] v2.4.10.7 -> v2.4.20.50 need upgrade.
 [00015658] [03 05] [00] v34.2.0.9 -> v34.2.0.9
 [00015698] [04 00] [00] v1.41.0.0 -> v1.48.0.0 need upgrade.
 [00015818] [11 00] [00] v1.8.0.0 -> v1.8.0.0
 [00015938] [11 01] [00] v1.8.0.0 -> v2.0.0.33, firmware v1.8.0.0 not support.
 [00015979] [11 01] [00] v1.8.0.0 -> v2.0.0.33
 [00016078] [12 00] [00] v1.10.0.0 -> v1.12.0.0 need upgrade.
 [00016178] [12 01] [00] v1.10.0.0 -> v1.12.0.0 need upgrade.
 [00016278] [12 02] [00] v1.10.0.0 -> v1.12.0.0 need upgrade.
 [00016379] [12 03] [00] v1.10.0.0 -> v1.12.0.0 need upgrade.
 [00028913] [15 00] [00] device not detected.
 [00029039] [17 00] [00] v1.1.1.7 -> v1.1.1.7
 [00029179] [17 01] [00] v1.0.2.7 -> v1.0.2.7
 [00041714] [19 00] [00] device not detected.
 [00041752] [01 00] [00] v1.26.4315 -> v1.32.5432 need upgrade
 [00041787] [01 01] [00] v1.26.4315 -> v1.32.5432 need upgrade
 [00054333] [08 00] [00] device not detected.
 [00054381] [09 00] [00] v0.0.0.0 -> v4.1.0.0 need upgrade.
 [00054415] Done.

 [00056449] Waiting for user confirm ...
 [00066482] Timeout, start upgrade automatically.

 [00066556] Firmware upgrading [1] ...
 [00078091] [01 00] Firmware upgrade start ...
 [00015133] [01 00] Firmware upgrade finished successfully.
 [00015173] Done.

 [00015207] Packet [C: \ P3X_FW_V01.11.0020.bin] resumed, card sn [0x9b54d1c4].
 [00015243] Packet vlink 01.11.0020 <-> 01.11.0020.
 [00015306] Version checking [2] ...
 [00015392] [03 06] [00] v2.4.10.7 -> v2.4.20.50 need upgrade.
 [00015472] [03 05] [00] v34.2.0.9 -> v34.2.0.9
 [00015517] [04 00] [00] v1.41.0.0 -> v1.48.0.0 need upgrade.
 [00015692] [11 00] [00] v1.8.0.0 -> v1.8.0.0
 [00015811] [11 01] [00] v1.8.0.0 -> v2.0.0.33, firmware v1.8.0.0 not support.
 [00015848] [11 01] [00] v1.8.0.0 -> v2.0.0.33
 [00016051] [12 00] [00] v1.10.0.0 -> v1.12.0.0 need upgrade.
 [00016153] [12 01] [00] v1.10.0.0 -> v1.12.0.0 need upgrade.
 [00016251] [12 02] [00] v1.10.0.0 -> v1.12.0.0 need upgrade.
 [00016351] [12 03] [00] v1.10.0.0 -> v1.12.0.0 need upgrade.
 [00028889] [15 00] [00] device not detected.
 [00028971] [17 00] [00] v1.1.1.7 -> v1.1.1.7
[... 1213 lines cut ...]
[0m [00002725] [CA9_0] ------- DjiAaaAeMeteringModeSetCallback ------- start [0m
[0m [00002725] [CA9_0] ------- DjiAaaAeMeteringModeSet ------- start [0m
[0m [00002725] [CA9_0] ------- DjiAaaAeMeteringModeSet ------- end [0m
[0m [00002725] [CA9_0] ------- DjiAaaAeMeteringModeSetCallback ------- end [0m
[0; 33m [00002725] [CA9_0] [DJI_UST] DjiUstAeMeterModeSet: End setting, id = 6. [0m
[0; 33m [00002725] [CA9_0] [DJI_UST] DjiUstEvBiasSet: Setting value (9). [0m
[0; 33m [00002725] [CA9_0] [DJI_UST] DjiUstEvBiasSet: Start setting, id = 8. [0m
[0m [00002725] [CA9_0] ------- DjiAaaAeEvBiasSetCallback ------- start [0m
[0m [00002725] [CA9_0] ------- DjiAaaAeEvBiasSet ------- start [0m
[0m [00002725] [CA9_0] evbias (data = 9) [0m
[0m [00002725] [CA9_0] ------- DjiAaaAeEvBiasSet ------- end [0m
[0m [00002725] [CA9_0] ------- DjiAaaAeEvBiasSetCallback ------- end [0m
[0; 33m [00002725] [CA9_0] [DJI_UST] DjiUstEvBiasSet: End setting, id = 8. [0m
[0; 33m [00002725] [CA9_0] [DJI_UST] DjiUstModeDialSetCallBack: End setting, id = 14. [0m
[0; 33m [00002725] [CA9_0] [DJI_UST] DjiUstModeDialSet: End setting, id = 33. [0m
[0; 33m [00002725] [CA9_0] [DJI_UST] DjiUstPhotoJpegQualitySet: Setting value (2). [0m
[0; 33m [00002725] [CA9_0] [DJI_UST] DjiUstPhotoSizeSet: Setting value (4). [0m
[0; 33m [00002725] [CA9_0] [DJI_UST] DjiUstPhotoSizeSet: Start setting, id = 2. [0m
[0m [00002725] [CA9_0] Capture state = 0x0000 (0x0001) [0m
[0; 33m [00004716] [CA9_0] [DJI_MIS] DjiLog2SdDetectProc: detect log record config file enable sd record log [0m

And I continue with the same thing, I cannot take photos, it detects the SD, but it does not take photos or film, every time I am going to delete the file
PFC300XFw3_updated.bin or firmware to update is always at 0byte
[image]
[image]

mefistotelis, 2021-02-19:
What made you believe this is Ambarella issue?

From the logs, Ambarella seem to be working. I wouldn't tell the same about Davinci.

rsrmdp, 2021-02-19:
Maybe this was wrong, but after a failed update I set Camera Error 01, and could not take photos, or put it in video recording mode. If you had live video in the DJI app