o-gs/dji-firmware-tools #195: DJI EEPROM Vault

sajjsamm, 2021-02-26:
There are various types of flash and EEPROM ICs in DJI products. Initiating a vault with known and healthy dumps of these kinds of storage ICs is a good idea. and in many cases, reflashing may bring dead or bricked devices back to life,

mefistotelis, 2021-02-26:
You mean, like this one?:
https://github.com/cs2000/DankDroneDownloader

sajjsamm, 2021-02-26:
something like that with more specific details for hardware engineers who want to interact directly with the hardware in case of bricked or corrupted system files and not only the drone itself but including the remote controllers.

mefistotelis, 2021-02-26:
So more like this description I made?:
https://github.com/o-gs/dji-firmware-tools/wiki/Flashing-firmware-on-DaVinci-media-processors

If you have knowledge about specific topics, you can post the detais here. I can prepare a wiki page from that.

sajjsamm, 2021-02-26:
yes, that's what I'm looking for. I repair electronic logic boards and I may get DJI products on my hands as well from time to time. I can contribute to the vault. I hope it would be useful to someone else who is doing a business like me.

mefistotelis, 2021-02-26:
We gladly accept contributions; you can either ask for a wiki change here is "issues", or clone the wiki repository and add your changes as new commits.

Representatives of few repair shops also lurk on #hardware group on slack. So for knowledge which exists but was never written down, you can ask there. Slack invite:
https://github.com/Bin4ry/deejayeye-modder#want-to-contribute-or-learn-some-stuff

closed 2021-03-01 by sajjsamm