o-gs/dji-firmware-tools #196: Mavic 2 Pro-Zoom Battery

scottWestveer, 2021-03-11:
I have a Mavic 2 Battery that was in storage too long and a cell got under voltage too far and the BMS Chip locked it out.

Opened it up hoping to use the tools to open and clear the PF flag but the BMS Chip is different.  I did charge the cells individually and they all appear to be find now, but the board still will not wake up.

I took some pictures for data to share in hopes of future support for this Battery:
BMS Chip is a TI BQ9003, SCL and SDA lines are easily accessible through the connector, don't even have to open the casing it appears..
[image]

[image]

mefistotelis, 2021-03-11:
DJI uses that chip for some time now, I'm actually working on its support right now. Its firmware claims the chip is BQ40z307, but that name seem to come from DJI and not from TI. The FW is heavily modified by DJI.

Can you take good quality photos of both sides of the board? I don't have it on the project wiki, yet.

scottWestveer, 2021-03-12:
Looking at DJI's site, it appears the BQ9003 is a "NDA" version of the chip that is likely just a modified version of the standard BQ you mention.. 
Here are some originals that hopefully show what you are looking for. If you need something specific.. where a trace goes or something, let me know.
[DJI Mavic 2 Battery.zip](https://github.com/o-gs/dji-firmware-tools/files/6128750/DJI.Mavic.2.Battery.zip)

From TI's site, this is the schematic of the bq40z50EVM Li-Ion Battery Pack Manager Evaluation Module which appears to be close for the circuitry:
[image]

I'm going to get out my old R-Pi, I think its a B+, and get it loaded up to try the tools. Are you using Raspbian or some other OS?

mefistotelis, 2021-03-12:
Thanks.

For the OS - any Linux will do, as long as the I2C kernel module is available. You also need Python 3.5 or newer.

The chip has some differences to BQ40z50, but I haven't found a closer match yet.
For example,  ChargingStatus bits do not contain temperature range bits (undertemp. / std temp / high temp etc.). These are probably somewhere in another register.

alximiktik, 2021-03-12:
> DJI uses that chip for some time now, I'm actually working on its support right now. Its firmware claims the chip is BQ40z307, but that name seem to come from DJI and not from TI. The FW is heavily [...]

if you need photo PCBA of any drone tell me. I will sent to you. My friend are repairing drones, i can ask him take photo

mefistotelis, 2021-03-14:
@alximiktik I am interested in photos of any board which is not on my wiki. Even for those which are on the wiki - if some chip markings are not readable, I'm still interested in more photos.

For specific boards, as I'm working on batteries right now - I don't have battery boards for Mavic Air 1 & 2, Spark, Mavic Mini 1 & 2, Inspire 2.

If you choose to help me with the photos, please make sure to shoot both sides, and do it in a way which makes chip markings visible.

tecdrone, 2021-03-22:
Olá amigo falo do Brasil .tbm tenho uma Assistência técnico especializa em drone dji .tenho a ev2300 e Ubrt para baterias .se precisa de placas tbm posso fazer fotos para vc

tecdrone, 2021-03-22:
> @alximiktik Estou interessado nas fotos de qualquer fórum que não esteja na minha wiki. Mesmo para aqueles que estão no wiki - se algumas marcas de chip não são legíveis, ainda estou interessado em mais fotos.
> 
> Para placas específicas, como estou trabalhando com baterias agora - não tenho placas de bateria para Mavic Air 1 e 2, Spark, Mavic Mini 1 e 2, Inspire 2.
> 
> Se você decidir me ajudar com as fotos, certifique-se de tirar fotos dos dois lados e faça-o de uma forma que torne as marcas visíveis.

Eu tenho placas bateria mavic air 1, Spark mavic 2 pro Phantom 4

mixeysan, 2021-04-10:
> @alximiktik I am interested in photos of any board which is not on my wiki. Even for those which are on the wiki - if some chip markings are not readable, I'm still interested in more photos.
>
> For [...]

Hello my friend :) Do you need a photo of mavic 2 pro motherboard with ISP eMMC pinouts? I was interested and I bought the motherboard of the drowned drone, drew the ISP eMMC pinout and read the data from it, and also restored photos and videos from the internal memory.

tecdrone, 2021-06-15:
> Tenho uma bateria Mavic 2 que ficou armazenada por muito tempo e uma célula ficou sob tensão demais e o chip BMS a bloqueou.
> 
> Abri na esperança de usar as ferramentas para abrir e limpar a bandeira PF, mas o Chip BMS é diferente. Eu carreguei as células individualmente e todas parecem estar bem agora, mas a placa ainda não acorda.
> 
> Tirei algumas fotos para compartilhar dados na esperança de um suporte futuro para esta bateria: O
> Chip BMS é um TI BQ9003, as linhas SCL e SDA são facilmente acessíveis através do conector, nem é preciso abrir a caixa que aparece ..
> [image]
> 
> [image]
> 
> [image]

Amigo você pode remoVer PF pelo programa UBRT uso sempre . Meu whatsapp +55 8198734-0010

mixeysan, 2021-06-15:
Use cp2112 and simple soft DJI battery killer [image] 
http://master-club.org/drone/DJi_Battery_Killer_with_dll_13.06.2021.7z

radenrezky, 2021-07-01:
> U can use cp2112 adapter and simple soft DJI battery killer [image]
> http://master-club.org/drone/DJi_Battery_Killer_with_dll_13.06.2021.7z

can you share the link.. i need this for my Mavic 2 Pro battery (unseal + clear PF + seal)

mixeysan, 2021-07-23:
http://master-club.org/drone/DJi_Battery_Killer_with_dll_26.06.2021.7z

mjnhchj, 2021-07-31:
I love this tool. 😍😍😍

mjnhchj, 2021-08-08:
Cell error.  How to fix it? DJI battery killer not fix this error 😣😣😣

mefistotelis, 2021-08-08:
Taking down PF flag does not 'fix' anything.
It's just a way to tell the battery that YOU fixed it, and it should no longer be treated as damaged.

mjnhchj, 2021-08-08:
> Taking down PF flag does not 'fix' anything.
> It's just a way to tell the battery that YOU fixed it, and it should no longer be treated as damaged.

Can i use Comm_sbs_bqctrl.py from DJI Firmware tools to fix this error?

mixeysan, 2021-08-09:
If the BMS board didn't after water and you connected other and good cells and also a drone writes with them that there is an error in the cells, then check the 200 Ohm resistors through which each cell is connected to the controller. Also check the controller soldering and it is advisable to solder the controller.

mjnhchj, 2021-08-09:
I checked the battery voltage in DJI Fly app. Both cell are balance :((

mixeysan, 2021-08-09:
Can U show logs from drones with show voltage for each cell?

mjnhchj, 2021-08-09:
How do I extract that data?  original battery cell i disassembled and assembled, error appeared!

maxdd, 2021-09-14:
> http://master-club.org/drone/DJi_Battery_Killer_with_dll_26.06.2021.7z
> 
> http, not https server[image]

Doest this work with BQ30Z55?

mixeysan, 2021-09-14:
> > http://master-club.org/drone/DJi_Battery_Killer_with_dll_26.06.2021.7z
> > 
> > http, not https server[image]
> 
> Doest this work with BQ30Z55?

New version work with bq30z55, but not free yet. Later.

maxdd, 2021-09-15:
Hi mixeysan, thank you for the feedback.
Will it work with CP2112 device?

mixeysan, 2021-09-15:
> Hi mixeysan, thank you for the feedback.
> Will it work with CP2112 device?

it already works with cp2112[image]

maxdd, 2021-09-15:
That's amazing! If you need a tester let me know.
Are you using external connector for SDA/SCL or are you soldering on the battery?

mixeysan, 2021-09-15:
> That's amazing! If you need a tester let me know.
> Are you using external connector for SDA/SCL or are you soldering on the battery?

External connection is UART, or am I missing something?
I need information on how to unseal the chip  or how to change the key without knowing this key, that is, when the key is not the default 0123456789ABCDEFFEDCBA9876543210 and when it is initially unknown.

maxdd, 2021-09-15:
Oh didnt know that, so soldering is a must for connecting to i2c right?
I can do the soldering and help you out if you want

noname443, 2021-09-19:
hello everyone, how good there are enthusiasts who can embrace almost anything. this topic helped me check out some AIR 1 batteries that I managed to buy some time ago. However, I have a problem with sealing. The program presented here does not have this option, but it is in the last screenshot. What is this version, because visually the interface looks identical, it is probably a newer development version. There is a chance to get access to this version? I have one more question. Has anyone tried to replace cels ( mavic AIR 1) with more capacity, but not 18650. I am asking because
I noticed repeated damage that the last cell in blocked batteries has reversed polarity and cannot be regenerated, charged. Thanks in advance for any suggestions.

maxdd, 2021-09-20:
Is there a reason why you cant use python scripts directy? mixeysan is still developing the app

mixeysan, 2021-09-20:
Hi guys. I'm working on the program yet. I think in a week or two I'll post a new version of the program, and there will be a lot of interesting things, I promise :)

radenrezky, 2021-09-22:
> 
> 
> Hi guys. I'm working on the program yet. I think in a week or two I'll post a new version of the program, and there will be a lot of interesting things :)
> But for now there will be only bq30z55 and bq9003.

Hi mixeysan, i bought the CP2112 and managed to connect to my Mavic2Pro intelligent battery using your software. may i  continue the process? before i put unseal, replace cell, clear PF and back seal again, i want to ask to you to makesure this apps can work and no bug then put my bq9003 (bq40z307). or i should wait you new release :) thanks a lot my friend!!

tungtranmobile, 2021-10-04:
> Hi guys. I'm working on the program yet. I think in a week or two I'll post a new version of the program, and there will be a lot of interesting things :) But for now there will be only bq30z55 and bq9003.

hello AD

tungtranmobile, 2021-10-04:
> Hi guys. I'm working on the program yet. I think in a week or two I'll post a new version of the program, and there will be a lot of interesting things :) But for now there will be only bq30z55 and bq9003.

Hi ad. you can get link to my email please? tugninhkieu@gmail.com. thanks

NHAMTAMPHUOCTHIEN, 2021-10-04:
> http://master-club.org/drone/DJi_Battery_Killer_with_dll_26.06.2021.7z
> 
> http, not https server[image]

Hi friend!
you can get link of software to my email please? huynhphuocnhan1907@gmail.com
 thanks you!

tungtranmobile, 2021-10-04:
Cảm ơn b. Sao mình kết nối ok mà read info ko đc nhỉGửi từ Galaxy của tôi
-------- Tin nhắn gốc --------Từ: NHAMTAMPHUOCTHIEN ***@***.***> Ngày: 04/10/2021  17:39  (GMT+07:00) Đến: o-gs/dji-firmware-tools ***@***.***> Cc: tungtranmobile ***@***.***>, Comment ***@***.***> Chủ đề: Re: [o-gs/dji-firmware-tools] Mavic 2 Pro-Zoom Battery (#196) 

http://master-club.org/drone/DJi_Battery_Killer_with_dll_26.06.2021.7z
http, not https server

Hi friend!
you can get link of software to my email please? ***@***.***
thanks you!

mixeysan, 2021-10-04:
U can find the last version here - https://github.com/o-gs/dji-firmware-tools/issues/214#issuecomment-927307835

tungtranmobile, 2021-10-04:
> U can find the last version here - #214 (comment)

cho mình sđt add zalo đc ko b

mjnhchj, 2021-10-04:
Đến lạy các cụ. Viết tiếng Việt rồi bắt người ta dịch sang tiếng anh à? Nước ngoài làm gì có xài zalo mà kết bạn?????

kingjc218, 2021-10-20:
> U can find the last version here - [#214 (comment)](https://github.com/o-gs/dji-firmware-tools/issues/214#issuecomment-927307835)

HI mixeysan,I have some question to ask you ,can you send your  email to me ,my email is tonyes@126.com

Hobbyleon, 2021-11-10:
Hello, I have 3 mavic air 1 batteries and the bq9003 chip is blocked, I was reviewing this forum and I am interested in the system to restore the battery, I would appreciate it, I would like to know how I access the chip (how do I connect it) I am a drone technician in Mexico. Cheers

0roblequad, 2021-11-10:
tienes que descargar el progama de 
mixeysan
 y comprar el reseteador de chip en Amazon

mjnhchj, 2021-11-10:
https://youtu.be/RkdUHQI2U_8

tadwood, 2021-11-29:
> @mixeysan First of all congratulations for your job. I'm trying to revive 2 Mavic air 1 batteries. I sucesfully connected your software to a working battery, but I can't do the same on the dead ones (by a long period of innactivity). One of them is open (plastic shell) and I used a Skyrc Imax B6 charger to partially charge it and balance the cells, the other one is intact. In both cases, I get the same errors: ERROR: Could not write Block to 0x44 ERROR: Could not perform SMBus read 0x00 Any ideas?

Did you ever get this to work for the dead batteries? I'm trying this on a Spark battery and getting the same exact errors.

mixeysan, 2021-11-29:
Guys, I'm sorry, but this topic is mephistotelis and I have no right to write in its topic. You can write to me in private messages or in telegrams.

davidhaluska, 2021-12-01:
Hi All, Is there possible to put battery into sleep mode - as when it comes from factory? There should be some command for that or? Thanks. David

mefistotelis, 2021-12-01:
> Is there possible to put battery into sleep mode - as when it comes from factory?

Yes there is. Read BMS chip documentation for details.
Such battery wakes up when connected to charger - so it will stop responding to commands until you do so.

cruzroy, 2021-12-10:
does anyone know how to setup bigger lipos with dji battery killer, change chemistry and or MAH ...thanks any help would be great

cruzroy, 2021-12-13:
how do you load battery killer so it works,,,,i cant get to work right

WilliansBR, 2021-12-21:
Nice Job!  I will try it on Mavic PRO 1. I lost 4 batterys

mixeysan, 2021-12-21:
sorry guys, but I am not allowed to use the mephistotelis forum for support the program. if you have any questions, write to me in telegram, my contacts are in the program

dronythinggoes, 2022-01-02:
> sorry guys, but I am not allowed to use the mephistotelis forum for support the program. if you have any questions, write to me in telegram, my contacts are in the program

hey bro what's up can you please send me dji battery killer intaller at stotomasromeo@gmail.com

Alessandro2023, 2022-01-16:
> Oi pessoal. Estou trabalhando no programa ainda. Acho que daqui a uma ou duas semanas vou postar uma nova versão do programa, e vai ter muita coisa interessante :) Mas por enquanto só vai ter bq30z55 e bq9003.

> Oi pessoal. Estou trabalhando no programa ainda. Acho que daqui a uma ou duas semanas vou postar uma nova versão do programa, e vai ter muita coisa interessante :) Mas por enquanto só vai ter bq30z55 e bq9003.

Boa tarde amigo gostaria de saber se vc conseguiu finalizar o dji battery killer 0.4 beta 2 se já conseguiu terminar vc poderia me passar o link pra mim baixar ele por favor ale3dprint@hotmail.com

dronythinggoes, 2022-01-16:
I am sorry i cannot understand, can you translate it in english?

Alessandro2023, 2022-01-16:
> Oi pessoal. Estou trabalhando no programa ainda. Acho que daqui a uma ou duas semanas vou postar uma nova versão do programa, e vai ter muita coisa interessante :) Mas por enquanto só vai ter bq30z55 e bq9003.

Good afternoon friend I would like to know if you managed to finish the dji battery killer 0.4 beta 2 if you were able to finish it could you give me the link to download it in my email please ale3dprint@hotmail.com

dronythinggoes, 2022-01-17:
Hello my friend, i have this firmware that i successfully used in my mavic 2 battery 
http://master-club.org/drone/DJi_Battery_Killer_with_dll_26.06.2021.7z

Alessandro2023, 2022-01-17:
> Oi pessoal. Estou trabalhando no programa ainda. Acho que daqui a uma ou duas semanas vou postar uma nova versão do programa, e vai ter muita coisa interessante :) Mas por enquanto só vai ter bq30z55 e bq9003.

sorry friend I wanted the dji battery killer 0.4 beta 2 that I can unlock and write on the bq30z55 Mixeysan has already launched I've seen a video of a person using this program

Alessandro2023, 2022-01-17:
I want to unlock and clean the PF of my phantom 4 battery can someone help me I have the cp2112 here

dronythinggoes, 2022-01-17:
I don't have it yet my friend i only use the veraion that i sent you

KnusperKnusper, 2022-01-20:
CP2112 and Mavic air 1 instrucion with ERROR 0X00:

Open Battry, Connect Charger to battery, Bridge PIN +13V to the Side of the battery where +13V would be or charge every cell on its own, connect SDL SCA and GND. 
If error appears switch sda scl.

Connect in the software, unseal etc...

dronythinggoes, 2022-01-21:
Thanks brother, but i have same error issue with mavic 2 battery, should i do the same? Do you mean i will use jumper wire from charger terminals to the bms board?

[47 more replies]