o-gs/dji-firmware-tools #199: Mavic Air 2 gimbal calibration error 40011

aytacengin, 2021-03-24:
Dou you have any solution ?  Downgrade firmware is not working for this solution.

dronesmedellin, 2021-04-16:
[](https://www.youtube.com/watch?v=5z8QmE1i2PM&ab_channel=InfinityDrones)

Haven´t tried it yet

dronesmedellin, 2021-04-16:
https://www.youtube.com/watch?v=5z8QmE1i2PM&ab_channel=InfinityDrones

aytacengin, 2021-04-27:
I know this solution. This is for Mavic Mini. I need a solution for Mavic Air2

robgode, 2021-06-29:
Same problem here, tried the Mini solution but no success

lrgc99, 2021-07-08:
Both for mini 2 and for Air 2 it is no longer possible to download firmware to eliminate error 40011.
No solution?

joao-grigorio, 2021-10-19:
> Você tem alguma solução? O downgrade do firmware não está funcionando para esta solução.

tenho a solução me chama por email contato.johnrpas@gmail.com e me mostra a tela do aplicativo com o erro.

weijieng, 2021-11-03:
I fixed 40011 error. If you want a solution. I can hep you. I can connect with ANYDESK application to your computer. We will fix together. You can contact me direclty.

@aytacengin hi can you help me

ginnolo, 2021-11-06:
> Ho corretto l'errore 40011. Se vuoi una soluzione. posso aiutarti. Posso connettermi con l'applicazione ANYDESK al tuo computer. Ripareremo insieme. Puoi contattarmi direttamente.

mi puoi aiutare? info@giannicenturione.com

954426470, 2021-11-18:
I will, you can contact me

954426470, 2021-11-18:
> 
> 
> > I fixed 40011 error. If you want a solution. I can hep you. I can connect with ANYDESK application to your computer. We will fix together. You can contact me direclty.
> 
> Hi @aytacengin may you help me

I will, you can contact me

954426470, 2021-11-18:
> 
> 
> > Ho corretto l'errore 40011. Se vuoi una soluzione. posso aiutarti. Posso connettermi con l'applicazione ANYDESK al tuo computer. Ripareremo insieme. Puoi contattarmi direttamente.
> 
> mi puoi aiutare? [info@giannicenturione.com](mailto:info@giannicenturione.com)

I will, you can contact me

weijieng, 2021-11-18:
is that any fee i need to pay?

954426470, 2021-11-18:
> is that any fee i need to pay?
> 
> 
> 
> > Ho corretto l'errore 40011. Se vuoi una soluzione. posso aiutarti. Posso
> > connettermi con l'applicazione ANYDESK al tuo computer. Ripareremo insieme.
> > Puoi contattarmi direttamente.
> >
> > mi puoi aiutare? ***@***.***
> >
> > I will, you can contact me
> >
> > or Android

Yes

Kakoroko, 2021-11-18:
Same isuss here need help how I contact you I have 3 errors

Kakoroko, 2021-11-18:
[image]

954426470, 2021-11-18:
My email address is 954426470 @ qq. com

Kakoroko, 2021-11-18:
Kakoroko1992@gmail.com

Kakoroko, 2021-11-18:
Hi can you help me

Sent from Mail.ru app for iOS

954426470, 2021-11-18:
I can help you. You need a remote software and a connection to the Chinese network 

 

 
 
 
 
 
 &gt;My email address is 954426470 @ qq. com&amp;nbsp;
 &gt;
 &gt;
 &gt;
 &gt;---Original---
 &gt;From: ***@***.***&amp;gt;
 &gt;To: ***@***.***&amp;gt;;
 &gt;Cc: ***@***.******@***.***&amp;gt;;
 &gt;
 &gt;
 &gt;
 &gt;
 &gt;
 &gt;Same isuss here need help how I contact you I have 3 errors
 &gt;
 &gt;—
 &gt;—

yaj29, 2021-11-19:
@aytacengin how?

alives2000, 2022-03-28:
> I fixed 40011 error. If you want a solution. I can hep you. I can connect with ANYDESK application to your computer. We will fix together. You can contact me direclty.
> 
> https://www.facebook.com/aytacengincom/

hi, how can i contact you?

aytacengin, 2022-03-28:
Dear Friends. I wrote upper comment.. Contact from my facebook link. It is very simple.

alives2000, 2022-03-29:
> Dear Friends. I wrote upper comment.. Contact from my facebook link. It is very simple.

I wrote you a private message on facebook but have not received a response so far

954426470, 2022-03-29:
I can handle it. You need to add my email address

954426470, 2022-03-29:
I don't have Facebook 。 Do you have any other way

alives2000, 2022-03-29:
> Dear Friends. I wrote upper comment.. Contact from my facebook link. It is very simple.

@aytacengin I wrote you a private message on facebook but have not received a response so far

ja4who, 2022-06-25:
Olá, alguém conseguiu resolve o erro 40011 ? Se alguém tiver solução, podemos ver algum valor significativo pelo auxílio.

yaj29, 2022-10-11:
how much?

yaj29, 2022-10-14:
> сколько?
> You can write me on WhatsApp +62 812-8430-9143

yes the ***@***.*** not working log in account

alxgacc, 2022-10-14:
@yaj29 I can remove errors 40011 and 40021 remotely, who needs this write me on whatsapp +62 812-8430-9143

yaj29, 2022-10-14:
How much?

954426470, 2022-10-22:
100rmb

954426470
***@***.***

954426470, 2022-10-26:
If you buy technology, you need $50

954426470
***@***.***

954426470, 2022-10-26:
Please give me your whatapps QR code

954426470
***@***.***

lrgc99, 2022-11-18:
> If you buy technology, you need $50 954426470 ***@***.*** &nbsp;
I'm interested: +39 3485722369

aytacengin, 2022-11-18:
> I fixed 40011 error. If you want a solution. I can hep you. I can connect with ANYDESK application to your computer. We will fix together. You can contact me direclty.
> 
> @aytacengin hi can you help me

yes sure.

ThedeonezoneIndia, 2023-02-25:
> I will, you can contact me

I am intrested in buying software how can i contact you . My email is rajivkumar5405@gmail.com
My whatsapp +91-9017890890