o-gs/dji-firmware-tools #200: DJI Phantom 3 Pro Unknown RC,AC Firmware version beeping (slow DD-DD) can't connect via camera view and can't bind RC to AC

JGabT, 2021-04-12:
Tried flashing P3X_FW_V01.11.0020.bin with FW_DEBUG but failed (steady long beep and red camera gimbal status)
Anything i should follow?

[P3X_FW_RESULT_AB.txt](https://github.com/o-gs/dji-firmware-tools/files/6295050/P3X_FW_RESULT_AB.txt)
[P3X_FW_LOG_AB.txt](https://github.com/o-gs/dji-firmware-tools/files/6295051/P3X_FW_LOG_AB.txt)

mefistotelis, 2021-04-12:
Read about modules:
https://github.com/o-gs/dji-firmware-tools/wiki#introduction

Check which modules were unresponsive and read the pages about them.

JGabT, 2021-04-12:
How do i determine what module is what?

mefistotelis, 2021-04-12:
From the list on the wiki I just shared?
https://github.com/o-gs/dji-firmware-tools/wiki/DJI-Firmwares

JGabT, 2021-04-12:
What do i do with that? (forgive me, im new to this stuff.)

JGabT, 2021-04-12:
Do i flash the P3X supported module to the RC/AC?

mefistotelis, 2021-04-12:
Do with stuff whatever you want. We won't judge.

You have LOG_AB. Skim through it. Notice new chunks are added at end. Look at latest chunk. Read. Think. Correlate to list of modules.

JGabT, 2021-04-15:
on flashing a firmware module, how do i download one? (https://github.com/o-gs/dji-firmware-tools/wiki/Firmware-m0306)

mefistotelis, 2021-04-15:
DankDroneDownloader

JGabT, 2021-04-15:
then download the firmware version specific to that module and unencrypt using dji-firmware-tools?

mefistotelis, 2021-04-15:
exactly.

JGabT, 2021-04-15:
yes i saw that [0ex000] [00 00] i know the [00 00] part is the module number but what is the [0x000] provided for example

mefistotelis, 2021-04-15:
I can't see neither [0ex000] nor [0x000] in the log. The only number loosely matching the patterns you provided is the card s/n:

``
# grep '\[[0-9a-f]*x[0-9a-f]*\]' P3X_FW_LOG_AB.txt
 [00016959]Packet [C:\P3X_FW_V01.01.0008.bin] detected, card sn [0xa0ce694e].
 [00016543]Packet [C:\P3X_FW_V01.01.0008.bin] detected, card sn [0xa0ce694e].
 [00016392]Packet [C:\P3X_FW_V01.11.0020.bin] detected, card sn [0xa0ce694e].
 [00016396]Packet [C:\P3X_FW_V01.10.0090.bin] detected, card sn [0xa0ce694e].
 [00016413]Packet [C:\P3X_FW_V01.11.0020.bin] detected, card sn [0xa0ce694e].
 [00015064]Packet [C:\P3X_FW_V01.11.0020.bin] detected, card sn [0x58c592da].
 [00193084]Packet [C:\P3X_FW_V01.11.0020.bin] detected, card sn [0x58c592da].
 [00014967]Packet [C:\P3X_FW_V01.11.0020.bin] detected, card sn [0x58c592da].
 [00015716]Packet [C:\P3X_FW_V01.11.0020.bin] detected, card sn [0xa0ce694e].
 [00041808]Packet [C:\P3X_FW_V01.11.0020.bin] detected, card sn [0xa0ce694e].
 [00035836]Packet [C:\P3X_FW_V01.11.0020.bin] detected, card sn [0xa0ce694e].
 [00013827]Packet [C:\P3X_FW_V01.11.0020.bin] detected, card sn [0xa0ce694e].
 [00013902]Packet [C:\P3X_FW_V01.11.0030.bin] detected, card sn [0xa0ce694e].
 [00041772]Packet [C:\P3X_FW_V01.11.0030.bin] detected, card sn [0xa0ce694e].
 [00016609]Packet [C:\P3X_FW_V01.10.0090.bin] detected, card sn [0xa0ce694e].
 [00014926]Packet [C:\P3X_FW_V01.11.0020.bin] detected, card sn [0xa0ce694e].
 [00015866]Packet [C:\P3X_FW_V01.06.0040.bin] detected, card sn [0xa0ce694e].
 [00016426]Packet [C:\P3X_FW_V01.06.0040.bin] detected, card sn [0xa0ce694e].
 [00013887]Packet [C:\P3X_FW_V01.01.0008.bin] detected, card sn [0xa0ce694e].
 [00015862]Packet [C:\P3X_FW_V01.11.0020.bin] detected, card sn [0xa0ce694e].
 [00015381]Packet [C:\P3X_FW_V01.11.0020.bin] resumed, card sn [0xa0ce694e].
 [00013897]Packet [C:\P3X_FW_V01.11.0020.bin] detected, card sn [0xa0ce694e].
 [00015377]Packet [C:\P3X_FW_V01.11.0020.bin] resumed, card sn [0xa0ce694e].
```

See CID and PSN within the sd-card protocol specification for more on these.
https://www.sdcard.org/

Also, be precise. Noone likes wasting time.

JGabT, 2021-04-15:
[0x000] [00 00] is an example. what does [0x000] in the log means?

mefistotelis, 2021-04-15:
There is no "x" in timestamp. But that's what you probably looking at.

JGabT, 2021-04-15:
[00017571][11 01][00] v1.8.0.0 -> v2.0.0.33 > oh so [00017571] is a timestamp or 5:57:1?

JGabT, 2021-04-15:
on DankDroneDownloader do i need to rename the file? to P3X_FW_Vnumber.bin?

JGabT, 2021-04-15:
i found that m800 and m1500 is device not detected. and i saw that P3X_FW_V01.09.0060 is going to update both.

JGabT, 2021-04-15:
how do i update the m800 and m1500 i saw that this issue covered it https://github.com/o-gs/dji-firmware-tools/issues/141
but can i do it the simple sd card update?

JGabT, 2021-04-15:
simple way meaning download P3X_FW_V01.09.0060 from DankDroneDownloader unencrypt it using dji-firmware-tools then put on sd card and update?

mefistotelis, 2021-04-15:
If you're really that interested in logging mechanisms within Ambarella, here are the logging macros used:

```
#define DJIUEG_LOG(lvl,fmt,...) \
{ \
  if (AmbaPrintk_Disabled() != 1) \
    AmbaPrintk(1, 1, 1, 0, 0, lvl, "[DJI_UEG] "fmt,args); \
}

#define DJIMSG_LOG(lvl,fmt,...) \
{ \
  if (AmbaPrintk_Disabled() != 1) \
    AmbaPrintk(1, 1, 1, 0, 0, lvl, "[DJI_MSG] "fmt,args); \
}

#define FWUPDATE_LOG(fmt,...) \
{ \
    AmbaKAL_MutexTake(&upgrade_log_mutex); \
    memset(printk_buf, 0, 1024); \
    int ttick = DjiTimerGetTickCount(); \
    sprintf(printk_buf," [%08d]"fmt,ttick,args); \
    DjiUegUpdateLogRecord(0); \
    DjiUegUpdateLogRecord(1); \
    AmbaKAL_MutexGive(&upgrade_log_mutex); \
}

#define FWUPDATE_LOG_RESULT(fmt,...) \
{ \
    AmbaKAL_MutexTake(&upgrade_log_mutex); \
    memset(upgrade_log_buf, 0, 1024); \
    sprintf(upgrade_log_buf, fmt, args); \
    DjiUegUpdateResultRecord(0); \
    DjiUegUpdateResultRecord(1); \
    AmbaKAL_MutexGive(&upgrade_log_mutex); \
}
```

You can't update a dead module by asking another module to contact it. It's dead. Dead don't tend to respond on requests from others.

JGabT, 2021-04-15:
so what do i do to fix it? i just do what #141 do?

JGabT, 2021-04-15:
or just replace the flight controller? if you were in my pants what would you do?

mefistotelis, 2021-04-15:
Read that:
https://github.com/o-gs/dji-firmware-tools/wiki/Firmware-m0800#ofdm-usb-interface
Then go to "By service serial port" in Flashing.

Why are you mentioning FC again? What is the problem you want to solve?
You seem to have no idea what you're doing, I'd give the drone to someone else.

JGabT, 2021-04-15:
Problem:
AC beeping DD-DD (still don't know what this means)
RC was linked to AC with green light but DJI GO not showing the camera view thing
Now i tried firmware flashing by adding the 1.11.30 and adding a zero bit file named 'P3X_FW_DEBUG'
Now i can't link RC to AC tried doing the C1+C2+RECORD Buttons then click the mode dial button then clicking the link button on the P3P which is beeping (DD-DD)

First Problem i want to fix is the DD-DD now the logs says that m800 and m1500 can you help me solve this first problem?

mefistotelis, 2021-04-15:
The questions you need to answer yourself:
What is beeping? Gimbal? main motors? only one? all of them?
Do you have compatible FW version on AC and RC?

If any of main motors is beeping, you should check logs through debug serial interface of FC:
https://github.com/o-gs/dji-firmware-tools/wiki/Firmware-m0306#debug-serial-interface

JGabT, 2021-04-15:
Gimbal is beeping DD then DD slowly so no need to check debug serial interface of FC. This P3P is from my uncle, i do not know what version is RC and AC since i cannot connect them to the DJI GO app. Now im flashing firmware of 01.09.0060 through sd card.

JGabT, 2021-04-15:
ill try letting it beep for 20 minutes and come back here. (https://forum.phantomhelp.com/t/phantom-3-pro-beeping-dd-dd/6233) apparently dji "gave" an advice to leave it for 20 minutes

JGabT, 2021-04-15:
after reading the wiki properly and thoroughly i now understand it. since the m800 and m1500 is not being detected it means it is dead because it needs a working communication between the chips on the camera module.
so i need to either flash it by service serial port or by chip maker method. thanks btw.

JGabT, 2021-04-15:
Ambarella is working fine btw since i can take a video/picture from camera

JGabT, 2021-04-16:
isnt there any way to fix the dead module except for serial flashing?

mefistotelis, 2021-04-16:
Every chip has a manufacturer, who provides its documentation. To learn flashing methods, you have to read that. The document may be named "Programming Reference", "Bootloader useage", "Flashing Reference", etc.

JGabT, 2021-04-16:
do you know any methods on the chip of m0800/m1500?

mefistotelis, 2021-04-16:
Already pointed you to that.

JGabT, 2021-04-16:
oh Programming Reference, Bootloader usage, Flashing Reference ill look it up :) thanks for your time.

mefistotelis, 2021-04-16:
For these specific modules, I actually pointed you to specific information earlier. In general, be sure to read related wiki pages before asking for support.

JGabT, 2021-04-16:
is it possible to use an arduino to serial flash the dm368?

JGabT, 2021-04-16:
ill probably create a tutorial on how to serial flash the dm368 using an arduino uno if this works lol

mefistotelis, 2021-04-16:
Would be interesting.
The first thing is to flash bootloader (u-boot). For that, TI published a tool, with source. You'd have to find that source (it's in C# I think) and re-implement the protocol for Arduino.

JGabT, 2021-04-16:
hmm yeah but worth trying arduino since i already have one and if fails ill get the usb-to-ttl thingy and try.

JGabT, 2021-04-20:
hey mefistotelis, how about the dd-dd on the gimbal? even when i put an empty micro sd card it beeps dd dd like its installing a firmware.

mefistotelis, 2021-04-20:
> like its installing a firmware.

Maybe it thinks it does. Or maybe reports unrelated gimbal issue.

JGabT, 2021-04-20:
how do i look at that? i tried searching for issues like that but cant find one.

JGabT, 2021-05-13:
i found this thread on the dji forum site. look at leighton's post he has the same issue i am facing. is it possible to create a python program the connects to the P3P and automatically force update, downgrade the firmware until it can find the working one?

mefistotelis, 2021-05-13:
Try reading this:
https://github.com/o-gs/dji-firmware-tools/wiki/Flashing-firmware-via-SD-card

Not sure if you can trust what every person writes on random forums. But you do what you want.

JGabT, 2021-05-13:
Do you plan on creating an auto firmware update program for DJI products? (ex. user selects what firmware versions to install on the AC and what order to install it). After inserting a SD Card it automatically formats it, writes the firmware files and can optionally add the p3x debug file then ejects, repeat.

mefistotelis, 2021-05-13:
I expect the users of my tools to be able to do that and much more, without such level of spoon feeding. So, no.

JGabT, 2021-05-13:
Thanks. Ill create a pr if i am successful in creating the autoupdate tool.

closed 2021-06-13 by mefistotelis