o-gs/dji-firmware-tools #202: Help! error? :(

sstsvetkov, 2021-04-18:
C:\DJI>py comm_og_service_tool.py com3 -vv WM240 GimbalCalib JointCoarse
Opened com3 at 9600

Info: The Gimbal will move through its boundary positions, then it will fine-tune its central position. It will take around 15 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 5d df 20 04 08 01 2f cb
Sending packet...
Waiting for reply...
Retrieved 2 packets (82b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a 5d df a0 04 08 01 6d 50
Error: Unrecognized response to calibration command JointCoarse request.

C:\DJI>py comm_og_service_tool.py com3 -vv WM240 GimbalCalib LinearHall
Opened com3 at 9600

Info: The Gimbal will slowly move through all positions in all axes, several times. It will take around 30 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 75 e8 20 04 08 02 81 b9
Sending packet...
Waiting for reply...
Retrieved 1 packets (14b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a 75 e8 a0 04 08 01 58 10
Error: Unrecognized response to calibration command LinearHall request.

lopixable, 2021-04-18:
Hi you can ignore the "Error: Unrecognized response to calibration command ... request."
If you see your gimbal moving then it's working.

sstsvetkov, 2021-04-18:
At the first stage, it gives an error (JointCoarse)
C: \ DJI> py comm_og_service_tool.py com3 WM240 GimbalCalib JointCoarse

Info: The Gimbal will move through its boundary positions, then it will fine-tune its central position. It will take around 15 seconds.

Error: Unrecognized response to calibration command JointCoarse request.

In this case, the camera starts to rut and does not upload the process

sstsvetkov, 2021-04-18:
If you go to the DJANGO app, the camera does not respond to turns, and does not see the suspension :(

sstsvetkov, 2021-04-18:
Before this calibration, the camera was slightly crooked, but it could be rotated. And now he is trying to calibrate himself and at the last stage he freezes looking to the left

sstsvetkov, 2021-04-18:
I tried all the options, maybe I'm doing something wrong?

devbrones, 2024-08-02:
having same issue with M2P both on windows 11 and linux hosts