o-gs/dji-firmware-tools #203: FPV Goggles and Air Unit Firmware extraction fails

j005u, 2021-05-04:
FPV Goggles and Air Unit Firmware extraction fails with the following:

`# python ../dji-firmware-tools/dji_imah_fwsig.py -vv -u -i gl150_2801_v10.01.11.41_20200327.pro.fw.sig
gl150_2801_v10.01.11.41_20200327.pro.fw.sig: Opening for extraction and un-signing
gl150_2801_v10.01.11.41_20200327.pro.fw.sig: Unpacking image...
OrderedDict([('magic', b'IM*H'),
            ('header_version', 2),
            ('size', 108525536),
            ('reserved', b'\x00\x00\x00\x00'),
            ('header_size', 224),
            ('signature_size', 256),
            ('payload_size', 108525056),
            ('target_size', 108525536),
            ('os', 0),
            ('arch', 0),
            ('compression', 0),
            ('anti_version', 0),
            ('auth_alg', 262146),
            ('auth_key', 'PRAK'),
            ('enc_key', 'UFIE'),
            ('scram_key', b'o\xc6\xca\xa2\xcbrs*\xf7RY\xef\xac\xc4\xda;'),
            ('name', '2801'),
            ('type', 'PIGE'),
            ('version', 167840553),
            ('date', 538968871),
            ('encr_cksum', 3855080739),
            ('reserved2', b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'),
            ('userdata', b''),
            ('entry', b'\x00\x00\x00\x00\x00\x00\x00\x00'),
            ('plain_cksum', 1126738375),
            ('chunk_num', 1),
            ('payload_digest', b'\x96\xe5X6\xa9Gu\x04\xd7\xe9gR\xc5\xc6\x10\x0cg\x03\xcaql\xf9P@\xf3z\x9dG\x82\x01\xd4\x88')])
Error: Image file head signature verification failed.`

Using -f to force the extraction seems to result in garbage data.

Any tips?

mefistotelis, 2021-05-04:
There are multiple PRAKs. The one for goggles isn't public yet.

tfurci, 2021-05-12:
So that means that dji would have to release it and then it would be posible to extract?

mefistotelis, 2021-05-12:
That's one possibility. They did published them in the past:
https://github.com/o-gs/dji-firmware-tools/commit/112b04bd154808db3838b4d47cad2f77b12dad8a

druckgott, 2022-02-13:
Is it already availible I testen it today and have Same problem.

j005u, 2022-02-15:
Yeah this issue can be closed. @druckgott You need to look up the correct keys in dji_imah_fwsig.py. The AU is WM150, The vista is LT150 and the Goggles V1 are GL150, they all share the same keys. Goggles V2 are GL170.

druckgott, 2022-02-17:
Ok I now use it like this:
 python .\dji-firmware-tools-master\dji_imah_fwsig.py -vv -k PRAK-2020-01 -u -i gp150_2805_v10.00.16.84_20210225.pro.fw.sig

But I get an error:
`gp150_2805_v10.00.16.84_20210225.pro.fw.sig: Image file head signature verification passed.
gp150_2805_v10.00.16.84_20210225.pro.fw.sig: Encrypted data checksum 0x3D3A38CB matches.
gp150_2805_v10.00.16.84_20210225.pro.fw.sig: Warning: 'UFIE' matches multiple keys; using first, 'UFIE-2021-06'
gp150_2805_v10.00.16.84_20210225.pro.fw.sig: Key choices: UFIE-2021-06, UFIE-2020-04, UFIE-2019-11, UFIE-2018-07, UFIE-2018-01
gp150_2805_v10.00.16.84_20210225.pro.fw.sig: Unpacking encrypted chunk '2805'...
OrderedDict([('id', '2805'), ('offset', 0), ('size', 175113994), ('attrib', 0), ('address', 0), ('reserved', b'\x00\x00\x00\x00\x00\x00\x00\x00')])
gp150_2805_v10.00.16.84_20210225.pro.fw.sig: Un-signed 1 chunks, skipped/truncated 0 chunks.
gp150_2805_v10.00.16.84_20210225.pro.fw.sig: Decrypted chunks checksum 0x581BDFE6, expected 0xD5D8EC7F.
Error: Decrypted chunks checksum verification failed.`

j005u, 2022-03-04:
@druckgott sorry for the late response.

You need to add -k UFIE-2020-04 for the decryption key. PRAK is just the verification key, which can also be ignored with -f.

druckgott, 2022-04-10:
Ok thx It works now with:
python ..\dji-firmware-tools-master\dji_imah_fwsig.py -vv -k PRAK-2020-01 -u -i gp150_2805_v10.00.16.84_20210225.pro.fw.sig

Now I have 2 ini files and one bin File. If I understand it right I now have to use: ./arm_bin2elf.py
But I do not realy undedrstand how I can get the right values for the input to the script. There is a .sh script in testfolder but I do not have a linux system.

j005u, 2022-04-10:
@druckgott it's a zip file, just extract it with 7zip or the like. Then you'll find a bunch of further encrypted partitions that require the TBIE key and the vendor and system partitions, which can be converted to ext4 images with https://github.com/xpirt/sdat2img. The latter two contain most of the interesting/easily readable parts.

You'll either need something that can read ext4 images under Windows, or I recommend using WSL (Windows Subsystem for Linux).

Here's how I usually do it under WSL, presuming dat2img is checked out in the parent folder of where I'm working.
```
python ../dat2img/sdat2img.py system.transfer.list system.new.dat system.img
python ../dat2img/vdat2img.py vendor.transfer.list vendor.new.dat vendor.img
mkdir mnt
sudo mount system.img mnt
mkdir system
mkdir vendor
cp -R mnt/* system/
sudo umount mnt
sudo mount vendor.img mnt
cp -R mnt/* vendor/
sudo umount mnt
rm system.img
rm vendor.img
```