o-gs/dji-firmware-tools #205: Hot to run "test_dji_imah_fwsig_rebin1.sh"  ??

9930wilson, 2021-05-15:
I was able to extract the wm161.sig files to .bin files.
And now trying to repack the .bin files to .sig again, but not being able to.

Anyone knows what command to use??
and which script to use for .bin to .sig conversion..

JefDrones, 2021-06-28:
How did you get? i'm trying this way but it's returning error.
If anyone else can help...thank you in advance.
[image]

mefistotelis, 2021-06-28:
https://letmegooglethat.com/?q=ModuleNotFoundError%3A+No+module+named+%27Crypto%27

JefDrones, 2021-07-01:
Thanks mefistotelis ... sorry.... i should have googled it first
I understand that for this tool I have to use linux or windows instead and install this Crypto library... I installed kali on a virtual machine and I need to learn how to work with linux and also find out that on this virtual machine the serial port won't work... .then i will format the pc and install a dual boot ....... hope i'm on the right path .... let's go

mefistotelis, 2021-07-01:
Most of the tools are written in Python, and therefore will work on any OS with Python available.

The repository also contains some shell (bash) scripts; these scripts execute some tools which are available for Linux and Windows, not sure about MacOS.

JefDrones, 2021-07-02:
[image]

mefistotelis, 2021-07-03:
File `*_head.ini` is created when unsigning. It contains additional information required to re-sign the file.

Example content of such file:
```
# DJI Firmware Signer main header file.
# Generated on 2021-07-02 22:08:19
name=bootloader
pkg_format=2018
version=00.00.00.00
anti_version=00.00.00.00
date=2017-12-27
enc_key=TBIE
auth_alg=2
auth_key=PRAK
os=0
arch=0
compression=0
type=TBLD
userdata=
entry=0000FCFF00000000
scramble_key=B6 50 7B 4C 3F 28 E4 B5 FE F5 39 CD 93 58 B0 28
modules=BLLK
```