o-gs/dji-firmware-tools #207: Flash A3 flight controller

cmillauriaux, 2021-05-26:
Hello everyone !

I found this awesome repository with every tools I need to make my custom firmware. It's a very complete solution, congrats to every developers behind it !

I'm just stuck on the flashing step : I own a matrice 600 pro and I understand, thanks to the wiki, I need to use a ST Link to flash the firmware. But i have two questions 
- How to wire the ST Link to the flight controller ? Do I need to open it or can i just use the pins exposed ?
- I repack the updated firmware with the right tool and obtain a .bin file. There is any warning or can I just flash my flight controller with this file ? I read from an issue that we need to be careful with encrypted datas, but if I could read and update parameters, I decrypted the files ?

Thanks to the community !

mefistotelis, 2021-05-26:
I never had Martice; some people described it here though:
https://dji.retroroms.info/howto/start
ST-link would work (if the FC is STM32 on that platform), but there are often ways to use DJIs fw update mechanism, which is easier.

cmillauriaux, 2021-05-26:
Thanks to your answer !
I tried to flash the firmware following their tutorial (http://dji.retroroms.info/howto/install_modded_m100_m600_fw) but the thing is, with Matrice m600 pro, it's not "DJI Assistant 2" but "DJI Assistant 2 Entreprises". I tried to use "DJI Assistant 2" but it only shows "LightBridge" module and not "M600" module, whereas with "DJI Assistant 2 Entreprises" it shows the two modules.