o-gs/dji-firmware-tools #209: MINI 2 ERRORS/ISSUES after GIMBAL/CAMERA replacement

plastinaf, 2021-06-12:
Hi
After crash i replace Gimball on MINI 2 but the process don't works for MINI2 (Work's fine on MINI 1 WM160)
I try with WM160 but not work
C:\dji-firmware-tools>C:\Python39\python ./comm_og_service_tool.py COM11 -vv WM160 GimbalCalib JointCoarse
Opened COM11 at 9600
Info: The Gimbal will move through its boundary positions, then it will fine-tune its central position. It will take around 15 seconds.
Prepared binary packet:
55 0e 04 66 0a 04 1e 0f 20 04 08 01 b4 3a
Sending packet...
Waiting for reply...
Retrieved 1 packets (14b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a 1e 0f 80 04 08 01 a5 2e
Error: Unrecognized response to calibration command JointCoarse request.

Code of MINI 2 is WM161
This code is missing from the PRODUCT_CODE
Developers have probably not yet finalized.
class PRODUCT_CODE(DecoratedEnum):
    A2     =  0 # Released 2013-09-04 A2 Flight Controller
    P330   =  1 # Released 2013-01-07 Phantom 1
    P330V  =  2 # Released 2013-10-28 Phantom 2 Vision
    P330Z  =  3 # Released 2013-12-15 Phantom 2 w/ Zenmuse H3-2D
    P330VP =  4 # Released 2014-04-07 Phantom 2 Vision+
    WM610  =  5 # Released 2014-11-13 Inspire 1
    P3X    =  6 # Released 2015-03-09 Phantom 3 Professional
    P3S    =  7 # Released 2015-03-09 Phantom 3 Advanced
    MAT100 =  8 # Released 2015-06-08 Matrice 100
    P3C    =  9 # Released 2015-08-04 Phantom 3 Standard
    MG1    = 10 # Released 2015-11-27 Agras MG-1
    WM325  = 11 # Released 2016-01-05 Phantom 3 4K
    WM330  = 12 # Released 2016-03-02 Phantom 4 (now referenced as Phantom 4 Standard)
    MAT600 = 13 # Released 2016-04-17 Matrice 600
    WM220  = 14 # Released 2016-09-28 Mavic Pro (also includes Released 2017-08-24 Mavic Pro Platinum)
    WM620  = 15 # Released 2016-11-16 Inspire 2
    WM331  = 16 # Released 2016-11-16 Phantom 4 Pro
    MAT200 = 17 # Released 2017-02-26 Matrice 200
    MG1S   = 18 # Released 2017-03-28 Agras MG-1S
    WM332  = 19 # Released 2017-04-13 Phantom 4 Advanced
    WM100  = 20 # Released 2017-05-24 Spark
    WM230  = 21 # Released 2018-01-23 Mavic Air
    WM335  = 22 # Released 2018-05-08 Phantom 4 Pro V2
    WM240  = 23 # Released 2018-08-23 Mavic 2 Pro/Zoom
    WM245  = 24 # Released 2018-10-29 Mavic 2 Enterprise
    WM246  = 25 # Released 2018-12-20 Mavic 2 Enterprise Dual
    WM160  = 26 # Released 2019-10-30 Mavic Mini
    WM231  = 27 # Released 2020-04-28 Mavic Air 2

ALT_PRODUCT_CODE = {
    'S800': 'A2', # Released 2012-07-25 Hexacopter frame, often sold with Dji A2 Flight Controller
    'S1000': 'A2', # Released 2014-02-24 Octocopter frame, often sold with Dji A2 Flight Controller
    'S900': 'A2', # Released 2014-08-04 Hexacopter frame, often sold with Dji A2 Flight Controller
    'PH3PRO': 'P3X',
    'PH3ADV': 'P3S',
    'PH3STD': 'P3C',
    'P3XW': 'WM325',
    'P4': 'WM330',
    'PH4': 'WM330',
    'PH4PRO': 'WM331',
    'PH4ADV': 'WM332',
    'SPARK': 'WM100',
    'MAVIC': 'WM220',
    'MAVAIR': 'WM230',
    'M2P': 'WM240',
    'M2Z': 'WM240',
    'M2E': 'WM245',
    'M2ED': 'WM246',
    'MMINI': 'WM160',
    'MAVAIR2': 'WM231',
if you have any suggestion please tell me how to

jefsilv, 2021-06-13:
I think it's the answer.
Code of MINI 2 is 

> WM161

This code is missing from the PRODUCT_CODE
Developers have probably not yet finalized.
class PRODUCT_CODE(DecoratedEnum...mm

JefDrones, 2021-06-20:
you can use WM160 and it will work normally, as long as there is no hardware problem