o-gs/dji-firmware-tools #212: error related dji_imah_fwsig.py for decryption

taek2021, 2021-07-07:
hello! 
i really thanks for this tool.  i got huge help

but i have some problem. 

1. I got some encrypted file after remove "IM*H" sign using dji_imah_fwsig.py.
*.ini, *.bin, and original file.

2. and then, i use dji_mvfc_fwpak.py for *.bin file.
like this line
> dji_mvfc_fwpak.py -dec -i filename.bin

and then, i got some error line.

Target:     0303
Version:    v17.222.109.209
Time:       1989-05-17 19:46:56
Size:       3921319770
Data MD5:   b'2b9627e5bcb7435ec04e6a628506a50f'
Header CRC: e031
Error: CRC of the header doesn't match!

i usually got above error line when i use dji_mvfc_fwpak.py for encrypted file after remove "IM*H" sign
can anyone help me about this situation??
if who know solutions, tell me plz

thanks!

mefistotelis, 2021-07-07:
If header fields have unrealistic values, this means different data is being interpreted as the header.
In other words, this file type is not supported by the tool.

taek2021, 2021-07-07:
> If header fields have unrealistic values, this means different data is being interpreted as the header.
> In other words, this file type is not supported by the tool.

thanks mefi!! **Thank you very much for your effort in making this tool.**

and then, how can i decrypt file without IM*H??

using your tool, i can got mini2 aircraft  firmware(wm161_0100) file like etc, bin, ...
but i fault decrypt wm161_0306, 0905 and 
mini2 controller firmware(wm161b) too.

1. in wm161, mini2 aircraft firmware, i only success wm161_0100's system file. 
in details, i only got wm161_0100_v01.43.00.39_20210201.pro.fw.bin's sys file(etc,bin, ...)

but i can't read other wm161_1100_v05.00.10.18_20200609_jp02.pro.fw_1100.bin, wm161_1100_v07.74.01.04_20201016_gl01.pro.fw_1100.bin, wm161_1200_v01.60.31.00_20201123.pro.fw_1200.bin,
wm161_1200_v01.60.31.00_20201123.pro.fw_1200.bin. 
-> because this file's type is tga. so i can't read more 

2. in wm161b, mini2 controller firmware file, i can enter rc-n1-wm161b_sys.bin -> wm161b_1301 -> ap.img, cp.img, ...etc -> ap_RTOS.bin, cp_RTOS.bin
=> i be blocked on this step. i can't decrypt *.RTOS.bin

I'm sorry to bother you but can you give me some solutions or guide about above problems?? 

thanks

mefistotelis, 2021-07-07:
For 2nd level encryption after IMaH - The published `mvfc` tool currently only supports pre-cryptoshell (pre-wm230) firmwares.
To have support of newer platforms, you'd have to figure out the algorithm and key, then make a tool for that.

For `cp_RTOS.bin` within `rc-n1-wm161b_1301_v04.11.00.13_20210406.pro.fw` - that specific file is decrypted properly by tests included in the repository. Check how it's done there.

taek2021, 2021-07-07:
> For 2nd level encryption after IMaH - The published `mvfc` tool currently only supports pre-cryptoshell (pre-wm230) firmwares.
> To have support of newer platforms, you'd have to figure out the [...]

Deer mefi!! i did try and back.

i used 14 tool in tests  include this repositiory for ap_RTOS_bin and cp_RTOS.bin
but i can't get noting.. 
can you teach what in 14 tools in tests and what option should i with that tool??
or can you tell me link about guide for tools in tests??
and then, i will try hard
sorry for bother you, mefi!

thanks!!

mefistotelis, 2021-07-07:
This is how I'm executing the tests:
```
./tests/test_imah_all.sh 2>&1 | tee ./test_imah_all.log
```

These tests have no ability do download the firmwares so you must first get them with DDD. The firmwares you need are all listed within `test_imah_all.sh`.

The tests will clean up afterwards - all extracted files will get deleted, leaving only logs. If you want to keep them, you'll have to modify the script.

taek2021, 2021-07-07:
> This is how I'm executing the tests:
>
> ```
> ./tests/test_imah_all.sh 2>&1 | tee ./test_imah_all.log
> ```
>
> These tests have no ability do download the firmwares so you must first get them with DDD. The [...]

Deer mefi!! i tried that and back..  i fault..

i tried under line

1. 
$ ~/tool/dji-firmware-tools/tests/test_imah_all.sh ap_RTOS.bin 2>&1 | tee test  
-> permisstion denied

2.
$sudo ~/tool/dji-firmware-tools/tests/test_imah_all.sh cp_RTOS.bin 2>&1 | tee test
-> not exist command

3. $ ~/tool/dji-firmware-tools/tests/test_imah_all.sh 2>&1 | tee test  
-> permisstion denied

4. $ sudo ~/tool/dji-firmware-tools/tests/test_imah_all.sh 2>&1 | tee test  
-> not exist command

i think i didn't understand in your solutions..

mefistotelis, 2021-07-07:
The command I proposed just executes a test script. If you're having trouble interpreting all the decoration around that command, just run:
```
./tests/test_imah_all.sh
```

taek2021, 2021-07-07:
> The command I proposed just executes a test script. If you're having trouble interpreting all the decoration around that command, just run:
> 
> ```
> ./tests/test_imah_all.sh
> ```
thanks mefi!!
but I think my computer have some problem

i can't execute ./tests/test_imah_all.sh

i tried 4 type under line

1.
$ ./tests/test_imah_all.sh
=> zsh: permisstion denied

2.
$ ./tests/test_imah_all.sh filename
=> zsh: permisstion denied

3.
$sudo ./tests/test_imah_all.sh
=> /tests/test_imah_all.sh: not exist command

4.
$sudo ./tests/test_imah_all.sh filename
=> /tests/test_imah_all.sh: not exist command

so i try finding this reason

sorry for bother you
i will eternally remember your warm help!!

mefistotelis, 2021-07-07:
Under UNIX, a file needs 'executable' attribute to be executed.
Learn about attributes, and how to change them.

taek2021, 2021-07-07:
> Under UNIX, a file needs 'executable' attribute to be executed.
> Learn about attributes, and how to change them.

thanks!!
i did chmod u+x test_imah_all.sh
and, i tried underline

& ~/tool/dji-firmware-tools/tests/test_imah_all.sh 2>&1 ap_RTOS.bin
=>
### SUITE sub-script not found; executed from wrong directory? ###

& ~/tool/dji-firmware-tools/tests/test_imah_all.sh ap_RTOS.bin
=>
### SUITE sub-script not found; executed from wrong directory? ###

is this tool support for ap_RTOS.bin/cp_RTOS.bin?? 
i did V04.11.0016_RC-N1-WM161b_dji_system.bin too
but i had same result

mefistotelis, 2021-07-07:
The script message says it all. Execute it properly.

Also, the test script does not take parameters. No idea why you're trying to pass a file there.
Read the script to see how it works.

It's a `bash` script. Read about bash if you don't understand the syntax.

taek2021, 2021-07-08:
> The script message says it all. Execute it properly.
> 
> Also, the test script does not take parameters. No idea why you're trying to pass a file there.
> Read the script to see how it works.
> 
> It's a `bash` script. Read about bash if you don't understand the syntax.

 Thank you very much for your kindness!! :)

mefistotelis, 2021-07-08:
Just to make sure you know what your goal is:
The test extract firmwares, then extract the files inside, then re-pack them and deletes everything.
You may either:
1. execute the test as-is, and text output of the script will contain command lines used to unsign every file, then copy these commands and execute them manually. or
2. Modify the script so that it does not delete all the binaries. Then you won't have to run any commands manually, the binaries will be where the script extracted them.

Either way, you need to understand how bash scripts work to use them correctly.

taek2021, 2021-07-08:
> Just to make sure you know what your goal is:
> The test extract firmwares, then extract the files inside, then re-pack them and deletes everything.
> You may either:
>
> 1. execute the test as-is, and [...]

thanks!! 👍 
Deer mefi! 
i want take mini2 controller's system file. and then i show how is mini2 do communication with controller... etc
ex) wifi setting... etc
because i think mini 2's controller is AP / mini 2's craft is client.

but i only catched "rc-n1-wm161b_sys.bin -> wm161b_1301 -> ap.img, cp.img, ...etc -> ap_RTOS.bin, cp_RTOS.bin"
i be blocked on this step.

above step is tried in formal Distribution File
ㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡㅡ
+ 
i tried to extracted *.bin File too from flash memory of mini2 controller by rasberry pi, minicom
but this file is unnormal. so I failed even though I used the tool like binwalk, ... your tool too.
The file doesn't budge from original *.bin 

can you tell me some tip this situations?

i goal is finding Communication Method of ocusink of mini2

i am so sorry and thank you always :)

mefistotelis, 2021-07-08:
The files flashed on the device are encrypted as well, and have IMaH format. First stage loader (which is not a part of firmware update) decrypts them when the chip is booting.

You can figure out how to decrypt the images by looking at the tests. Properly decrypted file looks like this:
[image]

taek2021, 2021-07-09:
> The files flashed on the device are encrypted as well, and have IMaH format. First stage loader (which is not a part of firmware update) decrypts them when the chip is booting.
> 
> You can figure out how to decrypt the images by looking at the tests. Properly decrypted file looks like this:
> [image]

wow..

i think I was entirely misled about the matter

0. i thinked some file without IM*H after dji_imah_fwsig.py is encrpyed 
but this file was wrote by CPU level syntax so i can't understand it. right?? 

1. is it assembly level syntax?? or other??

2. how am i anlyze/understand this type syntax??.. give me some hint if you don't feel bothering..

i always thanks you so much mefi :)

mefistotelis, 2021-07-09:
0. You can figure out if a file is encrypted by just looking at it. But if you don't see it, you can use any tool for entropy checking. `binwalk` is a popular tool which can do that. If entropy is uniform across file and very high - close to 1 - then the file is encrypted.

1. I didn't analyzed this file. On first look, this seem to be a typical binary which is placed at specific RAM address and executed by higher level bootloader. The content looks like ARM Thumb instruction set, and header looks like typical ISV table - first stacks position, then addresses of routines within the binary. These addresses can be used to guess base address where the binary is placed in RAM.

2. To analyze it, you figure out instruction set and base address, then open it in proper disassembler. For example IDA Pro or Ghidra. You may also re-construct ELF header of the binary, then more tools will be able to read it.

3. No idea what are files with "SDRH" magic. To understand that, you'd have to find SDK or documentation of the chip which it programs, and read there.

taek2021, 2021-07-09:
> 1. You can figure out if a file is encrypted by just looking at it. But if you don't see it, you can use any tool for entropy checking. `binwalk` is a popular tool which can do that. If entropy is uniform across file and very high - close to 1 - then the file is encrypted.
> 2. I didn't analyzed this file. On first look, this seem to be a typical binary which is placed at specific RAM address and executed by higher level bootloader. The content looks like ARM Thumb instruction set, and header looks like typical ISV table - first stacks position, then addresses of routines within the binary. These addresses can be used to guess base address where the binary is placed in RAM.
> 3. To analyze it, you figure out instruction set and base address, then open it in proper disassembler. For example IDA Pro or Ghidra. You may also re-construct ELF header of the binary, then more tools will be able to read it.
> 4. No idea what are files with "SDRH" magic. To understand that, you'd have to find SDK or documentation of the chip which it programs, and read there. If you can't find the documentation, you can also try checking which executable loads that file - maybe it is parsed on CPU side before being sent to hardware.

if you need this pic releated mini2 controller, i give you :)

[image]
[image]

mefistotelis, 2021-07-10:
Thank you! Could you re-do these though? The photos are smudged, your hand probably moved towards the board as you were taking them. Also, you don't mind, could you make markings of the thermal paste covered chips visible?

I am editing the photos before putting them on the wiki, fixing perspective and shadows and so on. So while these are acceptable photos, I'd have much easier task if they were sharper.

taek2021, 2021-07-12:
> Thank you! Could you re-do these though? The photos are smudged, your hand probably moved towards the board as you were taking them. Also, you don't mind, could you make markings of the thermal [...]

here you are :)

[image]
[image]

coaz, 2021-07-12:
Hello. I'm trying to decript the mini 2 firmware with dji_imah_fwsig.py. I got the bin file but as I understood dji_mvfc_fwpak.py doesnt works with bin files. Is this correct? 
I used arm_bin2elf.py to convert this file to elf.  Am I going well? I dont know how to edit this elf file. It looks strange.
[image]
I have a mini 2 I crashed, if you want I can share you the images.

mefistotelis, 2021-07-12:
> I got the bin file but as I understood dji_mvfc_fwpak.py doesnt works with bin files. Is this correct?

No, that's not correct.
Make sure you understand what binary files are, and you're able to distinguish file name from content.
`dji_mvfc_fwpak.py` works with 2nd level encryption of Mavic Pro FC firmwares, not with ELFs.

> I have a mini 2 I crashed, if you want I can share you the images.

Sure, if you can provide images of boards which are not on the wiki, or are on the wiki but you can make better quality photos, then please do. The wiki:
https://github.com/o-gs/dji-firmware-tools/wiki/DJI-Hardware#mini-2

coaz, 2021-07-19:
Thank you Mefistotelis. Another question: After get wm161_0306xxx.bin unsigning this file 161_0306xxx.pro.fw.sig with dji_imah_fwsig.py Im trying to extract flight parameters with dji_flyc_param_ed.py but Im getting this error: "Flight controller parameters array signature not detected in imput file." 

My question is that does the file could have some errors?

mefistotelis, 2021-07-19:
> My question is that does the file could have some errors?

You either decrypted both levels successfully and the raw data contains the parameter array structures, or you didn't and the data looks random and has nothing in it.

As I wrote before:
For 2nd level encryption after `IMaH` - The published `dji_mvfc_fwpak.py` tool currently only supports pre-cryptoshell (pre-wm230) firmwares.
To have support of newer platforms, you'd have to figure out the algorithm and key, then make a tool for that.

closed 2021-07-21 by taek2021

usatenko, 2022-07-07:
sorry, folks, where you download these firmware files, like rc-n1-wm161b_1301_v04.11.02.16_20220614.pro.fw.sig ?

usatenko, 2022-07-10:
> sorry, folks, where you download these firmware files, like rc-n1-wm161b_1301_v04.11.02.16_20220614.pro.fw.sig ?
ignore it, I figured it out