o-gs/dji-firmware-tools #213: Dji Mini 2 controller firmware

Andr3WTF, 2021-07-09:
I made backup from MINI 2 flash with firmware. Is it possible add it in wiki?
[image]
https://drive.google.com/file/d/1p7YoWlAe-zMeN1nhUmlVsWDEA8sIZRLd/view

mefistotelis, 2021-07-09:
Thank you, it is definitely worth adding some info on that, at least memory map if not the complete file.

Can you provide me with good quality photo of the board as well, both sides with chips visible?

taek2021, 2021-07-09:
[](url)> Thank you, it is definitely worth adding some info on that, at least memory map if not the complete file.
> 
> Can you provide me with good quality photo of the board as well, both sides with chips [visible?](url)

hello mefi! do you want this pic?

[image]
[image]

Andr3WTF, 2021-07-09:
Unfortunately i already assembled it and didn't plan open again in nearest future.
You can see inside pictures there: https://zhuanlan.zhihu.com/p/162144640
[image]

mefistotelis, 2021-07-10:
No problem; @taek2021 will provide me with photos.

tecdrone, 2021-09-11:
> Fiz backup do MINI 2 flash com firmware. É possível adicioná-lo no wiki? https://drive.google.com/file/d/1p7YoWlAe-zMeN1nhUmlVsWDEA8sIZRLd/view
> [image]
> [](https://drive.google.com/file/d/1p7YoWlAe-zMeN1nhUmlVsWDEA8sIZRLd/view)

Ola amigo você ainda tem o arquivo disponível para download?

tecdrone, 2021-09-11:
> Sem problemas; @ taek2021 vai me fornecer fotos.

Eu tem fotos de uma totalmente aberto .você pode me ajuda com o arquivo de atualização