o-gs/dji-firmware-tools #214: BQ40z307 Unseal do not work

vadda68, 2021-07-18:
$ python3 comm_sbs_bqctrl.py -vvv --dev_address 0x0b sealing Unseal
Opening i2c:1
Raw write: DATA=22 3e
Raw write: DATA=20 3e
Raw write: DATA=22 3e
Raw FirmwareVersion response: 0b 43 07 01 01 00 27 00 03 85 02 00 8f 8f 8f
Auto-selected chip: BQ40z307, Texas Instruments BQ40z307 chip for DJI
Importing comm_sbs_chips/BQ40z307.py
Store ManufacturerAccess.SecKeyWord0: 00 WORD=0x414
Write ManufacturerAccess: CMD=00 WORD=14 04
Store ManufacturerAccess.SecKeyWord1: 00 WORD=0x3672
Write ManufacturerAccess: CMD=00 WORD=72 36
Reading write_word_subcommand command at addr=0xb, cmd=0x0, type=uint16, opts={'subcmd': <MANUFACTURER_ACCESS_CMD_BQ40.OperationStatus: 84>}
Query ManufacturerAccess.OperationStatus: 00 WORD=0x54
Write ManufacturerAccess: CMD=00 WORD=54 00
Raw ManufacturerAccess.OperationStatus response: 04 06 03 04 00 c9
MA.OperationStatus:	0x00040306	bitfields	Operational Status bits
...
...
        SECURITY_MODE:	3=Sealed	[SEC]	Security Mode

mefistotelis, 2021-07-20:
I tested the unsealing on Mavic 1 battery I have, and it was working.
Why it didn't worked for you - I don't know. Maybe unseal key is different, and maybe the issue lies somewhere else.

mjnhchj, 2021-07-30:
that's right.  I can't do it with my MM battery either!

mefistotelis, 2021-07-30:
When you figure out the cause, make sure to share.

mjnhchj, 2021-07-30:
I used this tool https://mega.nz/fm/DB4mxSCL All work fine with CP2112 adapter

mixeysan, 2021-07-30:
https://github.com/o-gs/dji-firmware-tools/issues/196#issuecomment-885636238

islonina, 2021-08-13:
> I used this tool https://mega.nz/file/yEJGwJYR#O3jh40agtSWE11HRySdg5C6X5y2OeHW2c_A4hOcCVFs All work fine with CP2112 adapter

I can unseal battery without problem. But still no option for count number and seal option. Mavic mini 1

noname443, 2021-09-18:
hello everyone, how good there are enthusiasts who can embrace almost anything. this topic helped me check out some AIR 1 batteries that I managed to buy some time ago. However, I have a problem with sealing. The program presented here does not have this option, but it is in the last screenshot. What is this version, because visually the interface looks identical, it is probably a newer development version. There is a chance to get access to this version? I have one more question. Has anyone tried to replace cels ( mavic AIR 1) with more capacity, but not 18650. I am asking because 
I noticed repeated damage that the last cell in blocked batteries has reversed polarity and cannot be regenerated, charged. Thanks in advance for any suggestions.

mjnhchj, 2021-09-23:
That software is "DJI battery killer" by Mixeysan developer! It will release soon ...

mixeysan, 2021-09-26:
[Boom](url=http://master-club.org/drone/DJI_Battery_Killer_ver.0.4beta_26.09.2021.7z)[/Boom]

I have not finished everything yet. In particular, you still need to remove the unnecessary and disable to write of FW BQ30Z55, because the CP2112 adapter will not be able to do this anyway. But the change in capacity, voltages, charge max, reset of cycles seems to be plowing both on bq30z55 and on bq9003. To get full access in BQ30Z55 when the key is unknown, use the Full Access button.
There are still small shoals, with time I'll correct it. And I'll release it a little later when will be free time)

antonio18210, 2021-09-26:
Hola soy nuevo en este tema y me interesa mucho, he probado la version anterior y puedo leer datos de la bateria 
del mini1 es estupendo, espero la nueva version del programa, muchas gracias "mixeysan" y si puedo ayudar de alguna
manera decidmelo ...

NHAMTAMPHUOCTHIEN, 2021-10-04:
Mr. Mixeysan
Can i get the current software? if possible please send it to me, thank you very much!
my email: huynhphuocnhan1907@gmail.com

tungtranmobile, 2021-10-04:
> hello everyone, how good there are enthusiasts who can embrace almost anything. this topic helped me check out some AIR 1 batteries that I managed to buy some time ago. However, I have a problem [...]
you can get link to my email tungninhkieu@gmail.com best new version tool please. thanks

mixeysan, 2021-10-04:
> > hello everyone, how good there are enthusiasts who can embrace almost anything. this topic helped me check out some AIR 1 batteries that I managed to buy some time ago. However, I have a problem with sealing. The program presented here does not have this option, but it is in the last screenshot. [image] What is this version, because visually the interface looks identical, it is probably a newer development version. There is a chance to get access to this version? I have one more question. Has anyone tried to replace cels ( mavic AIR 1) with more capacity, but not 18650. I am asking because I noticed repeated damage that the last cell in blocked batteries has reversed polarity and cannot be regenerated, charged. Thanks in advance for any suggestions.
> > you can get link to my email [tungninhkieu@gmail.com](mailto:tungninhkieu@gmail.com) best new version tool please. thanks

U can find the last version here - https://github.com/o-gs/dji-firmware-tools/issues/214#issuecomment-927307835

mixeysan, 2021-10-04:
[DJI_Battery_Killer_ver.0.4beta_26.09.2021.7z.zip](https://github.com/o-gs/dji-firmware-tools/files/7279207/DJI_Battery_Killer_ver.0.4beta_26.09.2021.7z.zip)

tungtranmobile, 2021-10-04:
Thank you very much!

NHAMTAMPHUOCTHIEN, 2021-10-04:
> [DJI_Battery_Killer_ver.0.4beta_26.09.2021.7z.zip](https://github.com/o-gs/dji-firmware-tools/files/7279207/DJI_Battery_Killer_ver.0.4beta_26.09.2021.7z.zip) Thanks mefistotelis for some help :)
> 
> During the test, repairing the firmware in BQ9003 when the EEPROM is empty - FFFFFFFFF. The chip is always in bootloop. See video http://master-club.org/drone/BQ9003_bootloop_repair_empty_EEPROM.mp4

Hi Mr.mixeysan  
when i open the program an error message is not found file mfc140u.dll

mixeysan, 2021-10-05:
> when i open the program an error message is not found file mfc140u.dll

Check your files or system. You are the first who wrote about such a problem

0roblequad, 2021-10-29:
> @0roblequad Usually the boards came with those pins labeled. Check with your cp2112 vendor. (you must connect gnd pin also)

gracias

0roblequad, 2021-10-29:
sinisterRJ gracias ya lo solucione

mjnhchj, 2021-11-10:
https://youtu.be/RkdUHQI2U_8

0roblequad, 2021-11-12:
https://www.youtube.com/watch?v=_u0qoLGzXpg&t=51s&ab_channel=roblequad

mansigen, 2021-11-14:
hello, is possible unlock BQ30Z55 DJI RONIN MX 1580 BATTERY? dji not sell this battery, ronin mx is an obsolete product...
so i have tested your software... and run well, read and write, but battery after unseal, clear pf, and seal don't charge...
can I collaborate with us for develop a new version for unlock dji old gimbal battery? best regards

mixeysan, 2021-11-14:
> hello, is possible unlock BQ30Z55 DJI RONIN MX 1580 BATTERY? dji not sell this battery, ronin mx is an obsolete product...
> so i have tested your software... and run well, read and write, but battery after unseal, clear pf, and seal don't charge...
> can I collaborate with you for develop a new version for unlock dji old gimbal battery? best regards
> 

Please write me via telegram, I'll try help.

davidhaluska, 2021-12-01:
Hi All, Is there possible to put battery into sleep mode - as when it comes from factory? There should be some command for that or? Thanks. David

Alessandro2023, 2022-01-16:
Good afternoon friend I would like to know if you managed to finish the dji battery killer 0.4 beta 2 if you were able to finish it could you give me the link to download it in my email please ale3dprint@hotmail.com

cptrock, 2022-02-05:
Can a FT23RL chip be integrated with your app? Both windows and linux

ivn09, 2022-02-12:
Hello, I am using raspberry to unseal mini 2 battery, but without luck..  SEC=3 anyway
[image]

ledo77, 2022-06-02:
Hello, same here
[image]

gdereck, 2022-06-15:
Same here (tried on Raspberry Pi 1 & 4). I managed to unseal and unlock my DJI Mini 2 battery using the "Arduino Crack" software from Battery Killer's support channel and an Arduino Uno. I can't tell why that one works and this doesn't.

bornsnyder, 2022-07-11:
> Same here (tried on Raspberry Pi 1 & 4). I managed to unseal and unlock my DJI Mini 2 battery using the "Arduino Battery Crack" software from Battery Killer's support channel and an Arduino Uno. I can't tell why that one works and this doesn't.

Hi gdereck. Could you please provide more information on the process with the Arduino. Maybe share a link to the tool you used? I am also having problems using the Raspberry on my Mavic 2 Enterprise Zoom Battery as it will not unseal.

I recon that Mixeysan is having other problems at the moment than helping out people on the internet with the Russians attacking his country. :( Hope this madness is over soon.

gdereck, 2022-07-13:
> Hi gdereck. Could you please provide more information on the process with the Arduino. Maybe share a link to the tool you used? I am also having problems using the Raspberry on my Mavic 2 Enterprise Zoom Battery as it will not unseal.

Mixeysan posted an invite to their support channel in #245. I joined it and downloaded software posted by Chi Minh on May 9 I think. It's a hex file you need to upload into your Arduino and an exe you run on your PC with Arduino connected. Note: it's closed source and it's hard to tell what it does under the hood. I ran it on a VDI just in case.

I've also seen that someone posted an Arduino sketch - may work as well. That support channel is a good source of information.

bornsnyder, 2022-07-13:
> > Hi gdereck. Could you please provide more information on the process with the Arduino. Maybe share a link to the tool you used? I am also having problems using the Raspberry on my Mavic 2 Enterprise Zoom Battery as it will not unseal.
> 
> Mixeysan posted an invite to their support channel in #245. I joined it and downloaded software posted by Chi on May 9 I think. It's a hex file you need to upload into your Arduino and an exe you run on your PC with Arduino connected. Note: it's closed source and it's hard to tell what it does under the hood. I ran it on a VM just in case. Use with caution, at your own risk.
> 
> I've also seen that someone posted an Arduino sketch - may work as well. That support channel is a good source of information. Good luck!

Thanks gdereck for your swift reply. Sadly it seems that all download links in the support forum are dead. I guess I'll have to buy a pricey cp2112 board and use the battery killer software :/ It's really hard sitting there with a dead enterprise edition battery.

You don't happen to still have the hex and exe files you could provide? Never the less thanks for your response. 
Best regards

gdereck, 2022-07-13:
> You don't happen to still have the hex and exe files you could provide? Never the less thanks for your response. Best regards

The download works for me (iOS Telegram client), I've just downloaded them. I don't feel comfortable distributing someone else's software but if you don't manage to get it to work, I can try.

bornsnyder, 2022-07-13:
Seems I got you wrong about the support channel. I tried the link he posted in the other topic you mentioned (4dpa forum with 160+ pages in the thread) .. I just tried to contact via the t.me link provided in the abouts of the battery killer application but I'm not feeling comfortable contacting Mixeysan directly while ukraine is attacked - me asking for help with my shitty drone battery.You get what I mean. :/ Would be much appreciated if you could help out but if not I totally understand.

gdereck, 2022-07-13:
The point is that you can join the Telegram group and see the entire conversation history. It includes files posted by Mixeysan and others. If you want to find the files I mentioned, just scroll up to May 9th and find Chi's post. Or use the search feature and look for arduino battery crack.exe. There are many people in that group and lots of information in the history.

gdereck, 2022-07-13:
@bornsnyder, you could try this: https://github.com/o-gs/dji-firmware-tools/issues/232#issuecomment-1166430849 That custom key seems consistent with the DJI Arduino source code. I haven't tried it though.

bornsnyder, 2022-07-14:
> @bornsnyder, you could try this: [#232 (comment)](https://github.com/o-gs/dji-firmware-tools/issues/232#issuecomment-1166430849) That custom key seems consistent with the DJI Arduino source code. I haven't tried it though.

Thanks but already tried that without success ..

Now I finally managed to find and join the support channel and download the exe and hex file. (I was literally blind) Now I just need to figure out the wiring :D Thanks again for your help.

Happilytrying, 2022-08-25:
Hi everyone. I tried to reset 3 spark batteries using Texas Instruments battery  BQ studio but the chips for DJI spark do not seem to be in there it’s the BQ30Z55 I need right? I tried some others like BQ27 and Bq40 but the options don’t allow to resent PF and unseal codes don’t work.

with DJI battery killer the software doesn’t work with EV2400 I’ve tried (unless there is a driver needed?). Should I get hold of cp2112 adapter, and will it work with battery killer and spark chip ok without knowing the unseal code?

or is there another software for EV2400? I see TRB pontoon lab but it’s a paid subscription 

Thanks for the help

kloppinc, 2023-11-29:
I am also unable to unseal a bq40z307 inside a mavic air battery. Anyone has the correct key?

goliady, 2024-06-27:
> I used this tool https://mega.nz/file/yEJGwJYR#O3jh40agtSWE11HRySdg5C6X5y2OeHW2c_A4hOcCVFs All work fine with CP2112 adapter

link dont work

Shan-7, 2025-01-10:
> Boom HTTP link (not HTTPS) - http://master-club.org/drone/software/
> 
> I have not finished everything yet. In particular, still need to remove the unnecessary and disable to write of FW BQ30Z55, because the CP2112 adapter will not be able to do this anyway. But the change in capacity, voltages, charge max, reset of cycles seems to be plowing both on bq30z55 and on bq9003. To get full access in BQ30Z55 when the key is unknown, use the Full Access button. There are still small shoals, with time I'll correct it. And I'll release it a little later when will be free time)

send me softwware - shantanushendre@gmail.com