o-gs/dji-firmware-tools #216: does wm161b in mini2 controller have only boot loader file??

taek2021, 2021-07-26:
hello, anyone.

i want to access the Mini2 controller's system folder

so i did decrypted wm161b's firmware from wm161b's dumpfile or DDD

but i couldn't do it completely.

does wm161b have only boot loader file??

i have ap_RTOS.bin, cp_RTOS.bin and it is encrypted.

if who did decrypted it so did access it's system folder, tell me information.
or give me some tip or file plz..

mefistotelis, 2021-07-26:
> i have ap_RTOS.bin, cp_RTOS.bin and it is encrypted.

It looks like you've extracted module m1301 fw update, and then unsigned "ap.img"/"cp.img". The resulting files like "rc-n1-wm161b_1301_v04.11.00.13_20210406.pro.fw-test_1301.ap_RTOS.bin" are not encrypted.

> i want to access the Mini2 controller's system folder

Linux filesystem update is not included in the currently known "rc-n1-wm161b" firmware update file.
Some kind of filesystem should be accessible in the flash dump you provided.

taek2021, 2021-07-26:
thanks!!

> It looks like you've extracted module m1301 fw update, and then unsigned "ap.img"/"cp.img". The resulting files like "rc-n1-wm161b_1301_v04.11.00.13_20210406.pro.fw-test_1301.ap_RTOS.bin" are not encrypted.
=> i think both of them are encrypted because lt's entropy is near of 1
 
 
> Linux filesystem update is not included in the currently known "rc-n1-wm161b" firmware update file.
> Some kind of filesystem should be accessible in the flash dump you provided.
=> okay, i will try it at RTOS.bin in flash dump file!!

mefistotelis, 2021-07-26:
> i think both of them are encrypted because lt's entropy is near of 1

Then you didn't unsigned it properly. I've got:
[image]

> However, is there any possibility that only bootloader was included in wm161b?

Second stage bootloader is included in FW update, if that's what you're asking.
The dji-firmware-tools tests extracts it as `rc-n1-wm161b_1301_v04.11.00.13_20210406.pro.fw-test_1301.loader_p0_BLLK.bin`.
Look into `tests/test_imah_all.sh` to execute the tests.

taek2021, 2021-07-27:
did you get OS file in *_RTOS.bin??
 
[image]

i got this result..

mefistotelis, 2021-07-27:
The `test_dji_imah_fwsig_rebin1.sh` script accepts a whole TAR file (whole FW package). Not a single module.

taek2021, 2021-07-27:
well.. i got same result... 
[image]

taek2021, 2021-07-27:
> The `test_dji_imah_fwsig_rebin1.sh` script accepts a whole TAR file (whole FW package). Not a single module.
> Also, use `-sn` option to make it not delete extracted files at end.

hi mefi!! i try did it on your tip.
but i got follow error..
[image]
[image]

i want to get decrypted ap_RTOS.bin , cp_RTOS.bin file.. 

i always thanks you mefi!!

mefistotelis, 2021-07-27:
> i try did it on your tip. but i got follow error..

Ok, yes. I did made a mistake - the TAR gets extracted by `test_imah_all.sh`, so the other script gets single modules.

Anyway, just read the scripts.

taek2021, 2021-07-28:
> > i try did it on your tip. but i got follow error..
> 
> Ok, yes. I did made a mistake - the TAR gets extracted by `test_imah_all.sh`, so the other script gets single modules.
> 
> Anyway, just read the scripts.

well..  i am chaos..
[image]

Is there any other way not to use the tool in tests folder?
i have arrived this step using tool(like dji_imah_fwsig... (not in tests folder)

so i want decrypt one more step file in picture..

I think there's something wrong with the test file in my environment. Sorry to bother you mefi..
thanks!!

mefistotelis, 2021-07-28:
> Is there any other way not to use the tool in tests folder?

Yes. The test is just an example.
If you don't need examples, you can do it yourself.

> when you use binwalk tool on above your decrypted *RTOS.bin , were you get a linux os system folder

As the name suggests, these files store RTOS. RTOSes do not typically have filesystems, and there is none inside.

taek2021, 2021-07-28:
> Yes. The test is just an example.
> If you don't need examples, you can do it yourself.
>
>
> As the name suggests, these files store RTOS. RTOSes do not typically have filesystems, and there is none [...]

thanks mefi!!!!!

closed 2021-07-28 by taek2021