o-gs/dji-firmware-tools #217: Reset controller settings BQ30z55

alxgacc, 2021-07-29:
Hi. I want to change the parameters of the controller BQ30z55 battery ji mavic pro. I want to change the FullChargeCapacity and CycleCount parameters, but I cannot figure out which commands to do it

mefistotelis, 2021-07-29:
Implement the missing conversion, then you can use it.

alxgacc, 2021-07-29:
When i make a request i get this error:

pi@raspberrypi:/dji-firmware-tools-master $ python3.7 ./comm_sbs_bqctrl.py --short --dev_address 0x0b write RemainingCapacity 1900
Error: NotImplementedError: Converting the string value to cmd/subcmd type is not implemented.
Traceback (most recent call last):
  File "./comm_sbs_bqctrl.py", line 3681, in <module>
    main()
  File "./comm_sbs_bqctrl.py", line 3652, in main
    smart_battery_system_write(po.command, po.newvalue, vals, po)
  File "./comm_sbs_bqctrl.py", line 3056, in smart_battery_system_write
    v = smart_battery_system_cmd_value_from_text(cmd, subcmd, nval_str, po)
  File "./comm_sbs_bqctrl.py", line 2892, in smart_battery_system_cmd_value_from_text
    raise NotImplementedError('Converting the string value to cmd/subcmd type is not implemented.')
NotImplementedError: Converting the string value to cmd/subcmd type is not implemented.

mefistotelis, 2021-07-29:
Yeah. Again, implement the conversion - sending is done properly, but I never used 'write' from command line, so didn't made the conversion.

alxgacc, 2021-07-29:
I do not quite understand how implement the conversion, can you tell me, please?

mefistotelis, 2021-07-29:
I'm not working on this tool ATM.

The function which raises the exception gets parameters as strings. Based on type of the parameter - the one you want to set is integer number - it should convert that type into format which is accepted by the sender function - can't remember what exactly that is, integers or array of bytes; read the code for details.

The tool is just a wrapper for sending SMBus packets; these packets are simple, so understanding the whole flow isn't that hard.