o-gs/dji-firmware-tools #218: Exit boot rom mode Mavic mini battery

mjnhchj, 2021-07-30:
How to exit boot rom mode with comm_sbs_bqctrl.py in Raspberry Pi command? Thank you!

mefistotelis, 2021-07-30:
The tool does hint you on how to do it.
```
# ./comm_sbs_bqctrl.py -c BQ40z307 info ManufacturerAccess.ROMMode
Command/offset:
  ManufacturerAccess.ROMMode = 0x00.0x0f00
Access:
  Sealed '-'; Unsealed  '-'; Full Access  'W'
  Access by first writing sub-command word to ManufacturerAccess
Type:
  There is no value field in read/Write operations, it works as trigger
  The value has no unit or printing format preference
Description:
  Enables the ROM mode for IF update. On this command, device goes to ROM mode ready for re-programming firmware in Instruction Flash. Thit is often called BootROM mode. Use 0x08 to ManufacturerAccess() to return.
```

mjnhchj, 2021-07-30:
Sorry. I don't understand that command well.  Can you give me an illustrative example?  how to use "0x08 to ManufacturerAccesss".

mefistotelis, 2021-07-30:
The tool does not currently support bootrom commands. I just didn't had a need for them.

So this is how you trigger command 8 of MA:
```
./comm_sbs_bqctrl.py -c BQ40z307 trigger ManufacturerAccess.0x08
```
But it won't work out-of-the-box, because the tool will recognize that it knows command 0x08, and that command requires an argument (the tool will assume this is "normal mode" command.

So to make that work, you must edit the command within `comm_sbs_chips/BQ40z307.py` and make all its properties ('type', 'unit' etc.) identical to those from `MANUFACTURER_ACCESS_CMD_BQ40.ROMMode`, so that this command is treated as a trigger command.

If you'd like the tool to support boot mode commands properly - patches are welcome.

mjnhchj, 2021-07-30:
Thank you so much!