o-gs/dji-firmware-tools #219: Phantom4 battery change parameters

alives2000, 2021-07-31:
Who can tell me how I can update the FCC parameter on the phantom 4 battery, the bq30z55 chip, and can i somehow reset the charge counter and the production date?

alives2000, 2021-08-02:
I have already connected to the battery, I was able to read the information, reset the PF errors, but I cannot understand which commands should be used to change the battery parameters

mefistotelis, 2021-08-02:
Read here:
https://github.com/o-gs/dji-firmware-tools/issues/217

alives2000, 2021-08-02:
after executing the commands, now on the i2cdetect -y 1 command this is the output:

0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
10: 10 11 12 13 14 15 16 17 18 19 1a UU 1c 1d 1e 1f
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
70: 70 71 72 73 74 75 76 77

rebooting the raspberry does not help