o-gs/dji-firmware-tools #221: inspire 1 Pro update failed.Now cannt read firmware

kenhoung3553, 2021-08-08:
For this Inspire 1 Pro, I helped my friend write the parameters and write the firmware at the end. He saw that the gimbal was shaking and turned off the power. Oh, my God! Now it keeps calling when it is turned on. But it doesn’t write. The tail light does not turn on. .This platform has two gimbals, X5 and X3. Which one should be used to read? What version should be used to read. #

Result: Success.

========== 2014.01.01 00:00:06 remo-con disconnect======
Packet: WM610_FW_V01.06.00.40.bin
Result: Abort.
The firmware on the SD card is identical to or older than the current firmware on the aircraft.

========== 2016.01.14 18:15:09 =====================
Packet: WM610_FW_V01.06.00.40.bin
Result: Abort.
The firmware on the SD card is identical to or older than the current firmware on the aircraft.

========== 2016.01.15 11:20:46 =====================
Packet: WM610_FW_V01.06.00.40.bin
Result: Abort.
The firmware on the SD card is identical to or older than the current firmware on the aircraft.

========== 2014.01.01 00:00:06 remo-con disconnect======
Packet: WM610_FW_V01.06.00.40.bin
Result: Abort.
The firmware on the SD card is identical to or older than the current firmware on the aircraft.

========== 2014.01.01 00:00:06 remo-con disconnect======
Packet: WM610_FW_V01.06.00.40.bin
Result: Abort.
The firmware on the SD card is identical to or older than the current firmware on the aircraft.

========== 2014.01.01 00:00:06 remo-con disconnect======
Packet: WM610_FW_V01.06.00.40.bin
Result: Abort.
The firmware on the SD card is identical to or older than the current firmware on the aircraft.

========== 2014.01.01 00:00:06 remo-con disconnect======
Packet: WM610_FW_V01.06.00.40.bin
Result: Abort.
The firmware on the SD card is identical to or older than the current firmware on the aircraft.

========== 2016.01.16 09:50:47 =====================
Packet: WM610_FW_V01.06.00.40.bin
Result: Abort.
The firmware on the SD card is identical to or older than the current firmware on the aircraft.

========== 2016.01.16 09:54:52 =====================
Packet: WM610_FW_V01.06.00.40.bin
Result: Abort.
The firmware on the SD card is identical to or older than the current firmware on the aircraft.

========== 2016.01.16 09:58:51 =====================
Packet: WM610_FW_V01.06.00.40.bin
Result: Abort.
The firmware on the SD card is identical to or older than the current firmware on the aircraft.

========== 2016.01.19 09:06:40 =====================
Packet: WM610_FW_V01.06.00.40.bin
Result: Abort.
The firmware on the SD card is identical to or older than the current firmware on the aircraft.

========== 2016.01.19 09:39:03 =====================
Packet: WM610_FW_V01.06.00.40.bin
Result: Abort.
The firmware on the SD card is identical to or older than the current firmware on the aircraft.

========== 2016.01.19 09:41:07 =====================
Packet: WM610_FW_V01.06.00.40.bin
Result: Abort.
The firmware on the SD card is identical to or older than the current firmware on the aircraft.

========== 2016.01.19 10:15:58 =====================
[... 167 lines cut ...]
Packet: WM610_FW_V01.12.01.54.bin
Upgrading ...
Result: Success.

========== 2020.12.03 17:05:53 =====================
Packet: WM610_FW_V01.12.01.54.bin
Result: Abort.
The firmware on the SD card is identical to or older than the current firmware on the device.

========== 2014.01.01 00:00:06 remo-con disconnect======
Packet: WM610_FW_V01.11.01.50.bin
Result: Corrupted.
Firmware bin file is corrupted, download it again from the DJI offical website.

========== 2014.01.01 00:00:06 remo-con disconnect======
Packet: WM610_FW_V01.11.01.50.bin

========== 2021.08.02 12:10:41 =====================
Packet: WM610_FW_V01.11.01.50.bin

========== 2014.01.01 00:00:06 remo-con disconnect======
Packet: WM610_FW_V01.11.01.50.bin

========== 2021.08.02 17:32:17 =====================
Packet: WM610_FW_V01.11.01.50.bin

mefistotelis, 2021-08-12:
It isn't easy to understand you.

Anyway, you've probably half-updated one of the modules, making it impossible to boot.
You need to figure out which module you've damaged. For that, you need "LOG_AB". Google it. There are entry-level instructions on how to get LOG_AB on phantompilots forum.

What you inserted above is "RESULT_AB". It is useless.

Which gimbal, or anything else - how can you imagine a situation where that would even matter?
Figure out which module is damaged. That's what matters.

kenhoung3553, 2021-08-13:
> It isn't easy to understand you.
>
> Anyway, you've probably half-updated one of the modules, making it impossible to boot.
> You need to figure out which module you've damaged. For that, you need [...]

Thank you. ill try

arildo-yank, 2024-01-28:
Hi I got the same issue after the upgrade, would please get the solution by chance? thank youl