o-gs/dji-firmware-tools #223: P3P [0800, 1500] Module Problem

JGabT, 2021-08-16:
Looks like he had V01.03.0020 and upgraded to V01.04.0010. During this upgrade, some modules were left intact. This includes:
<pre> [00013627][15 00][00] v1.1.2.0 -> v1.1.2.0
 [00014004][08 00][00] v0.12.0.4 -> v0.12.0.4</pre>
This is where the issue started - he probably force-upgraded instead of force-downgrading.

Since then, the devices are "not detected":
<pre> [00028638][15 00][00] device not detected.
 [00041488][08 00][00] device not detected.</pre>

This means those two devices are still at firmware from V01.03.0020. To make them work, he needs to get the rest of the drone back to V01.03.0020 too. And I mean all components - they should all be at this version:
<pre> [03 05][00] v34.2.0.8
 [03 06][00] v2.1.6.18
 [04 00][00] v1.40.0.0
 [09 00][00] v1.7.0.3
 [11 00][00] v1.7.0.0
 [12 00][00] v1.8.0.0
 [12 01][00] v1.8.0.0
 [12 02][00] v1.8.0.0
 [12 03][00] v1.8.0.0
 [15 00][00] v1.1.2.0
 [17 00][00] v1.1.1.2
 [17 01][00] v1.0.2.4
 [19 00][00] v1.0.8.59
 [01 00][00] v1.23.3419
 [01 01][00] v1.23.3419
 [08 00][00] v0.12.0.4
</pre>

Now since he seem to have panicked and started trying every version he had, this will not be as simple as installing one downgrade. He should downgrade slowly, version by version. After he reaches V01.03.0020, he should post the log - we will compare his current versions with the list above to check which firmwares are not in its original state.

I think trying to up/downgrade the problematic modules will do nothing - the rest of the drone cannot communicate with them, so it cannot send them a new firmware. But I'm not 100% sure if it works this way.

_Originally posted by @mefistotelis in https://github.com/o-gs/dji-firmware-tools/issues/4#issuecomment-274350511_

JGabT, 2021-08-16:
So i will try downgrading my P3P with the m0800 and m1500 module problem one by one from the latest firmware v1.11.20 to v1.01.8. This drone was given to me. The last working firmware version is unknown. Currently i have flashed v1.11.20 to v1.09.6 with no success. Right now i am trying versions v1.08.8 all the way to v1.01.8 then after that upgrading to the latest v1.11.20. I am saving all logs for reference and any errors. If this is successful then we can confirm that this method works with the m0800, m1500 problem

JGabT, 2021-08-16:
I am downgrading with P3X_FW_DEBUG file.

JGabT, 2021-08-16:
01.10.0090 log attached is the hidden log and the result_ab.
[P3X_FW_RESULT_AB.txt](https://github.com/o-gs/dji-firmware-tools/files/6991373/P3X_FW_RESULT_AB.txt)
[P3X_FW_LOG_AB.txt](https://github.com/o-gs/dji-firmware-tools/files/6991376/P3X_FW_LOG_AB.txt)

JGabT, 2021-08-16:
01.09.0060 log attached is the hidden log and the result_ab.
[P3X_FW_RESULT_AB.txt](https://github.com/o-gs/dji-firmware-tools/files/6991378/P3X_FW_RESULT_AB.txt)
[P3X_FW_LOG_AB.txt](https://github.com/o-gs/dji-firmware-tools/files/6991379/P3X_FW_LOG_AB.txt)

JGabT, 2021-08-16:
01.08.0080 log attached is the hidden log and the result_ab. Failed.
[P3X_FW_LOG_AB.txt](https://github.com/o-gs/dji-firmware-tools/files/6991450/P3X_FW_LOG_AB.txt)

[P3X_FW_RESULT_AB.txt](https://github.com/o-gs/dji-firmware-tools/files/6991449/P3X_FW_RESULT_AB.txt)

mefistotelis, 2021-08-16:
You've cited my post from over 4 years ago. It was speculative. Since then, we've got full understanding of the m0800/m1500, and how the modules work in general.

Read the project wiki for details.

JGabT, 2021-08-16:
Hmm, seems that the hyperlink is not working on the "By Chip Maker Method" (https://github.com/o-gs/dji-firmware-tools/wiki/Firmware-m0800#by-chip-maker-method) would you mind replacing it?

JGabT, 2021-08-16:
Oh seems that the processors.wiki.ti.com is now end-of-life. But i found an internet archive snapshot (https://web.archive.org/web/20200615014317/https://processors.wiki.ti.com/index.php/Writing_Kernel_Image_to_NAND_Flash)

mefistotelis, 2021-08-16:
You may want to look at service serial port method instead.

JGabT, 2021-08-16:
> 
> 
> You've cited my post from over 4 years ago. It was speculative. Since then, we've got full understanding of the m0800/m1500, and how the modules work in general.
> 
> Downgrading may be required, but only in some rare cases. The module m0800 just needs to be fixed.
> 
> Read the project wiki for details.

"Downgrading may be required" Since it always fails to complete the firmware upgrade is there any chance of success by downgrading one by one to the oldest version using P3X_FW_DEBUG file then upgrading one by one without the P3X_FW_DEBUG?

mefistotelis, 2021-08-16:
> it always fails to complete the firmware upgrade

Stop thinking about firmware as one thing. It's a package.

Without serial port - don't know, I only used the serial port method.

JGabT, 2021-08-16:
> 
> 
> > it always fails to complete the firmware upgrade
> 
> Stop thinking about firmware as one thing. It's a package.
> 
> Without serial port - don't know, I only used the serial port method.

Is there any guide that you can share?

JGabT, 2021-08-16:
> 
> 
> Everything is on the wiki.

Alrighty. Connect my 3.3V TTL converter to service pads. which is 368_U0_RXP, 368_U0_TXP and GND for the P3X

I was really hoping for an easy way such as flashing it with a bin file but hopefully in the future. For now we can solely rely on the Serial Port method. I will report back here if the serial port flashing is successful.

JGabT, 2021-08-16:
Since this problem only come from the m0800 module i shouldn't need to touch/repair the m1500 module since if i fix the m0800 module it also fixes the m1500 right?

mefistotelis, 2021-08-16:
Yes.

JGabT, 2021-08-16:
Do you have any recommended guides on how to disassemble the Gimbal Board of P3X?

mefistotelis, 2021-08-16:
No. General youtube.

JGabT, 2026-02-07:
I fixed the DM365 chip by flashing using FTDI and now I have this problem with the 1900 module.. Any fixes for this?

JGabT, 2026-02-07:
[P3X_FW_LOG_AB.txt](https://github.com/user-attachments/files/25154901/P3X_FW_LOG_AB.txt)
[P3X_FW_RESULT_AB.txt](https://github.com/user-attachments/files/25154907/P3X_FW_RESULT_AB.txt)

Attached the logs and I tried flashing the firmware I tried last 5 years ago with the m0800 problem (P3X_FW_V01.08.0080.bin)

JGabT, 2026-02-08:
I’ve already performed the serial flash (sfh.exe) to restore the low-level bootloader and NAND. 
The DM36x is now out of the 'BOOTME' loop.

 I attempted SD card updates with several versions (v1.8.80, v1.9.60, v1.11.20) using the P3X_FW_DEBUG file.

Every attempt crashes at the m1900 module with the same error: [19 00] Firmware upgrade finish failed (step = 2, err = 0x08).

The drone beeps 'DD-DD' (Inconsistency) and the Remote Controller remains Solid Red. I've attempted a manual Link (C1+C2+Rec), but the RC flashes Blue indefinitely and never handshakes with the AC.

In my 2021 logs, module [19 00] was detected as v1.0.8.96. In my current logs, it shows as v0.0.0.0.

Since it now reports v0.0.0.0 and times out with err 0x08, does this confirm a physical BGA failure on the FPGA/Lightbridge chip, or is there a way to force the m1900 handshake after a serial reconstruct? Is there any way to skip the m1900 check entirely to allow the drone to arm?

mefistotelis, 2026-02-08:
> In my 2021 logs, module [19 00] was detected as v1.0.8.96. In my current logs, it shows as v0.0.0.0.

Then I guess you have a HW revision with FPGA onboard.

> does this confirm a physical BGA failure on the FPGA/Lightbridge chip

FPGA chip has no internal flash. The module description I linked explains that.
The FPGA isn't also in control of its bitstream content, MCU on the OFDM board handles writing it.
Check the flash chip.

Regardless of details, the issue seem to be with OFDM board, or its connection to other boards.