o-gs/dji-firmware-tools #227: Read dataflash from battery error

alxgacc, 2021-09-14:
/dji-firmware-tools-master $  ./comm_sbs_bqctrl.py --dev_address 0x0b raw-backup DataFlash /tmp/fw.bin
Error: OSError: [Errno 121] Remote I/O error
Traceback (most recent call last):
  File "./comm_sbs_bqctrl.py", line 3681, in <module>
    main()
  File "./comm_sbs_bqctrl.py", line 3660, in main
    smart_battery_system_raw_backup(po.addrspace, po.fname, vals, po)
  File "./comm_sbs_bqctrl.py", line 3159, in smart_battery_system_raw_backup
    v, l, u, s = smbus_read(bus, po.dev_address, cmd, opts, vals, po)
  File "./comm_sbs_bqctrl.py", line 2602, in smbus_read
    v, u = smbus_read_by_writing_word_subcmd_first(bus, dev_addr, cmd, subcmd, subcmdinf, po)
  File "./comm_sbs_bqctrl.py", line 2428, in smbus_read_by_writing_word_subcmd_first
    v = smbus_read_block_val_by_writing_word_subcmd_first(bus, dev_addr, cmd, subcmd, resp_type, resp_cmd, resp_wait, po)
  File "./comm_sbs_bqctrl.py", line 1897, in smbus_read_block_val_by_writing_word_subcmd_first
    b = smbus_read_raw_block_by_writing_word_subcmd(bus, dev_addr, cmd, subcmd, resp_type, resp_cmd, resp_wait, po)
  File "./comm_sbs_bqctrl.py", line 1762, in smbus_read_raw_block_by_writing_word_subcmd
    v = smbus_read_block_for_basecmd(bus, dev_addr, resp_cmd, basecmd_name, resp_type, po)
  File "./comm_sbs_bqctrl.py", line 1600, in smbus_read_block_for_basecmd
    bus.i2c_rdwr(part_write, part_read)
  File "/usr/local/lib/python3.7/dist-packages/smbus2/smbus2.py", line 658, in i2c_rdwr
    ioctl(self.fd, I2C_RDWR, ioctl_data)
OSError: [Errno 121] Remote I/O error

maxdd, 2021-09-22:
That's more a connection error in my opinion, can you show i2cdetect -l command?
Are you on raspberry?

mefistotelis, 2021-09-22:
It could also be coding error. I only tested that function once, and it was on Mav Mini battery.

alives2000, 2021-09-22:
> 
> 
> That's more a HW connection error in my opinion, can you show i2cdetect -l command?
> Are you on raspberry?

Yes, I do it on raspberry. I do not think that this is a connection problem, because the script executes other options without errors.