o-gs/dji-firmware-tools #229: I have a DJI FPV battery. Ic name BQ9006 . Can it work?

doquangquan93, 2021-10-08:
[image]
[image]

mjnhchj, 2021-10-08:
Hiện tại chỉ có các phiên bản phần mềm trả phí và công cụ đắt tiền mới có thể unlock được. Bạn có thể sử dụng DJI battery killer với CP2112 adaptor,  nhưng tôi không chắc nó có thể unlock.

mixeysan, 2021-10-19:
> I find pinout i2c battery.
> I read info battery oke . But it not function unseal :(((
> [image]
> [image]
> 
> 

Hi. I wrote you - "dji bat Killer" bq9006 not supported yet.
I have not yet had this chip in my hands.
if you have algorithms for working with this chip, then you can work with it manually using commands and cp2112

doquangquan93, 2021-10-19:
I understand . I just want to share a bit of information for anyone looking for scl and sda.

WP-Patrick, 2021-11-01:
The best is going to be to isolate the STM32 chip from BQ9006, otherwise, STM32 might interfere with EV adapter and BQ9006.
Also the STM32 likes to talk even if device is in BOOTMODE :( 

You can reset the lock of BMS from a rooted drone.

[image]

doquangquan93, 2021-11-02:
> The best is going to be to isolate the STM32 chip from BQ9006, otherwise, STM32 might interfere with EV adapter and BQ9006. Also the STM32 likes to talk even if device is in BOOTMODE :(
> 
> You can reset the lock of BMS from a rooted drone.
> 
> [image]

Where can I find root information.
i see https://drone-hacks.com but not about for reset the lock of BMS

WP-Patrick, 2021-11-02:
Yeah, the information is not yet public, sorry.

doquangquan93, 2021-12-16:
> Yeah, the information is not yet public, sorry.

Hello WP-Patrick, Can you share info rooted drone ?

WP-Patrick, 2021-12-16:
I'm not going to help MIXEYSAN by putting reset PF3 block public. Sorry.

cecca00, 2022-04-27:
> I find pinout i2c battery. I read info battery oke . But it not function unseal :((( [image] [image]

Thanks for the pinout, any update on this? Have you been able to unlock the battery?

mixeysan, 2022-12-13:
[BQ9006_DJi_FPV_37WDJ3BCR21124_backup_BatteryKiller.7z.zip](https://github.com/o-gs/dji-firmware-tools/files/10218954/BQ9006_DJi_FPV_37WDJ3BCR21124_backup_BatteryKiller.7z.zip)

WP-Patrick, 2022-12-16:
> [BQ9006_DJi_FPV_37WDJ3BCR21124_backup_BatteryKiller.7z.zip](https://github.com/o-gs/dji-firmware-tools/files/10218954/BQ9006_DJi_FPV_37WDJ3BCR21124_backup_BatteryKiller.7z.zip)
> 
> Probably soon there will be an update with interesting features ;)

Good job Michael, I've patched the communication on STM level so it doesn't return PF flag.

mixeysan, 2022-12-17:
> Good job Michael, I've patched the communication on STM level so it doesn't return PF flag.

It's not a problem for me.

wang909, 2023-02-10:
@mixeysan I have a DJI T30 battery with only STM32L071CZT6 chip on it    How to unlock the battery
[image]

aytacengin, 2023-04-09:
> [BQ9006_DJi_FPV_37WDJ3BCR21124_backup_BatteryKiller.7z.zip](https://github.com/o-gs/dji-firmware-tools/files/10218954/BQ9006_DJi_FPV_37WDJ3BCR21124_backup_BatteryKiller.7z.zip)
> Probably soon there will be an update with interesting features ;)

Is there any update brother? This is very important for us. We can not use battery. I have 10 More pieces battery. :(

vedmant, 2023-08-06:
Hi guy, any update on this, my battery just got locked, would be nice if I can restore it.

neich03, 2023-10-24:
> [BQ9006_DJi_FPV_37WDJ3BCR21124_backup_BatteryKiller.7z.zip](https://github.com/o-gs/dji-firmware-tools/files/10218954/BQ9006_DJi_FPV_37WDJ3BCR21124_backup_BatteryKiller.7z.zip) Probably soon there will be an update with interesting features ;)

You´re a hero dude! is there any place where we can make donations to thank your job?

borisbl4, 2024-01-14:
> The best is going to be to isolate the STM32 chip from BQ9006, otherwise, STM32 might interfere with EV adapter and BQ9006.
> Also the STM32 likes to talk even if device is in BOOTMODE :( 
> 
> You can reset the lock of BMS from a rooted drone.
> 
> [image]
> 
> 

Are there any paid solutions to the problem today?

AndKe, 2024-04-13:
What is the situation now? 
How is the BQ9006_DJi_FPV_37WDJ3BCR21124_backup_BatteryKiller files supposed to be used?
What is needed to clear PF from a DJI FPV battery? 
And what's with the rooted DJI FPV that can do this?

uavquadfpv, 2024-04-20:
To date, is there any solution to prove that the smart battery of this DJI FPV can be unlocked?

chimera15, 2024-05-07:
Same issues, as these drones get older this will be necessary.   what are the bin files?

werty1st, 2024-05-11:
Off topic: Its very frustrating that i can't use my fully working battery due to this software problem :(
Any ideas if the battery could used without clearing the flags?
How could a locked battery power a rooted drone to be unlocked via the drone?

AndKe, 2024-05-12:
@werty1st  - no the battery is not going to be usable with those flags. The whole point of making those flags set once, and not reset if conditions are good, is to make you buy new. 
You can take a new battery, disconnect and reconnect the BMS, and it will fail permanently(until reset) as if a cell was below some threshold.

Why is this community not open about rooting and such? Is is it just about money?

chimera15, 2024-05-12:
Almost every other dji battery has been hacked to allow the battery to be fixable, the dji fpv battery apparently has a really interesting way to go about it with the stm32 chip talking to the bms.  It's a really interesting hack, it will be done, a lot of people are working on it now and a hacker from slovakia has done it 2 years ago, but has not open sourced it, he has the company Flylusive, so it's doable, it just takes some knowledge.

chimera15, 2024-05-23:
> Use UBRT.

he speaks omg! ok I'll look it up...does it use stm32 or cp2112?  I'll check.

werty1st, 2024-05-23:
This is the closest I could get so far.
http://ubrt.com.ua/14-news/79-new-version-ubrt-7-1-30-00.html

But I can't find a download. And I still don't know if it can be used with a cheap adapter.

chimera15, 2024-05-23:
Ah ok so this uses ev2300/2400 right? Do you have to pay them for using their program as well like that other program that uses that?

AndKe, 2024-05-25:
This is a job for a logic analyzer that can capture the traffic, I can do that, anyone with a license please borrow it to me, then we'll have a open solution. :)

chimera15, 2024-05-25:
> Of course you have to pay. Yes, they're trusted - we partner with them.
> 
> And no, we didn't open source BQ9006 solution and are not planning to.
> 
> You will pay around 200€ for the license, and our price would have been around the same.

So if I pay them that, I don't see anywhere on their site where it specifically mentions the bq9006 and the fpv battery...do they give me a walkthrough of what I have to do..and then that board can no longer have the flag set again? Or is there a chance it could do it again?

AndKe, 2024-05-26:
> > This is a job for a logic analyzer that can capture the traffic, I can do that, anyone with a license please borrow it to me, then we'll have a open solution. :)
> 
> That would work if we haven't used a SHA-256 to unseal the battery. Sniffing is simply not enough - you cannot replicate it, as each time the battery gives different challenge.

And the private part of that key is in the STM or in the firmware of the DJI FPV?

chimera15, 2024-05-26:
> > > > This is a job for a logic analyzer that can capture the traffic, I can do that, anyone with a license please borrow it to me, then we'll have a open solution. :)
> > > 
> > > 
> > > That would work if we haven't used a SHA-256 to unseal the battery. Sniffing is simply not enough - you cannot replicate it, as each time the battery gives different challenge.
> > 
> > 
> > And the private part of that key is in the STM32 or in the firmware of the DJI FPV?
> 
> BQ9006 BRL mode. (and no, you cannot read it without patching the loader)
> 
> STM32 does not have anything to do with this, it literally just forwards commands. One can just use STM32 to communicate with BQ9006 directly.

Why does the 2300 work and not the stmlink or cp? I was communicating with the board with the cp but I couldn't seem to get it to communicate with the stm, but maybe my solders were bad or something.