o-gs/dji-firmware-tools #237: remove all errors after replacing DJI gimbal camera assembly? What is his software name?

mjnhchj, 2021-11-11:
[](https://www.youtube.com/watch?v=nTgwv_avEhk&t=48s&ab_channel=JCDRONES) in the end of his video!

mjnhchj, 2021-11-11:
[image]

mjnhchj, 2021-11-11:
[image]

954426470, 2021-11-18:
I can

mjnhchj, 2021-11-18:
> I can

Please say software name? Thank you

mEfis-afk, 2021-11-27:
> eu posso

Contact?

954426470, 2021-11-27:
？

mEfis-afk, 2021-11-27:
> ？
> […](#)

Your contact?

954426470, 2021-11-27:
> Your contact?

My mailbox 954426470@qq.com
My WeChat +86 17353342224

mjnhchj, 2021-11-27:
My drone: Mavic Air has an error: code 3 (camera) and cant active device :( can you help me ...

954426470, 2021-11-27:
Sorry, I can't do this

mEfis-afk, 2021-11-27:
> My drone: Mavic Air has an error: code 3 (camera) and i cant active this device :( can you help me ...

Is it not activating?

mjnhchj, 2021-11-27:
> 

Yes, cant active

mEfis-afk, 2021-11-27:
> > 
> 
> Yes, cant active

Error when activating?

mjnhchj, 2021-11-27:
> > > 
> > 
> > 
> > Yes, cant active
> 
> Error when activating?

Yes

mEfis-afk, 2021-12-09:
> My drone: Mavic Air has an error: code 3 (camera) and i cant active this device :( can you help me ...

I can solve your problem. Leave your WhatsApp

Clearvisionsenthil, 2021-12-16:
> I can

I am interested
WhatsApp +919842193180

linoko, 2021-12-30:
do may this error be got away, help

atulakbari, 2022-10-23:
Please say software name

atulakbari, 2022-10-23:
I am interested whatsapp 9925611770

Clearvisionsenthil, 2022-10-25:
No sir, I don't know the software.

954426470, 2022-10-26:
I can eliminate the 40011 error with DH
I can eliminate the 40011 error with DHI can eliminate the 40011 error with DH

954426470
***@***.***

954426470, 2022-10-26:
Please give me your whatapps QR code

954426470
***@***.***

Clearvisionsenthil, 2022-10-26:
+919842193180   whatsApp no