o-gs/dji-firmware-tools #242: Mavic Air 2 Gimbal Calibration ERROR: INVALID Choice

tonikolic, 2021-11-22:
Can someone help me to fix this to can calibrate gimbal
**argument product: invalid choice:**

I tried on another 2PC, different com number but the same problem
ty.

C:\DJI\dji-firmware-tools-master>py comm_og_service_tool.py com5 WM231 GimbalCalib JointCoarse
usage: comm_og_service_tool.py [-h] (--port PORT | --bulk) [-b BAUDRATE] [-w TIMEOUT] [--dry-test] [-v] [--version]
                               product command ...
comm_og_service_tool.py: error: argument product: invalid choice: 'COM5' (choose from 'A2', 'P330', 'P330V', 'P330Z', 'P330VP', 'WM610', 'P3X', 'P3S', 'MAT100', 'P3C', 'MG1', 'WM325', 'WM330', 'MAT600', 'WM220', 'WM620', 'WM331', 'MAT200', 'MG1S', 'WM332', 'WM100', 'WM230', 'WM335', 'WM240', 'WM245', 'WM246', 'WM160', 'WM231', 'WM232', 'WM260')

mefistotelis, 2021-11-28:
You have the help info on the screen and you still can't understand it?

To get even more help on screen, use "--help" param.
This will show you the help info in different form, hopefully allowing you to understand it.

closed 2021-11-28 by mefistotelis

tonyintn, 2021-12-28:
I am also having this error but for a Mavic Mini using the command:
py comm_og_sevice_tool.py com6 WM160 GimbalCalib JointCoarse I have views many videos and others seem to use this command with no issues I can only conclude that maybe the py command has been changed a bit.  I have looked at the help file and tired it a couple different ways but can not get it to run.

Please help