o-gs/dji-firmware-tools #243: BQ30z55 on Ronin M Battery not unsealing

Blackbird12621, 2021-11-25:
I have a Ronin M battery on which I would like to reset the PF flags. 

Connecting and reading was a breeze. Thanks for the amazing work. 

Now I can't get the BQ30z55 to unseal. I guess they used a different key? Is there any chance you can help?

Thanks

mixeysan, 2021-11-25:
Hi, for bq30z55 unseal and fas use DJi battery killer program. But be careful, and make backups.

mefistotelis, 2021-11-27:
You didn't provided much details on your setup. If you're using RPi, try affecting impedance by the command explained on this video:
https://www.youtube.com/watch?v=P5PNOO2GebY

tommyponny, 2022-02-24:
Hi,

I am also trying to get my DJI Ronin M battery running again. I am using a Raspberry Pi 3b+ to do this. 
unfortunately when I read the addresses I only see 0x40. Can someone help me solve the problem? 

thank you very much

tommyponny, 2022-02-24:
Hi, finally i was able to read the Chip. Now i am Stuck on the Same Point. I cant unseal the Chip. The unseal Parameter doesnt Change to 2. Maybe the Chip News another Password? Can anyone Help please?

mefistotelis, 2022-02-24:
Some tools have a database of passwords, and there are some which in many cases can even do what is needed without it.
The one I made doesn't have that.

So, try another tool. Ie. the one [mixeysan](https://github.com/mixeysan)  created.

tommyponny, 2022-02-24:
@mefistotelis thank you dir the fast reply.
I have ordered a cp2112 wich hopefully works with the tool from @mixeysan . The device will arrive on monday. Do you have any experience with the tool? Have a nice day :)
[image]

mefistotelis, 2022-02-24:
No.

chemaam, 2022-07-12:
@tommyponny have you had good results by using the cp2112 USB board? I'm thinking of using it with battery killer but it says that jus support bq9003 (bq40z307) chips

tommyponny, 2022-07-12:
@chemaam yes ist works. but never use the wrote Firmware/EPROM function. It destroys 2 of my batteries. First you have to unseal the Chip with the battery Killer Tool. Then you can follows the steps with the Raspberry Pi. The battery Killer Tool wont Change the Lock flag. This musst bei Döner via the Raspberry Pi. Regards

vinny-ferrari, 2023-08-17:
> @chemaam yes it works. but never use the write Firmware/EPROM function. It destroys 2 of my batteries. First you have to unseal the Chip with the battery Killer Tool. Then you can follows the steps with the Raspberry Pi. The battery Killer Tool wont Change the Lock flag. This must be done via the Raspberry Pi. Regards

Hi @tommyponny I can read the Ronin battery using the DJI Battery Killer  with the cp2112 and also using the raspberry, the unseal says it works, but the status remains as 3(Sealed) any tips?

...

Opening i2c:1
Raw write: DATA=22 3e
Raw write: DATA=20 3e
Raw write: DATA=22 3e
Raw FirmwareVersion response: 0d 05 50 00 36 00 34 00 03 80 00 01 00 03 c5
Auto-selected chip: BQ30z55, Texas Instruments BQ30z55 chip
Importing comm_sbs_chips/BQ30z554.py
Write ManufacturerAccess: CMD=00 WORD=31 00
Raw ManufacturerAccess.UnSealDevice response: 14 b4 28 f0 57 e1 7f 9a f0 54 e0 7f 9a c8 57 ad a9 ef 55 e0 7e 11
Prepared ManufacturerAccess.UnSealDevice B1=0123456789abcdeffedcba98765432107ee055efa9ad57c89a7fe054f09a7fe157f028b4
Computed ManufacturerAccess.UnSealDevice HMAC1=54ae410e5f0632cc1a9029ae858bd171b3912b0e
Computed ManufacturerAccess.UnSealDevice HMAC2=73c15120c681833622874a8b3da13af547030f7e
Write ManufacturerAccess.UnSealDevice: CMD=2f BLOCK=7e 0f 03 47 f5 3a a1 3d 8b 4a 87 22 36 83 81 c6 20 51 c1 73
Reading write_word_subcommand command at addr=0xb, cmd=0x0, type=uint16, opts={'subcmd': <MANUFACTURER_ACCESS_CMD_BQ30.OperationStatus: 84>}
Query ManufacturerAccess.OperationStatus: 00 WORD=0x54
Write ManufacturerAccess: CMD=00 WORD=54 00
Raw ManufacturerAccess.OperationStatus response: 04 00 73 10 00 d9
...

 SECURITY_MODE:  3=Sealed        [SEC]   Security Mode

negis02, 2023-08-19:
hi, for unseal you need to use newer version of DJI_Battery_Killer_ver.0.4beta_26.09.2021: https://github.com/o-gs/dji-firmware-tools/issues/214
with USB to I2C atater: https://diyprojectslab.com/fix-dji-battery-with-dji-battery-killer/
and for reset flags on rpi:
sudo python3 comm_sbs_bqctrl.py -vvv --dev_address 0x0b --chip BQ30z55 trigger ManufacturerAccess.PermanentFailDataReset

this all worked for me on DJI Ronin M Battery

vinny-ferrari, 2023-08-21:
> hi, for unseal you need to use newer version of DJI_Battery_Killer_ver.0.4beta_26.09.2021: #214 with USB to I2C atater: https://diyprojectslab.com/fix-dji-battery-with-dji-battery-killer/ and for reset flags on rpi: `sudo python3 comm_sbs_bqctrl.py -vvv --dev_address 0x0b --chip BQ30z55 trigger ManufacturerAccess.PermanentFailDataReset`
> 
> to chek falgs: `sudo python3 comm_sbs_bqctrl.py -v --bus "smbus:1" --dev_address 0x0b --chip AUTO --short monitor BQStatusBitsMA `
> 
> this all worked for me on DJI Ronin M Battery

Hi @negis02 very thanks about the sealing! It works using 0.4 Beta! But even using "sudo python3 comm_sbs_bqctrl.py -vvv --dev_address 0x0b --chip BQ30z55 trigger ManufacturerAccess.PermanentFailDataReset" with pi3 i still with [PF=1].  

----
Importing comm_sbs_chips/BQ30z554.py
Writing write_word_subcommand command at addr=0xb, cmd=0x0, type=uint16, v=b'', opts={'subcmd': <MANUFACTURER_ACCESS_CMD_BQ30.PermanentFailDataReset: 41>}
Store ManufacturerAccess.PermanentFailDataReset: 00 WORD=0x29
Write ManufacturerAccess: CMD=00 WORD=29 00
MA.PermanentFailDataReset:      trigger SUCCESS Trigger switch write accepted
---

But the flag still.

rokisuk, 2023-08-23:
> hi, for unseal you need to use newer version of DJI_Battery_Killer_ver.0.4beta_26.09.2021: https://github.com/o-gs/dji-firmware-tools/issues/214
>
> with USB to I2C atater: [...]

Labas. Gal galeciau asikonsultuoti su tavimi apie Dji ronin mx baterija? Turiu dvi baterijos ir abi nesikrauna ir neijungia ronino. Kaip su jumis susisiekti?

labodezao, 2024-03-30:
@negis02 and @vinny-ferrari 

Did you use the Rpi command or I just Need the Dji battery killer for my ronin M batteries ?
Thanks

demonpixel, 2024-11-12:
I understand that in order to clear the PF Flag on a Ronin M / MX battery, you need to use both the Windows program DJI_Battery_Killer_ver.0.4beta_26.09.2021 and a Raspberry Pi.

I was just curious as to why the Windows program DJI Battery Killer is unable to clear the flag. In other words, why do you have to switch to the Raspberry Pi to clear it? I don't mind doing it that way; just wondering.

negis02, 2024-11-13:
Hi, i think it was limitation in that version DJI_Battery_Killer_ver.0.4beta_26.09.2021,

I successfully did again just using never version, there is info:
need to find DJi Battery Killer ver. 0.5 Beta with dlls
[link to WEB archive](https://web.archive.org/web/20221223191927/http://master-club.org/drone/software/)
[DJi_Battery_Killer_ver.0.5beta2_01.02.2022_with_dlls.7z](https://web.archive.org/web/20221223191927/http://master-club.org/drone/software/DJi_Battery_Killer_ver.0.5beta2_01.02.2022_with_dlls.7z)
[same file on GoogleDrive](https://drive.google.com/file/d/1xOsiAeo3YO8gZglvpwOueL4mxNg4DHL5/view)
https://www.youtube.com/watch?v=kg1g2aWtz4Y

yucelyilmaz, 2025-01-16:
> Hi,
> 
> I am also trying to get my DJI Ronin M battery running again. I am using a Raspberry Pi 3b+ to do this. unfortunately when I read the addresses I only see 0x40. Can someone help me solve the problem?
> 
> thank you very much

Hello tommy,

After three years maybe you don't remember. How did you solve this problem ? Because I am stuck on same point. I just see 0x40 address. I would like appreciate your comments.

Cleric-K, 2025-06-25:
> Hello tommy,
>
> After three years maybe you don't remember. How did you solve this problem ? Because I am stuck on same point. I just see 0x40 address. I would like appreciate your comments.

@yucelyilmaz , if you still need this, here's what worked for me.

First make sure you _manually_ charge the battery to operable range

The main insight came from https://github.com/o-gs/dji-firmware-tools/issues/354#issuecomment-1637972578 where @mefistotelis says that `0x40` is the address of the MCU (MSP430) and not the battery controller (BQ30z55). This made me look harder for the `0xB` slave and why it was not showing. There were two factors in my case:

1. I had lowered the clock to 10 000 Hz, expecting that lower frequencies are more likely to work. However, it turns out that my battery doesn't respond at all unless the clock was set to 40 000 Hz.
2. I had to press the button on the battery and then query (double press and hold preferrable). Probably the chip goes into sleep otherwise.

BTW: I've made a small I2C bridge that allows to use the `comm_sbs_bqctrl.py` script with a PC and Arduino (instead of using a Raspberry PI with hardware I2C). https://gist.github.com/Cleric-K/dba550cf68f242c6fc1771d90de53039

Wianshi, 2026-09-08:
> > MERHABA,
> > DJI Ronin M'imin pilini tekrar çalıştırmaya çalışıyorum. Bunun için Raspberry Pi 3b+ kullanıyorum. Maalesef adresleri okuduğumda sadece 0x40 görüyorum. Bu sorunu çözmeme yardımcı olabilir misiniz?
> > çok teşekkür ederim
> 
> Merhaba Tommy,
> 
> Üç yıl sonra belki hatırlamıyorsunuzdur. Bu sorunu nasıl çözdünüz? Çünkü ben de aynı noktada takılı kaldım. Sadece 0x40 adresini görüyorum. Yorumlarınız için teşekkür ederim.

> > MERHABA,
> > DJI Ronin M'imin pilini tekrar çalıştırmaya çalışıyorum. Bunun için Raspberry Pi 3b+ kullanıyorum. Maalesef adresleri okuduğumda sadece 0x40 görüyorum. Bu sorunu çözmeme yardımcı olabilir misiniz?
> > çok teşekkür ederim
> 
> Merhaba Tommy,
> 
> Üç yıl sonra belki hatırlamıyorsunuzdur. Bu sorunu nasıl çözdünüz? Çünkü ben de aynı noktada takılı kaldım. Sadece 0x40 adresini görüyorum. Yorumlarınız için teşekkür ederim.

Wianshi, 2026-09-08:
@yucelyilmaz sorunu çözebildinizmi acaba