o-gs/dji-firmware-tools #245: Dji battery killer last version updated

NHAMTAMPHUOCTHIEN, 2021-11-26:
Hi Mr Mixeysan!
Can you get link of Dji battery killer updated software to me?
Thanks you!

mixeysan, 2021-11-26:
http://master-club.org/drone/software/

NHAMTAMPHUOCTHIEN, 2021-11-27:
Thanks you very much!

mefistotelis, 2021-11-27:
To make things clear:

O-Gs are neither endorsing nor testing the 'Dji battery killer' tool by @mixeysan.
We do not check if it works, or whether it does additional malicious actions. The tool is not open-source like our releases.

Judging from user responses, the tool seem to work and provide features not implemented in `comm_sbs_bqctrl.py`, so we tolerate it.

mixeysan, 2021-11-27:
> 

sorry for getting into your topic ;)

baonamnguyen, 2021-12-02:
Hi @mixeysan, do you have the Guideline or Readme for your tools? I can't find it on the download link

mjnhchj, 2021-12-23:
Guideline and readme available at: https://youtube.com/c/minyc

GabilUAV, 2021-12-30:
Hi @mixeysan . I cant install Dji battery killer to my windows 7. There is writing missed some DLL files. can you help me please

mixeysan, 2021-12-30:
Write to me via telegram. It is not my support forum.
PS use DLLs from old versions of the programs.

tommyponny, 2022-02-27:
Hi @mixeysan, I cant download the new version. the link no longer exists. can you send it by mail or somehow else? thank you

mixeysan, 2022-02-27:
> Hi @mixeysan, I cant download the new version. the link no longer exists. can you send it by mail or somehow else? thank you

Hi everyone. On February 24, 2022, the army of the Russian Federation has now officially invaded our country and is killing Ukrainians. The server is intermittent due to a power outage.

mefistotelis, 2022-02-27:
Hang in there, @mixeysan. Otherwise, we in Poland are next.

dangdinhhiep, 2022-07-07:
> http://master-club.org/drone/software/ also U can find new versions [here](http://4pda.to/forum/index.php?s=&showtopic=1003090&view=findpost&p=111152949) check http (not https)

Please give me the DJI battery killer software if you can, thank you so much

GrinFactor, 2022-08-11:
I'm getting the following errors: 
ERROR: Could not perform SMBus read 0x00
and
Failed to connect to 9999:  HID_SMBUS_DEVICE_IO_FAILED
[image]

MAKTrax60, 2022-08-13:
I am getting same error as GrinFactor.  Tried on 2 PCs.

joycatch, 2022-08-30:
I'm also getting:

"ERROR: Could not perform SMBus read 0x00"

Does anybody know how to resolve that? I have  Spark battery...

[image]

gd510, 2022-09-18:
+1. I am having the same exact problems. Getting both of these errors. Batteries were in storage for a long time when I bought my used drone, and now some of them seem to be dead (no LED at all), but some work. They are the same age, so I think they were in hibernation mode for too long, but I can't bring them out of it. 

Were you able to resolve this?

Thanks.
- George

> I'm also getting:
> 
> "ERROR: Could not perform SMBus read 0x00"
> 
> Does anybody know how to resolve that? I have Spark battery...
> 
> [image]

SH4LT1S, 2022-09-23:
same error ERROR: Could not perform SMBus read 0x00

fluffhead123, 2022-10-11:
anyone know if this software works with EV2400? it seems to detect it correctly but when i hit connect it gives error  

Failed to connect to 5410585127002200:
HID SMBUS DEVICE IO FAILED

venom-code, 2022-10-21:
does this tool work with the Mavic Pro 1, if yes someone kindly guide me to the correct path, inside the tool i did not see support for the MP that is why i am asking if i might have missed something,

drkenb, 2022-10-23:
[image]

I am having absolutely no luck whatsoever.  I've tried several different version of the DJI Battery Killer app on both Win10 and Win11 systems with no luck.  As you can see, the program seems to load fine but when I go to Read, I get the "Could not perform SMBus read 0x00" error. I've checked the pinouts several times and they are correct on both battery and CP2112 board.   Help!!!

drkenb, 2022-10-23:
Oh, one more thing: When I tried the HELP button, I got a popup window that I could not close.  Had to go into Task Manager and End Task.  The latest version I am trying is dated Feb 2022.

gd510, 2022-10-23:
I’m having the exact same issues. So far I have not found a fix. - George

JUJ1K, 2022-11-04:
Im wondering... Does dji battery killer works on cp2102. I have logistical problems to get 2112 but have 2102 in store. 
Someone tryied this prog on other hub?

bad1dea, 2022-11-18:
> 

did you find out if it works for mavic pro battery 1?

donchapri, 2022-11-21:
Is there anyone who can help me solve some problems, same problem?

Tony4464, 2022-12-22:
Hello everyone. I tried every link to downloaded the DJI Battery Killer software but none of the Zip files will open. Tried with WinZip and several others programs but it still wont open. Any help will be greatly appreciated. Thank you, in advanced.

drkenb, 2022-12-22:
Never got any of it to work at all.  Waste of time & money.
________________________________________________________________________

Hello everyone. I tried every link to downloaded the DJI Battery Killer software but none of the Zip files will open. Tried with WinZip and several others programs but it still wont open. Any help will be greatly appreciated. Thank you, in advanced.

mixeysan, 2023-01-09:
> Never got any of it to work at all. Waste of time & money.
> […](#)

Change the pad between the chair and the ass - then it might work.
PS The program is free, I don't know what you spent money on.

mefistotelis, 2023-01-09:
As requested, closing.

drkenb, 2023-01-09:
Mixeysan - First, your childish comment, "Change the pad between the chair and the ass - then it might work." speaks volumes about you as a person.  Cut the crap and act adult.
Second, your comment "PS The program is free, I don't know what you spent money on." The money spent is for the parts needed to work with your non-functioning software.

mixeysan, 2023-01-09:
> Mixeysan - First, your childish comment, "Change the pad between the chair and the ass - then it might work." speaks volumes about you as a person.  Cut the crap and act adult.
> Second, your comment [...]

Can I ask you what equipment you spent the money on, the CP2112? 3 dollars? Isn't it funny for you to write about this? At least 400 people have the program working (the number of people in our group), supposedly it doesn't work for you, so who's the fool?
PS It seems to me that you are either Patrick's (trb2300) second nickname or someone else who is offended by life and doing dirty tricks to others.

fluffhead123, 2023-01-09:
FWIW once I got a cp2112 this software worked amazingly well and I was able to bring back 3 spark batteries from the the dead!  I recently upgraded to the mini 3 pro and i’m hoping the software will be update to support that. I haven’t been able to find a pin out for those batteries and have been unable to get the software to work with those yet. Thanks for the great tool!

almol, 2023-01-12:
Can someone send me a copy of DjiBatteryKiller? I get one but it does not work and master-club.org are not available, 4pda.to can't download file.
[image]

tommyponny, 2023-01-12:
@almol Here you can find a version that I used and that worked for me. However, this only worked in combination with a Raspberry Pi. I have used the DJI Battery Killer via CP2112 only to unlock the battery . for this I have often clicked on Unseal, Unseal Full Acces etc.. Then I connected the battery to the Raspberry Pi (without CP2112) and followed the instructions from the Youtube video. https://www.youtube.com/watch?v=P5PNOO2GebY&ab_channel=mefistotelis
at some point the battery was unlocked and I could reset the flag. Now the battery works again.
In any case, it is not an easy undertaking. I am a mechatronics engineer in the development industry and it was not an easy undertaking for me either. I keep my fingers crossed for you.

Software: https://drive.google.com/drive/folders/1VJGhkWqAU3KcpL_NnZssmSVSj9_UXkn5?usp=sharing

hdboycom, 2023-03-13:
Hi, 
I have issue with my DJI FPV battery, it was sitting on my desk for couple months and now won't charge, LEDs doesn't light up. I'm sure because low voltage.
Is this software work with DJI FPV battery also? Do you guys know this battery pinouts?

cyberhemmo, 2023-03-13:
Hei, olen lomalla 13.3.2023 asti. 
Työasioissa ota yhteyttä ***@***.***  

Terveisin
Heikki Savolainen

JanCihak, 2023-03-20:
Hi, 
I have a problem with DJI_Battery_Killer everything looks fine, but after try to run apllication, the program says:

[image]

Translate error: The code cannot run because: mfc140u.dll was not found. Try to fix this problem by reinstalling the program.

But I dont know how this program reinstalling and others version tell the same error message. 

Do you know how to solve this problem? 

Thank you for your help. 
Kindly Jan  

P.S. I have a windows 11

Dlow6977, 2023-04-26:
PLease help with finding diji battery killer app...

VMALMAO, 2023-05-30:
any solution for "ERROR: Could not perform SMBus read 0x00" ?

zagnuts, 2023-08-22:
> any solution for "ERROR: Could not perform SMBus read 0x00" ?

In my case I found this error was due to the battery not having sufficient power left to allow it to be read. Need to provide power via a separate power supply and/or carefully (to avoid damaging the pins) connect the battery up whilst also pushing the battery into the normal charger.

anandakrishnas, 2023-09-11:
> http://master-club.org/drone/software/ also U can find new versions [here](http://4pda.to/forum/index.php?s=&showtopic=1003090&view=findpost&p=111152949) check http (not https)

HELO, YOUR WORK IS VERY NICE, I USED IT FOR MANY LAPTOP BATTERIES ALSO, IF POSSIBLE CAN YOU SHARE THE SOURCE FILES,?

softa74, 2024-01-04:
Hello, can I help? Why is the current under 0A and 0B negative, I connected other batteries that are correct and they have a value of 0 on those items. The thing is that the batteries are full but the drone reads that the battery is critically empty?
[image]

saqibns1, 2024-01-05:
can this software work with arduino?

anandakrishnas, 2024-01-05:
> can this software work with arduino?

Arduino not supported, works with only cp2112

linoko, 2024-02-17:
@mixeysan please add me in telegram group r signal group i need your help. My phone is +359889654068

kardiyalog, 2024-09-20:
Can I use rp2040 zero instead of cp2112?

hoo676, 2025-02-20:
> Can I use rp2040 zero instead of cp2112?

hi,can you send the soft to me?thankyou! hu676@qq.com

ayoubriahi, 2025-05-11:
Hey everyone! 👋

I really need some help. I have a DJI Spark drone, and by accident, I flashed a Mavic 2 Pro battery firmware onto the Spark battery using DJI Killer. 😓

I only have this one battery  it’s part of an important school project and I’m stuck.

If anyone has a working Spark battery and could extract the original firmware (Spark_IntelBat_FW.bin) and share it, or knows how I can fix this issue, I’d be super grateful 🙏

ayoubriahi, 2025-05-11:
> Ok, since `PUEK-2017-09` isn't published, here you go.
> 
> [wm100_1100_v01.00.00.60_20170502.pro.fw_1100.bin.txt](https://github.com/user-attachments/files/20149230/wm100_1100_v01.00.00.60_20170502.pro.fw_1100.bin.txt)
> 
> What's that? See extraction logs: https://github.com/o-gs/dji-firmware-tools/actions/runs/14681720726/job/41205177918 (type wm100 in search field)
> 
> Even more info: https://github.com/o-gs/dji-firmware-tools/wiki/Firmware-m1100
> 
> EDIT: oh wait, that module wasn't even encrypted. I should've just pointed you to command lines. Oh well..

thanks for the replay but whene i tryed to uplaod the file it says that the file should be a multiple of 32k

linoko, 2025-05-13:
> Hey everyone! 👋
>
> I really need some help. I have a DJI Spark drone, and by accident, I flashed a Mavic 2 Pro battery firmware onto the Spark battery using DJI Killer. 😓
>
> I only have this one battery [...]

[dbk_fw_wm100_1100_v01.00.00.60_20170502.pro.fw_1100.zip](https://github.com/user-attachments/files/20187532/dbk_fw_wm100_1100_v01.00.00.60_20170502.pro.fw_1100.zip)

HERE IS THE FILE

ayoubriahi, 2025-05-13:
> [dbk_fw_wm100_1100_v01.00.00.60_20170502.pro.fw_1100.zip](https://github.com/user-attachments/files/20187532/dbk_fw_wm100_1100_v01.00.00.60_20170502.pro.fw_1100.zip)
>
> HERE IS THE FILE

thanks linoko for your replay, but are you sure that this firmware is for the spark cause i tested and it didn't work😓

ayoubriahi, 2025-05-15:
please i need help still stuck with the same problem

> Hey everyone! 👋
>
> I really need some help. I have a DJI Spark drone, and by accident, I flashed a Mavic 2 Pro battery firmware onto the Spark battery using DJI Killer. 😓
>
> I only have this one battery [...]

accmohammed22, 2025-05-18:
[image]

[image]

[image]

Hello my friends, there is a solution to the error

accmohammed22, 2025-05-18:
[image]

Hello my friends, there is a solution to the error