o-gs/dji-firmware-tools #248: Mavic 2 zoom\pro mainboard eMMC pinouts (Happy New Year! :) )

mixeysan, 2021-12-25:
[image]
Please add to wiki

mixeysan, 2021-12-26:
Also I have a dumps from this mainboard. I can upload it to the wiki.

mefistotelis, 2021-12-26:
Sure, please do.
For binary dumps, we typically upload it somewhere else, and only link on Github.

Will add the images somewhere, not sure if to wiki or to harware-schematics repo.

Alessandro2023, 2022-01-16:
Olá Mixeysan tudo bem....queria ver se co segue me ajudar vc consegue me passa o link do dji battery killer 0.4 beta 2 meu email é  ale3dprint@hotmail.com

mtietke, 2022-07-19:
@mixeysan , Could you provide the Link to the Memory-Dump?

KJY5713, 2024-12-03:
Hi,
I want to get the eMMC Memory-Dump file.
Could you provide the Link?