o-gs/dji-firmware-tools #251: Missing TBIE key

milenovic, 2021-12-28:
Is there any chance to determine TBIE key used in WM232 firmware (V02.04.1604 - V02.04.2120) bootarea.img/rtos.img/normal.img partitions? The correct PRAK key for them is PRAK-2020-01, but none of the TBIE keys available in this repository are correct.
Correct UFIE key is UFIE-2020-04.

mefistotelis, 2021-12-28:
If you wish to start looking into keys, make sure to read:
https://github.com/o-gs/dji-firmware-tools/wiki/Firmware-m0901#boot-process
https://github.com/o-gs/dji-firmware-tools/wiki/Firmware-m0801#keys-derivation
.. and the whole pages above.

mefistotelis, 2022-12-14:
Hm, you're right - `TBIE-2018-01` is still not public..

Ok then, file attached.

[wm230_0801_v10.00.07.12_20180126.pro.fw_0801.bootarea_p0_BLLK.bin.gz](https://github.com/o-gs/dji-firmware-tools/files/10227127/wm230_0801_v10.00.07.12_20180126.pro.fw_0801.bootarea_p0_BLLK.bin.gz)

milenovic, 2023-12-20:
One year later...
Maybe it's time to make the key public?
0x7b, 0xca, 0x59, 0x6f, 0x22, 0x73, 0xc5, 0x19, 0x5e, 0x41, 0x42, 0xaa, 0x3d, 0x20, 0x1e, 0x25

gogisoft, 2023-12-20:
Thanks for sharing. Tring to get a Mini 3 gimbal working again on spare time. Does your post imply DJI has ecryprion keys protecting the firmware?

milenovic, 2023-12-20:
That question is best answered by fully reading this amazing repository ;)

mefistotelis, 2024-10-05:
Finally added with:
[dji_imah_fwsig; Added TBIE-2020-02](https://github.com/o-gs/dji-firmware-tools/commit/ad3e351731f074452ff79d1026ea2487b280cf79)

closed 2024-10-05 by mefistotelis