o-gs/dji-firmware-tools #252: How to solve the error 40011 on DJI mini2 and air2

linoko, 2021-12-29:
Hi dear guys,
I have bad experience with using the python script for calibration of gimbal on both MINI2 and AIR2. The script works but after restart aircraft shows "Gimbal calibration error 40011" Any suggestions or help to solve that issue?

mjnhchj, 2021-12-30:
You need DJI VPN and DJI account to login to DJI server.

linoko, 2021-12-30:
> You need DJI VPN and DJI account to login to DJI server.

Do you can help me, Do you have vpn

mjnhchj, 2021-12-30:
I do not have.

linoko, 2021-12-30:
Does any one have a solution, may pay reasonable price

coptersafe, 2021-12-31:
i have ,  pm me  to email memo.problem@gmail.com

coptersafe, 2021-12-31:
i have , pm me

coaz, 2022-01-03:
> i have , pm me

I contacted you via email.

linoko, 2022-01-03:
> i have , pm me

I also write you at mail

mEfis-afk, 2022-02-18:
I can remove the error. Leave the contact!

coaz, 2022-02-19:
alexander.allenz@gmail.com @mEfis-afk

coaz, 2022-02-19:
+502 5388 0018

Ardumen, 2022-03-01:
I also need to fix error 40011 on mavic mini 2. Who can help?

mEfis-afk, 2022-03-01:
> Também preciso corrigir o erro 40011 no mavic mini 2. Quem pode ajudar?

Leave your WhatsApp for contact

alives2000, 2022-03-03:
@mEfis-afk Hello, I also need help with error 40011, I got the error after calibrating through the comm_og_service_tool.py script. Can you give me your contacts WA or email?

mEfis-afk, 2022-03-03:
> @mEfis-afk Olá, também preciso de ajuda com o erro 40011, recebi o erro após calibrar através do script comm_og_service_tool.py. Você pode me dar seus contatos WA ou e-mail?

Leave your WhatsApp for contact

alives2000, 2022-03-03:
@mEfis-afk my whatsapp +77772137795

Ardumen, 2022-03-03:
For Mavic Mini, the error is solved by rolling back the drone firmware to version V01.00.0302 and calibrating the gimbal.

alximiktik, 2022-03-21:
> i have , pm me

please help to remove 40011 after calibration

mEfis-afk, 2022-03-21:
Number WhatsApp, please

à(s) 12:57:

alximiktik, 2022-03-21:
> > i have , pm me
> 
> please help to remove 40011 after calibrati

+

> […](#)

+37494504633

QiwSam, 2022-06-15:
> i have , pm me

Hello, I also need help with error 40011, I got the error after calibrating through the comm_og_service_tool.py script.

mEfis-afk, 2022-06-15:
> > eu tenho, me mande
> 
> Olá, também preciso de ajuda com o erro 40011, recebi o erro após calibrar através do script comm_og_service_tool.py.

Leave your WhatsApp and I'll contact you

QiwSam, 2022-06-16:
> Leave your WhatsApp and I'll contact you

+380501653990

ja4who, 2022-06-25:
Olá, também preciso de ajuda com o erro 40011. Se puder entrar em contato. +5566996685331

thangnd85, 2022-07-07:
> i have , pm me

Could you check email

Leandroosorio, 2022-07-12:
Tengo el mismo error 40011 alguien me puede ayudar? leandrorosorio@gmail.com

mEfis-afk, 2022-07-12:
> Tengo el mismo error 40011 alguien me puede ayudar? [leandrorosorio@gmail.com](mailto:leandrorosorio@gmail.com)

Leave your WhatsApp

pratikalwaysrocks, 2022-07-24:
I need help in solving error after running python script.
My WhatsApp +919664514503

Please help, Thanks

Asdertila, 2022-07-26:
The same problem with error 40011
Has anyone been helped via WhatsApp or is it a scam?

mEfis-afk, 2022-07-26:
> O mesmo problema com o erro 40011 Alguém foi ajudado via WhatsApp ou é uma farsa?

Leave your WhatsApp and I'll contact you

Asdertila, 2022-07-27:
+375293885334

guewald, 2022-07-27:
> > O mesmo problema com o erro 40011 Alguém foi ajudado via WhatsApp ou é uma farsa?
> 
> Leave your WhatsApp and I'll contact you

+5551999948379

alxgacc, 2022-07-28:
Who needs to remove error 40011, you can write to me on whatsapp +62 812-8430-9143

DJI-Solution, 2022-11-01:
This guy can remotely fix DJI Error 40011 and 40021, I just connect my drone to my computer and he do all the work.
I know him by this video:
https://youtu.be/1AgMwswzWkg

He is so friendly and quick reply, You can Telegram / Whatsapp him +85589898982
Hope it you help you, but it will cost a little, i can't tell the price but it's fast and cost less then send to Service Center.

DJIGuru, 2023-02-13:
Have found the solution to clear error. WhatsApp me: +61 43 631 0173

DJI-Solution, 2023-03-22:
Fix Error 40021 and 40011 for DJI Mini1,SE,Mini2,Mini3 and Air 1,2,2s. It also possible to remotely fix, you just need to connect your drone to your pc. for more information telegram us: t.me/djisolution

dronedegaragem, 2025-07-31:
My whatapps : +5585988395152 . @mEfis-afk