o-gs/dji-firmware-tools #253: WM331 GimbalCalib JointCoarse and GimbalCalib LinearHall

rsrmdp, 2021-12-30:
F:\dji-ftm>comm_og_service_tool.py COM7 -vv WM331 GimbalCalib JointCoarse
Opened COM7 at 9600

Info: The Gimbal will move through its boundary positions, then it will fine-tune its central position. It will take around 15 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 66 f7 20 04 08 01 6b 1b
Sending packet...
Waiting for reply...
Timeout while waiting for reply.
Retrieved 10 packets (527b), dropped 0 fragments (0b)
Prepared binary packet:
55 0e 04 66 0a 04 66 f7 20 04 08 01 6b 1b
Sending packet...
Waiting for reply...
Timeout while waiting for reply.
Retrieved 1 packets (68b), dropped 0 fragments (0b)
Prepared binary packet:
55 0e 04 66 0a 04 66 f7 20 04 08 01 6b 1b
Sending packet...
Waiting for reply...
Timeout while waiting for reply.
Retrieved 7 packets (341b), dropped 0 fragments (0b)
No response received.
Error: No response on calibration command JointCoarse request.

F:\dji-ftm>comm_og_service_tool.py COM7 -vv WM331 GimbalCalib LinearHall
Opened COM7 at 9600

Info: The Gimbal will slowly move through all positions in all axes, several times. It will take around 30 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 42 02 20 04 08 02 ce 05
Sending packet...
Waiting for reply...
Timeout while waiting for reply.
Retrieved 2 packets (139b), dropped 0 fragments (0b)
Prepared binary packet:
55 0e 04 66 0a 04 42 02 20 04 08 02 ce 05
Sending packet...
Waiting for reply...
Timeout while waiting for reply.
Retrieved 8 packets (396b), dropped 0 fragments (0b)
Prepared binary packet:
55 0e 04 66 0a 04 42 02 20 04 08 02 ce 05
Sending packet...
Waiting for reply...
Timeout while waiting for reply.
Retrieved 2 packets (128b), dropped 0 fragments (0b)
No response received.
Error: No response on calibration command LinearHall request.

I would think that there is some hardware error, but I already repaired a sensor and from the app I did not correct the stabilization.

linoko, 2021-12-31:
That happend when do this callibration via python. Is the gimbals leveled? May be you received error message "Gimbal calibration 40011)

rsrmdp, 2021-12-31:
> That happend when do this callibration via python. Is the gimbals leveled? May be you received error message "Gimbal calibration 40011)

that error does not appear

linoko, 2021-12-31:
> > That happend when do this callibration via python. Is the gimbals leveled? May be you received error message "Gimbal calibration 40011)
> 
> that error does not appear

So what is the major issue that you have, describe it please?

rsrmdp, 2021-12-31:
> > > That happend when do this callibration via python. Is the gimbals leveled? May be you received error message "Gimbal calibration 40011)
> > 
> > that error does not appear
> 
> So what is the major issue that you have, describe it please?

apparently as it reads I am not receiving information from the gimbal.  If I find a solution I will post it here

linoko, 2022-01-03:
> apparently as it reads I am not receiving information from the gimbal. If I find a solution I will post it here

You will not see incoming info from gimbal, it only seen on DJI Spark

linoko, 2022-01-06:
Someone have solution to share?

vserena82, 2022-04-03:
Hello,

I have same problem. I have P4P with raplaced gimbal yaw motor, II'm trying to use JointCoarse and LinearHall with WIN10:
```

C:\DJI>py comm_og_service_tool.py COM4 -vv WM331 GimbalCalib JointCoarse
Opened COM4 at 9600

Info: The Gimbal will move through its boundary positions, then it will fine-tune its central position. It will take around 15 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 d0 f3 20 04 08 01 f3 ec
Sending packet...
Waiting for reply...
Timeout while waiting for reply.
Retrieved 1 packets (68b), dropped 0 fragments (0b)
Prepared binary packet:
55 0e 04 66 0a 04 d0 f3 20 04 08 01 f3 ec
Sending packet...
Waiting for reply...
Timeout while waiting for reply.
Retrieved 8 packets (401b), dropped 0 fragments (0b)
Prepared binary packet:
55 0e 04 66 0a 04 d0 f3 20 04 08 01 f3 ec
Sending packet...
Waiting for reply...
Timeout while waiting for reply.
Retrieved 1 packets (68b), dropped 0 fragments (0b)
No response received.
Error: No response on calibration command JointCoarse request.
```

There is a solution?
Thank you

rsrmdp, 2022-04-06:
there is no solution at the moment.

rsrmdp, 2022-04-06:
Only
1
https://youtu.be/4q9fmd5iWVM
2
https://youtu.be/0G7p9EroalM
I hope... it works for you.  but they changed the bell.  maybe changing the Hall sensors will fix it.