o-gs/dji-firmware-tools #254: DJI Mavic Air 2 Gimbal Calibration

Ianf223, 2021-12-30:
After replacing the Air 2 gimbal i followed instructions on calibrating it from (https://www.youtube.com/watch?v=6MMpbMhy20Y), everything went right up until i entered 'py comm_og_service_tool.py COMPORT MODEL GimbalCalib JointCoarse' it gave me the error 'usage: comm_og_service_tool.py [-h] (--port PORT | --bulk) [-b BAUDRATE] [-w TIMEOUT] [--dry-test] [-v] [--version]
                               product command ...
comm_og_service_tool.py: error: argument product: invalid choice: 'COM3' (choose from 'A2', 'P330', 'P330V', 'P330Z', 'P330VP', 'WM610', 'P3X', 'P3S', 'MAT100', 'P3C', 'MG1', 'WM325', 'WM330', 'MAT600', 'WM220', 'WM620', 'WM331', 'MAT200', 'MG1S', 'WM332', 'WM100', 'WM230', 'WM335', 'WM240', 'WM245', 'WM246', 'WM160', 'WM231', 'WM232', 'WM260')' I've tried different computers, cables and it doesn't seem to calibrate, I was able to find the com port so that's why I'm confused, If anyone can help it will be greatly apricated.

mefistotelis, 2021-12-30:
Have you tried reading with understanding?

linoko, 2021-12-31:
> After replacing the Air 2 gimbal i followed instructions on calibrating it from (https://www.youtube.com/watch?v=6MMpbMhy20Y), everything went right up until i entered 'py comm_og_service_tool.py [...]

use like this:
comm_og_service_tool.py --port COM4 -vv WM241 GimbalCalib LinearHall
comm_og_service_tool.py --port COM4 -vv WM241 GimbalCalib LinearCoarse

Important is to find correct com port in my case its com4. You may easy find it if first start dji Assistant 2 (Consumer drones) than plug drone and in device manager will see com port which number is accociated. then open command prompt and prepare to write text above, but before you start must close DJI assistant.

Ianf223, 2021-12-31:
@linoko THANK YOU SM everything works now but I know about the codes still being in the app but again thank you again!

linoko, 2022-01-03:
> @linoko THANK YOU SM everything works now but I know about the codes still being in the app but again thank you again!

They will be apper, still no free cure for them

aanestaa, 2022-09-05:
@linoko hi can you help me with mavic 3 pro calibration?

Drafthorseadmin, 2025-01-12:
I know this in an old thread, but I just came across this problem when I replaced the gimbal assy (with the camera) of my Mini 2. The starting point was that the new camera was crooked to one side, even though it seems to work otherwise. Naturally loads of errors, such as Gimbal motor overload, incompatible data, calibration error, etc.. 

It tried this python method on a Windows 11 PC and got tired of tweaking the security settings, ports and shit and decided to attempt this on my Mac (M2 Pro, Sequoia 15.0.1). 

After installing python, the pyserial and the setup, I was ready to go. I tried the command:
comm_og_service_tool.py --port COM4 -vv WM160 GimbalCalib JointCoarse
in this format and got: comm_og_service_tool.py: error: argument product: invalid choice: 'WM160\xa0GIMBALCALIB'
FInding the right port has not been made easy in Win11 nor Mac.

At this point I still had the DJI Assistant running, and I figured it was using the port and had to be released. So I quit the DJI Assistant and left the drone connected via usb. I also wanted to avoid the guesswork and see where my mini was connected and run: ls /dev/tty.*
That returned:
/dev/tty.Bluetooth-Incoming-Port	/dev/tty.debug-console
/dev/tty.Joy-0573			/dev/tty.usbmodem103

After identifying that my Mini 2 was using /dev/tty.usbmodem103, I updated  the command to:
comm_og_service_tool.py --port /dev/tty.usbmodem103 -vv WM160 GimbalCalib JointCoarse and It worked as it should. Then I run the second command: comm_og_service_tool.py --port /dev/tty.usbmodem103 -vv WM160 GimbalCalib LinearCoarse and the gimbal and camera are dead on, and the errors were gone. 

Even though I did this with my Mini 2, this should work on other models as well.

raniere692hp, 2025-05-01:
https://github.com/o-gs/dji-firmware-tools/issues/254#issuecomment-1003321740
#linoko poderia me ajudar
estou tendo problema em atualizar firmware do gimbal 
se quiser pode fazer do meu pc