o-gs/dji-firmware-tools #257: Error: ModuleNotFoundError: No module named 'smbus2'

Padke9, 2022-01-19:
How to solve this error in Raspberry pi 3

./comm_sbs_bqctrl.py  --dev_address 0x0b --short read BatteryStatus

Error: ModuleNotFoundError: No module named 'smbus2'
Traceback (most recent call last):
  File "./comm_sbs_bqctrl.py", line 3681, in <module>
    main()
  File "./comm_sbs_bqctrl.py", line 3594, in main
    smbus_open(po.bus, po)
  File "./comm_sbs_bqctrl.py", line 1469, in smbus_open
    import smbus2
ModuleNotFoundError: No module named 'smbus2'

Padke9, 2022-01-19:
./comm_sbs_bqctrl.py  -vvv --bus "i2c:1"  --dev_address 0x0b --short read BatteryStatus
Opening i2c:1
Error: ModuleNotFoundError: No module named 'smbus2'
Traceback (most recent call last):
  File "./comm_sbs_bqctrl.py", line 3681, in <module>
    main()
  File "./comm_sbs_bqctrl.py", line 3594, in main
    smbus_open(po.bus, po)
  File "./comm_sbs_bqctrl.py", line 1469, in smbus_open
    import smbus2
ModuleNotFoundError: No module named 'smbus2'

Padke9, 2022-01-19:
i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- 0b -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

Padke9, 2022-01-19:
dtparam=i2c_arm=on
dtparam=i2c_baudrate=66000
dtparam=i2c1_baudrate=66000
dtparam=i2c_arm_baudrate=66000
#dtparam=i2s=on
dtparam=spi=on

Padke9, 2022-01-19:
ls /dev/*i2c*
/dev/i2c-1

mefistotelis, 2022-01-19:
Is there something unclear in the message?
```
Error: ModuleNotFoundError: No module named 'smbus2'
```

Also, if you see an error message, might help googling it.

Padke9, 2022-01-19:
it means that the smbus2 library is not installed

but

 pip install smbus2
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting smbus2
  Downloading https://files.pythonhosted.org/packages/c8/bf/62ef029fb7077fc87c3539f7365859bccc6cedb2bb20796b737b788c8d09/smbus2-0.4.1-py2.py3-none-any.whl
Installing collected packages: smbus2
Successfully installed smbus2-0.4.1

./comm_sbs_bqctrl.py -v  --dev_address 0x0b --chip BQ20z45 --short monitor BQStatusBitsMA
Opening i2c:1
Error: ModuleNotFoundError: No module nad 'smbus2'
Traceback (most recent call last):
  File "./comm_sbs_bqctrl.py", line 3681, in <module>
    main()
  File "./comm_sbs_bqctrl.py", line 3594, in main
    smbus_open(po.bus, po)
  File "./comm_sbs_bqctrl.py", line 1469, in smbus_open
    import smbus2
ModuleNotFoundError: No module named 'smbus2'

Padke9, 2022-01-19:
sudo apt-get install python3-smbus
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-smbus is already the newest version (4.1-1).
0 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.

Padke9, 2022-01-19:
dpkg-query -L python3-smbus
/.
/usr
/usr/lib
/usr/lib/python3
/usr/lib/python3/dist-packages
/usr/lib/python3/dist-packages/smbus.cpython-36m-arm-linux-gnueabihf.so
/usr/lib/python3/dist-packages/smbus.cpython-37m-arm-linux-gnueabihf.so
/usr/share
/usr/share/doc
/usr/share/doc/python3-smbus
/usr/share/doc/python3-smbus/changelog.Debian.gz
/usr/share/doc/python3-smbus/changelog.gz
/usr/share/doc/python3-smbus/copyright

Padke9, 2022-01-19:
sudo apt-get install python-smbus python3-smbus python-dev python3-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
python-dev is already the newest version (2.7.16-1).
python-smbus is already the newest version (4.1-1).
python3-dev is already the newest version (3.7.3-1).
python3-smbus is already the newest version (4.1-1).
0 upgraded, 0 newly installed, 0 to remove and 7 not upgraded

Padke9, 2022-01-19:
pip show smbus2
Name: smbus2
Version: 0.4.1
Summary: smbus2 is a drop-in replacement for smbus-cffi/smbus-python in pure Python
Home-page: https://github.com/kplindegaard/smbus2
Author: Karl-Petter Lindegaard
Author-email: kp.lindegaard@gmail.com
License: MIT
Location: /usr/local/lib/python2.7/dist-packages
Requires:
Required-by:

closed 2022-01-19 by Padke9