o-gs/dji-firmware-tools #258: Dji spark battery repair

HonestMstk, 2022-01-23:
I saw a tutorial on youtube on how to fix a mavic air battery using your (excellent)  tools. A commenter said that it worked on his spark batteries. So i bought a raspberry pi and set everything up but when i tried to unseal, it didnt work. Is there any way to make it work? Thank you for the amazing software.

mefistotelis, 2022-01-23:
With Mavic Pro batteries, I had to enable line sampling on RPI to make the unseal work. I can't remember if it was the same with Spark.

Watch my tutorial for details:
https://www.youtube.com/watch?v=P5PNOO2GebY

kipz, 2022-01-23:
FWIW - I get no errors when unsealing a Spark battery but it doesn't actually work (security mode remains equal to 3).

HonestMstk, 2022-01-23:
> With Mavic Pro batteries, I had to enable line sampling on RPI to make the unseal work. I can't remember if it was the same with Spark.
> 
> Watch my tutorial for details: https://www.youtube.com/watch?v=P5PNOO2GebY

Thank you very much, i will try it and inform you of the results!

HonestMstk, 2022-01-23:
> FWIW - I get no errors when unsealing a Spark battery but it doesn't actually work (security mode remains equal to 3).

Same here

HonestMstk, 2022-01-23:
> With Mavic Pro batteries, I had to enable line sampling on RPI to make the unseal work. I can't remember if it was the same with Spark.
> 
> Watch my tutorial for details: https://www.youtube.com/watch?v=P5PNOO2GebY

It was your tutorial i was following (mavic air), thanks for the help, dont know if enabling sampling was in the video, i think it was and it still wouldnt unseal.

mefistotelis, 2022-01-24:
> Could it be that i have to use a different interval?

Unlikely. This interval is already very small.
Maybe your battery just doesn't use the default password?

Try using the Windows tool shared is some threads here.
Some tools can use special features made by DJI to skip the password, though I don't know which ones or any details.

kipz, 2022-01-24:
Yeah, that was my guess too - something auth related. Unfortunately I'm not in a position to test the windows stuff - no windows and no adapter - pi only.

I've seen someone do this with a TRB2300 https://www.youtube.com/watch?v=M2wbo2IdIJY - so we know it's possible.

HonestMstk, 2022-01-24:
> 
I ordered the cp2112 and will try the other tools after it arrives, probably 2 months from now. 
Thanks again mate.

kipz, 2022-01-24:
@mefistotelis in the mean-time, any advice on how to debug this? These other devices/softwares aren't magic - they just know something we don't!

mefistotelis, 2022-01-24:
SMBus is very easy to sniff. That's probably why there are so many such tools.

It's not that this is hard, sniffing or just figuring out by experiments. I didn't finished with support of newer batteries mostly because I just got bored of this subject.

Read the SBS spec of similar chip, and it should be easy to start trying stuff.
DJI FW has few modifications to TIs original, but it mostly adds new commands, it doesn't change the existing ones (besides a few which seem to have been unintentionally skewed).

HonestMstk, 2022-01-24:
> @mefistotelis in the mean-time, any advice on how to debug this? These other devices/softwares aren't magic - they just know something we don't!

I too thought about looking into the code for a different default password or sth, but as mefistotelis said, its different tricks that they use to bypass it. My knowledge is severely lacking so its best to wait for the tool to arrive.

kipz, 2022-01-25:
I read the contents ManufacturerAccess.SecurityKey, and from what I can tell from docs for similar chips, this is actually the unseal key. Can I be right about this? If so, then we are not currently sending the right key during unseal - we are sending the default. Similarly for FullAccess.

mefistotelis, 2022-01-25:
_"The unseal key can be read and changed via the MAC SecurityKey() command when in the FULL ACCESS mode."_

Yes, the docs are right.
And yes, by default the tool sends the default key, obviously.

kipz, 2022-01-25:
OK, thanks. I assumed we'd read/write the key automatically if this were possible. In any case, the only reason I mention it is because it also doesn't seem to work when I use the MAC SecurityKey, and thought maybe I was screwing something up.

mjnhchj, 2022-01-27:
You can use CP2112 + DJI BK to unseal and full option for DJI Spark battery. https://youtu.be/XsoB7Mjrtho the Spark battery is using BQ9003 chip.

kipz, 2022-01-27:
Thanks @mjnhchj - I have a raspberry pi, and I can talk i2c/smbus to it, so ideally I'd like to get it going like this. I don't really want to buy anything or run windows (or some random exe on it). Thanks again though.

Some useful context from that video though perhaps:

[image]

ivn09, 2022-02-13:
> Thanks @mjnhchj - I have a raspberry pi, and I can talk i2c/smbus to it, so ideally I'd like to get it going like this. I don't really want to buy anything or run windows (or some random exe on it). Thanks again though.

Hello, do you have any success with unsealing BQ9003 using **comm_sbs_bqctrl.py**

[image]

DIYlover, 2022-05-25:
Sorry for noob question!  May I ask if the spark battery will ever response to (no matter CP2112/Rpi/arduino) via I2C port if it had runout of battery and stayed for a very long time? 

p.s. It shows NO led light lid when I clicked the only button on all of my 3 spark batteries but all 3 will lights up 1 LED when I put them into original DJI battery charging hub.

truncj, 2022-07-12:
@DIYlover yes it will. You can provide 11.4v @ 0.5A to the power/ground pins of the battery and it will show up in the tool of your choice. I did this using the pins from the hub battery charger power supply to the battery itself. Unfortunately it looks like comm_sbs_bqctrl.py doesn't support the DJI Spark as of now.

DIYlover, 2022-07-13:
> @DIYlover yes it will. You can provide 11.4v @ 0.5A to the power/ground pins of the battery and it will show up in the tool of your choice. I did this using the pins from the hub battery charger power supply to the battery itself. Unfortunately it looks like comm_sbs_bqctrl.py doesn't support the DJI Spark as of now. I was able to execute the unseal command once (despite running it dozens of times) but it still appears as sealed.
> 
> ```
> dji-firmware-tools $ ./comm_sbs_bqctrl.py -vvv --bus "i2c:3" --dev_address 0x0b --chip BQ40z50 sealing Unseal
> Opening i2c:3
> Importing comm_sbs_chips/BQ40z50.py
> Store ManufacturerAccess.SecKeyWord0: 00 WORD=0x414
> Write ManufacturerAccess: CMD=00 WORD=14 04
> Store ManufacturerAccess.SecKeyWord1: 00 WORD=0x3672
> Write ManufacturerAccess: CMD=00 WORD=72 36
> Reading write_word_subcommand command at addr=0xb, cmd=0x0, type=uint16, opts={'subcmd': <MANUFACTURER_ACCESS_CMD_BQ40.OperationStatus: 84>}
> Query ManufacturerAccess.OperationStatus: 00 WORD=0x54
> Write ManufacturerAccess: CMD=00 WORD=54 00
> Raw ManufacturerAccess.OperationStatus response: 04 80 77 00 00 14
> MA.OperationStatus:	0x00007780	bitfields	Operational Status bits
>       SYS_PRESENT_LOW:	0=Inactive	[PRES]	System present input state low
>        DSG_FET_STATUS:	0=Inactive	[DSG]	Discharge FET status
>        CHG_FET_STATUS:	0=Inactive	[CHG]	Charge FET Status
>       PCHG_FET_STATUS:	0=Inactive	[PCHG]	Precharge FET Status
>           FUSE_STATUS:	0=Inactive	[FUSE]	FUSE input status
>       TRIP_POINT_INTR:	1=Active	[BTPI]	Battery Trip Point Interrupt
>         SECURITY_MODE:	3=Sealed	[SEC]	Security Mode
>     SHUTDOWN_LOW_VOLT:	1=Active	[SDV]	Shutdown triggered via low pack voltage
>         SAFETY_STATUS:	0=Inactive	[SS]	Safety mode status
>     PERMANENT_FAILURE:	1=Active	[PF]	Permanent Failure mode status
>  DISCHARGING_DISABLED:	1=Active	[XDSG]	Discharging Disabled
>     CHARGING_DISABLED:	1=Active	[XCHG]	Charging Disabled
>            SLEEP_MODE:	0=Inactive	[SLEEP]	Sleep mode condition met
>        SHUTDOWN_BY_MA:	0=Inactive	[SDM]	Shutdown activated by SMBus command
>           LED_DISPLAY:	0=Off	[LED]	LED Display state
>          AUTH_ONGOING:	0=Inactive	[AUTH]	Authentication ongoing
>    AUTO_CC_OFFS_CALIB:	0=Cal Done	[ACALM]	Auto CC offset calibration by SMBus cmd
>     RAW_ADC_CC_OUTPUT:	0=Inactive	[CALOC]	Raw ADC/CC data calibration output
>     RAW_CCOFFS_OUTPUT:	0=Inactive	[CALOO]	Raw CC offset data calibration output
>      XLOW_SPEED_STATE:	0=Inactive	[XL]	400 kHz transmission SMBus mode
>           SLEEP_BY_MA:	0=Inactive	[SLEPM]	SLEEP mode activated by SMBus command
>      INIT_AFTER_RESET:	0=Inactive	[INIT]	Initialization after full reset
>        SMB_CAL_ON_LOW:	0=Not in Cal	[SLCAL]	Auto CC calibration when the bus is low
>     ADC_MEAS_IN_SLEEP:	0=Inactive	[SLPAD]	ADC Measurement in SLEEP mode
>      CC_MEAS_IN_SLEEP:	0=Inactive	[SLPCC]	Current Check measurement in SLEEP mode
>        CELL_BALANCING:	0=Inactive	[CB]	Cell balancing status
>    EMERGENCY_SHUTDOWN:	0=Inactive	[EMSHT]	Emergency Shutdown
> ```

Thx for confirmation 🙏🏻

mavicmod, 2022-12-13:
> Thanks @mjnhchj - I have a raspberry pi, and I can talk i2c/smbus to it, so ideally I'd like to get it going like this. I don't really want to buy anything or run windows (or some random exe on it). Thanks again though.
> 
> Some useful context from that video though perhaps:
> 
> [image]

can you send me this version of dji ba ttery killer ? thanks

whiteduck22, 2022-12-15:
Did anyone find a fix for Spark battery unlocking on a Raspberry Pi? Thank you

dev8edss, 2023-01-02:
> Did anyone find a fix for Spark battery unlocking on a Raspberry Pi? Thank you

i was able to unlock my spark batteries with this command
./comm_sbs_bqctrl.py --dev_address 0x0b sealing --i32key 0xCCDF7EE0 Unseal

whiteduck22, 2023-01-02:
Oh brilliant I will try that thanks

whiteduck22, 2023-01-05:
Made some progress with this after realising Raspi has i2c onboard (didn't need that cp2112) but can't seem to get it to auto detect the Spark battery chipset. It fails although it does manage to read some registers like BatteryStatus. Any ideas?

adurtydog, 2023-03-31:
> I finally resolved this after realising Raspberry Pi 3B+ actually has i2c onboard (didn't need that cp2112 which doesn't seem to have a driver in the kernel so I wasted an evening figuring that one out and a lot of head scratching and cable aligning!).
> 
> Firstly a big thanks to others on this thread for your shared wisdom above especially @dev8edss - thank you! Then thanks to this site and the person who recorded the video here - https://wiki.dji-rev.com/howto/comm_sbs_bqctrl
> 
> Here is what I did that worked with my DJI Spark Batteries - **USE AT YOUR OWN RISK**:
> 
> Connect pin2, pin3 and pin5 of the Raspberry Pi to SDA, SCL and GND - i.e. the corresponding Spark battery terminals (far left pin is SCL and far right pin is SDA.
> 
> SCL | - | + | + | GND | SDA
> 
> I found I didn't have any suitable crocodile clips to connect to the "Fly More Combo" charger so I figured out using a PP3 9V battery was enough to power the logic board via the other - and one of the +.
> 
> Downloaded just the comm_sbs_bqctrl.py and created the comm_sbs_chips directory and its contents.
> 
> `./comm_sbs_bqctrl.py -vvv --chip BQ40z50 monitor BQStatusBitsMA ` Following the helpful video you will find that this shows if you have something more serious but in my case I just had PF (permanent failure) and under-voltage. The video is for a Mavic but the commands are the same. **Please don't deconstruct your battery like he had to do!**
> 
> Then unseal the battery to allow you to make changes: `./comm_sbs_bqctrl.py -vvv --chip BQ40z50 sealing --i32key 0xCCDF7EE0 Unseal `
> 
> Then reset the PF flag (**do this at your own risk!**) `./comm_sbs_bqctrl.py -vvv --chip BQ40z50 trigger ManufacturerAccess.PermanentFailDataReset `
> 
> **Note: I found I had to run this command several times to see the PF flag clear.**
> 
> When you get it working then seal the battery again: `./comm_sbs_bqctrl.py -vvv --chip BQ40z50 sealing --i32key 0xCCDF7EE0 Seal `
> 
> Now disconnect the battery carefully and put it on to a charger. I suggest you keep an eye on the battery and charger just to be safe i.e. don't go to bed or go out until you are confident it is charging safely - call me old fashioned but I don't trust LiPos.
> 
> **Note: If a command fails (or errors) try it again but be sure to check your connections.**
> 
> That's it! Success thanks again to those who pointed me in the right direction! I've charged the batteries, ran the the Spark for a while and recharged them and all still good.

Many thanks for posting this. 
To get it to work I had to crack the Batteries open and indavidualy charge each cell with a TP4056 and add the i2c address into the commands but now have re cycled two batteries.

whiteduck22, 2023-03-31:
I really didn't want to deconstruct the batteries so I wonder if you had a different series of battery.

jayemel, 2023-04-03:
I can confirm that @whiteduck22's guide worked for my Spark batteries, I was successfully able to reset the permanent failure flag and they're working perfectly. I had to power the pack by supplying ~10V. The pack drew <10mA (I suppose the BMS has disconnected the cells and only the BMS is being powered). A 9V alkaline battery might work in a pinch.

Thanks so much @whiteduck22!

whiteduck22, 2023-04-20:
Dom in my original post I connected a 9V battery to the other - and + which was enough to power the logic board and for a single LED to light up.

jayemel, 2023-04-20:
@gosymbian as @whiteduck22 says, it was simultaneous to it being connected to the Raspberry Pi. My understanding is that when the pack is in PF the cells are disconnected from the connector, so it isn't possible to charge the cells without dismantling the pack. All that this is doing is powering the BMS to allow for communication with the Raspberry Pi and resetting the PF. Once this has been done it is possible to charge the cells as per normal with the charger.

neich03, 2023-09-05:
> > Did anyone find a fix for Spark battery unlocking on a Raspberry Pi? Thank you
> 
> i was able to unlock my spark batteries with this command ./comm_sbs_bqctrl.py --dev_address 0x0b sealing --i32key 0xCCDF7EE0 Unseal

Huge thanks to this community.  Specially to [mefistotelis](https://github.com/mefistotelis) and the team for developing this incredible tool and sharing your video for the community. Giving us the power to do something just impossible for mortals....it´s just insane, Thanks! 

Thanks also to [whiteduck22](https://github.com/whiteduck22) and [dev8edss](https://github.com/dev8edss) for the specific Spark guide!

superdioz, 2024-09-13:
First of all thanks to [dev8edss](https://github.com/dev8edss), then I want also to add an issue that I encountered and its fix.

My batteries were long term discharged (just 4v left in the three cells), after a manual charge connecting the batteries directly to a source of 12v (to bring them from 4v to 15v), the raspberry still couln't read the address 0x0b with:
`i2cdetect -y 1` , I discovered that pluggin it to the dji charger and clicking on the battery button to kick in the led error sequence for some reasons "woke up" the board and then the rasp was able to read it and ran all the commands to reset the failure.

I recovered 4 batteries in this way and I can't thank the community enough for it.

RanzCuriosity, 2024-12-26:
> I can confirm that @whiteduck22's guide worked for my Spark batteries, I was successfully able to reset the permanent failure flag and they're working perfectly. I had to power the pack by supplying [...]

Hi!! Do you have a video of how you did it! Many thanks 😊 🙏

whiteduck22, 2024-12-26:
No sorry I didn't manage to film it as it was pretty fiddly trying to hold the wires from the 9v battery onto the Spark battery and type commands and a tripod would have just got in the way. However those steps above should work. Good luck!

ayoubriahi, 2025-05-09:
Hi everyone,

I’m a student and made a huge mistake I accidentally flashed my DJI Spark battery with Mavic 2 firmware using DJI Killer. Now it’s completely dead (no lights/sound). I can’t afford a new battery, so I’m desperately asking for help:
Could someone with a working Spark battery extract the original firmware and share it? I’d need the .bin file to reflash via DJI Killer.
If extraction isn’t possible, is there a modified firmware or method to revive it?

What I’ve tried:
 -DJI Assistant 2 (fails with errors).
 -DJI Killer (worked for flashing, but now the battery won’t respond).
 -I didn’t back up the original firmware, and I’m stuck. 
Any guidance or files would save me thanks a ton!

Firecul, 2025-05-09:
> Hi everyone,
>
> I’m a student and made a huge mistake I accidentally flashed my DJI Spark battery with Mavic 2 firmware using DJI Killer. Now it’s completely dead (no lights/sound). I can’t afford a [...]

Obviously saying this is why you should always back up firmwares before doing anything else is redundant now (we all learn that the hard way sooner or later) but an important question came to mind. 
Do you have any communication with the battery at all now? if not, even getting a back up from someone else will not help you I fear.

ayoubriahi, 2025-05-09:
Thanks for the heads-up i’ve definitely learned the hard way about backups! 
To answer your question:
Current Status: The battery is completely unresponsive (no lights/sound), but based on the DJI Killer logs ([screenshot](https://image.png/)), there’s still SMBus/I2C communication (errors imply the chip is "talking" but rejecting commands). The HidSmbus_Open(); HID_SMBUS_SUCCESS line confirms the hardware interface is alive.

if the BQ chip is locked/sealed, I’m prepared to physically reflash it—I’m an EE student with lab access to programmers/SMBus tools.

[image]

ayoubriahi, 2025-05-09:
> Obviously saying this is why you should always back up firmwares before doing anything else is redundant now (we all learn that the hard way sooner or later) but an important question came to mind. [...]

i was able to connected back and powred up but still got some erorr

accmohammed22, 2025-05-19:
[image]

Dji battery killer error 
Battery air2 ![image]

[image]

ayoubriahi, 2025-05-22:
make sure the connection between the cp2112 and the battery is established (check the wiring) if everything is okay try to power up the battery with an external power supply

dvdsosa, 2026-09-28:
Adding a data point for Spark batteries that are *very* deeply discharged, since the usual "unseal → PermanentFailDataReset → seal → charger" sequence didn't work for me on its own.

Two MB1-1480mAh-11.4V packs from 2017/2018, cells at 1.77–2.2 V. The Spark key `0xCCDF7EE0` from this thread unsealed them fine and `0x0029` cleared the PF, but **the PF re-latched 2–3 s after the device reset** (I polled OperationStatus/PFStatus every ~0.8 s: PF clear at 0.6 s, CHG/DSG disabled at 1.5 s, PF latched again at 3.3 s). The lowest cell was ~2.18 V, so the Safety Cell Undervoltage threshold seems to be around 2.2 V. No way to get the pack onto the charger in time.

What worked: during that 2–3 s window the BMS precharges from the wake-up supply (12 V through a 100 Ω resistor on pins 3/5 → ~13 mA). Repeating unseal → `0x0029` → reset lifts the cells ~6–10 mV per round. On the first pack the PF stopped re-latching at round 3 (lowest cell 2.20 V), and from then on the BMS stayed in normal precharge (`PCHG` set) with all protections active; cells went from 2.2 V to 2.7 V in a few minutes. Both packs now charge normally on the DJI charger.

I put it together as an Arduino Uno/Nano sketch + bash scripts (pumping loop with safety checks, and a live per-cell monitor), MIT licensed:
https://github.com/dvdsosa/dji-spark-battery-unbrick

Thanks to everyone here, especially for the Spark key.