o-gs/dji-firmware-tools #260: DJI Mavic Mini | Gimbal 40012 after ribbon / flex cable replacement

ma-mod, 2022-01-28:
Hi Guys,

I've bought an Mavic Mini which has a broken arm and the Ribboncable to the Gimbal Yaw motor driver board was broken/tattered. 
I've ordered an replacement cable and replaced it, all plugs are plugged in fine but unfortunately I am getting these errors:

"Gimbal unable to connect. Repairs required. Contact DJI Support for assistance" - Code: 40012
"IMU initializing. Retrieving IMU data" - Code 30066
"Gimbal calibration error. Restart aircraft or contact DJI Support if the issue persists" - Code: 40011
"Gimbal stuck. Check whether gimbal cover is removed and make sure gimbal can rotate freely on all three axes" - Code: 40002

Interestingly, the first 2 errors were added after the replacement. before (with broken flexcable, they were not here)

- I dont have issues to find GPS Signal, this is working fine
- I am able to calibrate the IMU via DJI Fly App without any problems
- I am not able to calibrate the Gimbal via DJI Fly App (it will be stuck at 0% and move the Gimbal slowly into one corner)
- If I start the calibration via APP and simultanously start the LinearHall + JointCoarse Command, it will go up to 30% and will then be stuck there)

When I do:

 py comm_og_service_tool.py --port com4 WM160 GimbalCalib LinearHall
 py comm_og_service_tool.py --port com4 WM160 GimbalCalib JointCoarse

It works fine and the Gimbal is working fine until I restart the Mavic Mini. 
But in the App it will still show the error messages. 

Does anyone know what could be the problem or has an Idea how to fix this?

Thanks in advance!

linoko, 2022-02-07:
There is no cure for now. I spoke with service and they told that this message will disapper only if drone gone to their service. So dont worry its only warning message and not interfere working of gimbal

alxgacc, 2022-10-14:
Who needs remove error 40011 or 40021, you can write me on WhatsApp +62 812-8430-9143