o-gs/dji-firmware-tools #262: DJI Mavic 2 pro problem does not charge the battery

Lineamoto, 2022-02-05:
I own a DJI Mavic 2 pro model: L1P and I have a problem with one battery that won't charge. I tried to solve with CP2112 board but I didn't succeed because there seems to be something wrong with the electronic board
I saw that this could be solved by changing the electronic board inside the battery itself, so I wanted to ask you if you have that board and what the price is. Can anyone help me
[image]
[image]

linoko, 2022-02-16:
You need to check the voltage of cells, individually and in series. If voltage is under 2.5V by cell it might be not possible to see the battery. What software you use?

Lineamoto, 2022-02-16:
The battery is new, I haven't used it in over a month