o-gs/dji-firmware-tools #265: Mavic mini - on tools calibration USB connection lost before completing

WACITDJ, 2022-02-21:
On windows 10 the allocated com port for the usb only stays available for a short time. When running tools calibration, command below, it starts and before completion loses the usb/com connection (windows make noise indicating it is lost contact) and calibration just stops. Com port 3 shows initially in device manager and then after about 10 seconds disappears. Anyone any ideas.

py comm_og_service_tool.py --port com3 WM160 GimbalCalib JointCoarse

mefistotelis, 2022-02-21:
You're not completely right about the calibration progress.

Anyway, you might try to call `do_assistant_unlock()` before the calibration - maybe that will help.

I'm not sure whether only that is needed, or some kind of heartbeat. I remember seeing one of newer drones having programmed the need of heartbeat, which is just another packet, sent periodically. So if the unlock call won't help, that's the lead to follow next.

WACITDJ, 2022-02-21:
Thanks, just found out I had to run the DJ assistant 2 first, then shut it down. Found in another thread and did not realise you had to do this. This will hold the USB port and stop it going into charging mode. Then use the tools. The JointCoarse works ok. But LinearHall throws an error:
PS C:\dji> py comm_og_service_tool.py --port com3 WM160 GimbalCalib LinearHall

Info: The Gimbal will slowly move through all positions in all axes, several times. It will take around 30 seconds.

Error: Unrecognized response to calibration command LinearHall request.