o-gs/dji-firmware-tools #266: Mavic mini calibration LinearHall error

WACITDJ, 2022-02-21:
JointCoarse calibration goes ok. Linearhall causing error. Anyone any ideas:

PS C:\dji> py comm_og_service_tool.py --port com3 -vv WM160 GimbalCalib LinearHall
Opened com3 at 9600

Info: The Gimbal will slowly move through all positions in all axes, several times. It will take around 30 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 88 a2 20 04 08 02 77 c1
Sending packet...
Waiting for reply...
Retrieved 1 packets (14b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a 88 a2 80 04 08 01 fd e7
Error: Unrecognized response to calibration command LinearHall request.

mjnhchj, 2022-03-07:
> Its a normal message. py comm_og_service_tool.py --port com4 WM160 GimbalCalib LinearHall sends this: Prepared binary packet: 55 0e 04 66 0a 04 88 a2 20 04 08 02 77 c1 Sending packet... Waiting for reply... Retrieved 1 packets (14b), dropped 0 fragments (0b) Received response packet: 55 0e 04 66 04 0a 88 a2 80 04 08 01 fd e7 Error: Unrecognized response to calibration command LinearHall request.
> 
> And this is normal, because this us used by DJI SPARK, but works on other drones. Note: After do this , and restart drone you will receive warning message on App : error 40011& 40012

Same too. After do this. I received warning "gimbal calibration error" 40011. Although before, it is normally

linoko, 2022-03-07:
> > Its a normal message. py comm_og_service_tool.py --port com4 WM160 GimbalCalib LinearHall sends this: Prepared binary packet: 55 0e 04 66 0a 04 88 a2 20 04 08 02 77 c1 Sending packet... Waiting for reply... Retrieved 1 packets (14b), dropped 0 fragments (0b) Received response packet: 55 0e 04 66 04 0a 88 a2 80 04 08 01 fd e7 Error: Unrecognized response to calibration command LinearHall request.
> > And this is normal, because this us used by DJI SPARK, but works on other drones. Note: After do this , and restart drone you will receive warning message on App : error 40011& 40012
> 
> Same too. After do this. I received warning "gimbal calibration error" 40011. Although before, anything is normal

Its normal to have this warning, because its for spark

mjnhchj, 2022-03-10:
I want to remove warning ..... How can i do it?

alxgacc, 2022-10-11:
Who needs remove error 40011 or 40021, you can write me on WhatsApp +62 812-8430-9143

Ljubo32, 2022-11-04:
You must have two drones with same model and replace camera from one to another and do calibration, you may have error incompatible gimbal data but that is no problem, and replace again to return camera and do calibration again and that fix the problem. I fixed mavic air 2 one is old firmware us version, second is new firmware eu version.