o-gs/dji-firmware-tools #267: BQ30Z55 - DJI Ronin M Battery wont unseal

tommyponny, 2022-02-24:
Hi, I am trying to get my deep discharged dji Ronin battery back to life. I can connect to the BQ30Z55 with the Raspberry Pi and read it out as well. But unfortunately the unseal command does not work. It runs through and also writes that it has written something, but when I check the status the battery is still in status 3 sealed. I assume that another password must be used. Can someone help me ? I would appreciate any help. Thanks a lot.

jensschiefer, 2022-06-26:
Same problem - Received byte[20] from command ManufacturerAccess.UnSealDevice has invalid length - Thanks a lot - grate project

Krisjehaes, 2022-07-17:
Same issue here. I received a not working battery that was unsealed. I sealed it, but I cannot unseal it again. Also I have a battery status ERROR CODE:7. Not sure what I can do with that, No PF flafs found, all fine.

pi@batterypi:~/toolz/dji-firmware-tools $ ./comm_sbs_bqctrl.py -vvv --dev_address 0x0b sealing FullAccess
Opening i2c:1
Raw write: DATA=22 3e
Raw write: DATA=20 3e
Raw write: DATA=22 3e
Raw FirmwareVersion response: 0b 43 07 01 01 00 27 00 03 85 02 00 8f 8f 8f
Auto-selected chip: BQ40z307, Texas Instruments BQ40z307 chip for DJI
Importing comm_sbs_chips/BQ40z307.py
Store ManufacturerAccess.SecKeyWord0: 00 WORD=0xffff
Write ManufacturerAccess: CMD=00 WORD=ff ff
Store ManufacturerAccess.SecKeyWord1: 00 WORD=0xffff
Write ManufacturerAccess: CMD=00 WORD=ff ff
Reading write_word_subcommand command at addr=0xb, cmd=0x0, type=uint16, opts={'subcmd': <MANUFACTURER_ACCESS_CMD_BQ40.OperationStatus: 84>}
Query ManufacturerAccess.OperationStatus: 00 WORD=0x54
Write ManufacturerAccess: CMD=00 WORD=54 00
Raw ManufacturerAccess.OperationStatus response: 04 06 03 04 00 c9
MA.OperationStatus:	0x00040306	bitfields	Operational Status bits
      SYS_PRESENT_LOW:	0=Inactive	[PRES]	System present input state low
       DSG_FET_STATUS:	1=Active	[DSG]	Discharge FET status
       CHG_FET_STATUS:	1=Active	[CHG]	Charge FET Status
      PCHG_FET_STATUS:	0=Inactive	[PCHG]	Precharge FET Status
          FUSE_STATUS:	0=Inactive	[FUSE]	FUSE input status
            SMOOTHING:	0=Inactive	[SMOTH]	CEDV Smoothing status
      TRIP_POINT_INTR:	0=Inactive	[BTPI]	Battery Trip Point Interrupt
        SECURITY_MODE:	3=Sealed	[SEC]	Security Mode
    SHUTDOWN_LOW_VOLT:	0=Inactive	[SDV]	Shutdown triggered via low pack voltage
        SAFETY_STATUS:	0=Inactive	[SS]	Safety mode status
    PERMANENT_FAILURE:	0=Inactive	[PF]	Permanent Failure mode status
 DISCHARGING_DISABLED:	0=Inactive	[XDSG]	Discharging Disabled
    CHARGING_DISABLED:	0=Inactive	[XCHG]	Charging Disabled
           SLEEP_MODE:	0=Inactive	[SLEEP]	Sleep mode condition met
       SHUTDOWN_BY_MA:	0=Inactive	[SDM]	Shutdown activated by SMBus command
          LED_DISPLAY:	0=Off	[LED]	LED Display state
         AUTH_ONGOING:	1=Active	[AUTH]	Authentication ongoing
   AUTO_CC_OFFS_CALIB:	0=Cal Done	[ACALM]	Auto CC offset calibration by SMBus cmd
    RAW_ADC_CC_OUTPUT:	0=Inactive	[CALOC]	Raw ADC/CC data calibration output
    RAW_CCOFFS_OUTPUT:	0=Inactive	[CALOO]	Raw CC offset data calibration output
     XLOW_SPEED_STATE:	0=Inactive	[XL]	400 kHz transmission SMBus mode
          SLEEP_BY_MA:	0=Inactive	[SLEPM]	SLEEP mode activated by SMBus command
     INIT_AFTER_RESET:	0=Inactive	[INIT]	Initialization after full reset
       SMB_CAL_ON_LOW:	0=Not in Cal	[SLCAL]	Auto CC calibration when the bus is low
    ADC_MEAS_IN_SLEEP:	0=Inactive	[SLPAD]	ADC Measurement in SLEEP mode
     CC_MEAS_IN_SLEEP:	0=Inactive	[SLPCC]	Current Check measurement in SLEEP mode
       CELL_BALANCING:	0=Inactive	[CB]	Cell balancing status
   EMERGENCY_SHUTDOWN:	0=Inactive	[EMSHT]	Emergency Shutdown

BatteryStatus:	0x48c7	bitfields	Battery's Alarm and Status bit flags
                ERROR_CODE:	7=Unknown Error	[EC]	Function error code
          FULLY_DISCHARGED:	0=Not fully	[FD]	Battery capacity is depleted
             FULLY_CHARGED:	0=Not fully	[FC]	Battery is full
               DISCHARGING:	1=Yes	[DSG]	Battery is discharging
               INITIALIZED:	1=Recalibrate	[INIT]	State of calibration/configuration
      REMAINING_TIME_ALARM:	0=Inactive	[RTA]	Remaining time to depletion alarm tripped
  REMAINING_CAPACITY_ALARM:	0=Inactive	[RCA]	Remaining capacity alarm tripped
 TERMINATE_DISCHARGE_ALARM:	1=Active	[TDA]	Battery capacity is depleted
     OVERTEMPERATURE_ALARM:	0=Inactive	[OTA]	Temperature is above pre-set limit
    TERMINATE_CHARGE_ALARM:	1=Active	[TCA]	Charging should be suspended
        OVER_CHARGED_ALARM:	0=Inactive	[OCA]	Battery is fully charged

negis02, 2023-08-19:
hi, for unseal you need to use  DJI_Battery_Killer_ver.0.4beta_26.09.2021:  https://github.com/o-gs/dji-firmware-tools/issues/214
with USB to I2C atater: https://diyprojectslab.com/fix-dji-battery-with-dji-battery-killer/
and for reset flags  on rpi: 
`sudo python3 comm_sbs_bqctrl.py -vvv --dev_address 0x0b --chip BQ30z55 trigger ManufacturerAccess.PermanentFailDataReset`

this all worked for me on DJI Ronin M Battery

un48ures, 2025-12-14:
Tried it back in 2025 now and had some problems.
I struggled even with doing it like @negis02 told.
For me the key moment was to use "Full Access" before "unseal" in the dji battery killer tool.

Also by using

`./comm_sbs_bqctrl.py -v --bus "smbus:1" --dev_address 0x0b --chip BQ30z55 --short monitor BQStatusBitsMA`

one can also check the SEC and PF Flags. SEC=1 (or also SEC=2?) showed me that it is unsealed.