o-gs/dji-firmware-tools #270: DJI Smartcontroller SDR UART debug port

mkameke, 2022-03-09:
Hello, 

does somebady try to get a connection on the UART of the SDR board from the Smartcontroller?

There some activity during the boot but not in a readable format... Does somebody know the port settings or protocol? 
It seams there is ARM CPU, maybe this is a debug port...

Faresthug, 2022-09-25:
Hello did you get any answer because itry the same thing to update the system ?? Help

casept, 2025-07-08:
I get the following output when set to `921600` baud. The console does not seem to be interactive, though it does echo back key presses:

```
HA(cpu0) runing.

U-Boot 2011.12 (Jun 16 2021 - 15:53:37)

chip id: 0x44414971, rom id: 0x1 bb_id: 0x282e05
shutdown by keyon disable.
parse_shutdown_reason: value from 0x5d reg is: 0x0
----------last end of day reason: unknown
parse_bootup_reason: value from 0x5c reg is: 0x4
+++++++++start of day reason: charger cable plug in
Main battery voltage = 3642 mv
Meet low battery voltage and USB plug in, voltage 3642
mmc set rst_n_function failure!
SD/MMC1: 0
ddr init
vendor:[Man 700100 Snr 012ff014] product:[EE4ED] revision[3.4]
get ddr mid=0x70010045, did=0x45344544!!
use default ddr!!
zq_select 0
training :1 0
DRAM:  512 MiB
mmc set rst_n_function done!
SD/MMC1: 0
boot normal
board_id: volt1 = 1644mV, volt2 = 1641mV
board is RM500_RC_V3

Starting kernel at 0x1f800000...
Uncompressing Linux... done, booting the kernel.
[    0.048651] c0 1 (swapper/0) i2c-0: clock: 26000000, speed:400000
[    0.049825] c0 1 (swapper/0) i2c-1: clock: 26000000, speed:400000
[    0.051050] c0 1 (swapper/0) i2c-4: clock: 10660000, speed:100000
[    0.199549] c0 1 (swapper/0) Initramfs unpacking failed: junk in compressed archive
[    0.224912] c0 1 (swapper/0) comip_mmc0_set_power: set card vdd ret = -6 vdd = 16
[    0.250271] c0 1 (swapper/0) Failed to create heap debugfs at /ion/heaps/lc_ion_lcmem_heap
[    0.266557] c0 1 (swapper/0) [UREG]Ureg Driver Registered
[    0.291789] c0 37 (kworker/u10:1) comip_mmc0_set_power: set card vdd ret = -6 vdd = 0
[    0.668385] c0 1 (init) init: /dev/hw_random not found
[    0.669993] c0 1 (init) init: cannot open '/initlogo.rle'
[    1.442240] c2 1 (init) init: /dev/hw_random not found
[    1.444669] c2 1 (init) init: The sn.txt is not exist in the path of amt!
[    1.493317] c2 1 (init) init: property 'sys.powerctl' doesn't exist while expanding '${sys.powerctl}'
[    1.495043] c2 1 (init) init: powerctl: cannot expand '${sys.powerctl}'
[    1.496289] c2 1 (init) init: using deprecated syntax for specifying property 'persist.sys.usb.config', use ${name} instead
[    4.869542] c3 1 (init) android_usb: already disabled
[    4.870820] c3 1 (init) init: using deprecated syntax for specifying property 'sys.usb.config', use ${name} instead
[    4.873679] c3 82 (ueventd) init: add network interface: rndis0
[    4.882676] c0 82 (ueventd) init: exec CMD: /system/bin/ifconfig rndis0 192.168.42.2
[    4.891823] c3 1 (init) init: using deprecated syntax for specifying property 'sys.usb.config', use ${name} instead
[    5.094536] c0 82 (ueventd) init: android_usb CONNECTED, usb_acc_mode1=0
[    5.210570] c3 82 (ueventd) init: android_usb CONFIGURED, usb_acc_mode1=0
[    5.245001] c2 329 (dji_hdvt_gnd) i2c-1: wait rx timeout!
[    5.246040] c2 329 (dji_hdvt_gnd) i2c-1-0x11: xfer error -110.
[    5.482998] c1 1 (init) init: untracked pid 480 exited
```

jankais3r, 2026-09-24:
The UART connection is in fact interactive and provides a root shell, but you have to interrupt the standard boot by sending a key early.

The easiest way is to use an automated Python script:
```python
import serial, time
s = serial.Serial('/dev/tty.usbmodem5C4C0540161', 921600, timeout=0)
t0 = time.time()
while time.time() - t0 < 20:
    s.write(b'\x03 \r') # Ctrl+C, space, Enter
    data = s.read(4096)
    if data:
        print(data.decode(errors='replace'), end='', flush=True)
    time.sleep(0.005)
```

The moment you see a bunch of `#`s being printed in your Python console, stop the script and switch to a normal serial emulator.

```
% picocom -b 921600 /dev/tty.usbmodem5C4C0540161
Type [C-a] [C-h] to see available commands
Terminal ready
# id; getenforce; uptime
uid=0(root) gid=0(root)
Disabled
up time: 00:00:31, idle time: 00:01:47, sleep time: 00:00:00
# cat /proc/cmdline
watchdog_thresh=3 console=ttyS3,921600 vmalloc=412M maxcpus=4 android firmware_class.path=/vendor/firmware isolcpus=4 initrd=0x07400000,1M lcpart=mmcblk0=gpt:0:2000:200,ddr:2000:2000:200,env:4000:2000:200,panic:6000:2000:200,amt:8000:20000:200,factory:28000:4000:200,factory_out:2c000:4000:200,recovery:30000:8000:200,normal:38000:8000:200,system:40000:40000:200,vendor:80000:20000:200,cache:a0000:80000:200,blackbox:120000:400000:200,userdata:520000:1ec000:200  chip_sn=162b4144 board_sn=13RCK3X0093B2R production quiet board_id=0xe5001001
# cat /proc/partitions
major minor  #blocks  name

  31        0         32 mtdblock0
 179        0    3702784 mmcblk0
 179        1       4096 mmcblk0p1
 179        2       4096 mmcblk0p2
 179        3       4096 mmcblk0p3
 179        4       4096 mmcblk0p4
 179        5      65536 mmcblk0p5
 179        6       8192 mmcblk0p6
 179        7       8192 mmcblk0p7
 179        8      16384 mmcblk0p8
 179        9      16384 mmcblk0p9
 179       10     131072 mmcblk0p10
 179       11      65536 mmcblk0p11
 179       12     262144 mmcblk0p12
 179       13    2097152 mmcblk0p13
 179       14    1007616 mmcblk0p14
 179       96        512 mmcblk0rpmb
 179       64       2048 mmcblk0boot1
 179       32       2048 mmcblk0boot0
>/dev/null || ls -l /dev/block/by-name/                                       <
lrwxrwxrwx root     root              1980-01-01 00:14 amt -> /dev/block/mmcblk0p5
lrwxrwxrwx root     root              1980-01-01 00:14 blackbox -> /dev/block/mmcblk0p13
lrwxrwxrwx root     root              1980-01-01 00:14 cache -> /dev/block/mmcblk0p12
lrwxrwxrwx root     root              1980-01-01 00:14 ddr -> /dev/block/mmcblk0p2
lrwxrwxrwx root     root              1980-01-01 00:14 env -> /dev/block/mmcblk0p3
lrwxrwxrwx root     root              1980-01-01 00:14 factory -> /dev/block/mmcblk0p6
lrwxrwxrwx root     root              1980-01-01 00:14 factory_out -> /dev/block/mmcblk0p7
lrwxrwxrwx root     root              1980-01-01 00:14 gpt -> /dev/block/mmcblk0p1
lrwxrwxrwx root     root              1980-01-01 00:14 normal -> /dev/block/mmcblk0p9
lrwxrwxrwx root     root              1980-01-01 00:14 panic -> /dev/block/mmcblk0p4
lrwxrwxrwx root     root              1980-01-01 00:14 recovery -> /dev/block/mmcblk0p8
lrwxrwxrwx root     root              1980-01-01 00:14 system -> /dev/block/mmcblk0p10
lrwxrwxrwx root     root              1980-01-01 00:14 userdata -> /dev/block/mmcblk0p14
lrwxrwxrwx root     root              1980-01-01 00:14 vendor -> /dev/block/mmcblk0p11
# mount
rootfs / rootfs ro 0 0
tmpfs /dev tmpfs rw,nosuid,relatime,size=8192k,mode=755 0 0
devpts /dev/pts devpts rw,relatime,mode=600 0 0
proc /proc proc rw,relatime 0 0
sysfs /sys sysfs rw,relatime 0 0
[... 243 lines cut ...]
tinymix
tinypcminfo
tinyplay
toolbox
top
touch
udtperf
umount
uptime
ureg_test
usbmuxd
valgrind
vmstat
vold
watchprops
wipe
wpa_cli
# ls /sbin
adbd
dji_verify
healthd
system_check
ueventd
watchdogd
```